NicolasBernaerts/tasmota #1: Restart Exception 3

Vincent835, 2020-05-11:
Bonjour,

Pour commencer, merci pour ce développement.

Je suis nouveau sur github.com, alors je ne sais pas si je respecte le protocole pour ouvrir une "issue".
Je me permets aussi, par facilité, de rédiger en francais.

J'ai installé votre version tasmota téléinfo sur un sonoff basic (sonoff basic RF R2 Power v1.0)
Le problème est que dès que j'active la téléinfo dans le menu, le sonoff redémarre.
L'erreur est la suivante :

> {"RestartReason":{"Exception":3,"Reason":"Exception","EPC":["4000bf64","00000000","00000000"],"EXCVADDR":"4025a8e7","DEPC":"00000000"}}

Après ce redémarrage, si je reste sur la console, cela fonctionne correctement. Ci-dessous un extrait des données téléinfo :

> 22:21:50 MQT: tele/tasmota/SENSOR = {"Time":"2020-05-11T22:21:50","ENERGY":{"TotalStartTime":"2020-05-11T20:51:17","Total":39626.512,"Yesterday":0.000,"Today":39626.512,"Period":8,"Power":[880,880,440],"Current":[4.000,4.000,2.000]},"Timezone":{"STD":{"Offset":60,"Month":10,"Week":0,"Day":1},"DST":{"Offset":120,"Month":3,"Week":0,"Day":1}},"Teleinfo":{"ADCO":"031776013513","OPTARIF":"HC..","ISOUSC":"20","HCHC":"008524506","HCHP":"008294743","PTEC":"HC..","IINST1":"004","IINST2":"002","IINST3":"002","IMAX1":"060","IMAX2":"060","IMAX3":"060","PMAX":"06103","PAPP":"01790","HHPHC":"A","MOTDETAT":"000000","PPOT":"00","SINSTS1":"716","SINSTS2":"716","SINSTS3":"358"}}

Mais dès que je reviens sur le menu principal et que je sélection un des menus, le sonoff reboote.

Par précaution, j'ai fait plusieurs reset : d'abord reset 6, puis reset 2.
Cela ne change rien.
Avant d'activer la téléinfo, le sonoff marche bien. Je peux aller d'un menu à l'autre.
dès que la téléinfo est activée (Teleinfo 1200) le sonoff reboote lors d'un changement de menu.

Pour info :
- j'ai configuré le module avec GPIO1 - TX et GPIO3 - RX.
- L'affichage des données Téléinfo sur l'interface Web n'apparait plus quand la téléinfo est activée.

Auriez-vous une explication, un paramètre à changer?

Je vous remercie
Cordialement

Vincent

NicolasBernaerts, 2020-05-11:
Bonsoir Vincent,
Votre retour est très intéressant car il semble que vous utilisiez un compteur avec une configuration différente de ce que j'ai eu sous la main pour les tests de cette extension Teleinfo.
D'après les données Teleinfo, vous avez un contrat triphasé avec heures pleines / heures creuses.
Je n'ai jamais eu ce type de compteur en test.
Le problème doit sans doute provenir d'une trame teleinfo inconnue qui n'est pas traitée correctement.
Vous serait-il possible de m'envoyer un log de la sortie série de votre adaptateur Teleinfo afin d'analyser les trames ?
Cela devrait ressembler à quelque chose comme cela :
`ADCO 020522001636 2
OPTARIF BASE 0
ISOUSC 30 9
BASE 026645008 *
PTEC TH.. $
IINST 008 _
IMAX 026 G
PAPP 01780 1
MOTDETAT 000000 B

ADCO 020522001636 2
OPTARIF BASE 0
ISOUSC 30 9
BASE 026645009 +
PTEC TH.. $
IINST 007 ^
IMAX 026 G
PAPP 01760 /
MOTDETAT 000000 B`
Si vous pouviez capturer une bonne centaine de messages, cela me permettra d'essayer de reproduire le problème.
A bientôt.
Nicolas

Vincent835, 2020-05-11:
Bonsoir,

Merci pour cette réponse rapide.
Voilà la capture.
[log teleinfo 12-05-2020.txt](https://github.com/NicolasBernaerts/tasmota/files/4612458/log.teleinfo.12-05-2020.txt)

Vincent

NicolasBernaerts, 2020-05-11:
Vincent,
Classiquement une trame Teleinfo est constituée de la manière suivante :

    STX (002 h) indique le début de la trame.
    Corps de la trame
    ETX (003 h) indique la fin de la trame.

Dans vos logs, je ne vois pas trace des caractères 0x02 (STX) et 0x03 (ETX).
Le problème doit venir de là. Du coup, le buffer de réception doit faire un overflow.
Est-ce que votre compteur est en mode Historique ou en mode Standard ?

Vincent835, 2020-05-11:
Nicolas,

C'est un compteur Linky en mode historique.
J'ai refait une capture plus longue avec des alertes dépassement d'intensité.
[log teleinfo 12-05-2020_b.txt](https://github.com/NicolasBernaerts/tasmota/files/4612527/log.teleinfo.12-05-2020_b.txt)

NicolasBernaerts, 2020-05-12:
C'est très étrange.
Dans vos logs, les trames Teleinfo ne sont pas séparées par les caractères STX et ETX.
Le problème vient sans doute de là, car le sonoff exploite ces caractères pour séparer les trames.
Quel adaptateur utilisez-vous entre le compteur et le Sonoff ?

Vincent835, 2020-05-12:
Bonjour,

Voici de nouvelles captures.
Dans le log txt, il apparait un caractère spécial et dans l'hex il y a des couples 03 02.
Mon adaptateur correspond exactement au schéma que vous proposez. Je l'avais pris sur le site hallard.me.

[serial_20200512_1330.txt](https://github.com/NicolasBernaerts/tasmota/files/4615330/serial_20200512_1330.txt)
[serial_hex_20200512_1332.txt](https://github.com/NicolasBernaerts/tasmota/files/4615331/serial_hex_20200512_1332.txt)

Bonne analyse et bonne journée

Vincent

Vincent835, 2020-05-13:
Bonjour Nicolas, 

Je viens de réaliser que je ne vous ai pas donné la bonne référence de la teleinfo v2. Ce n'est pas la vôtre, mais le fork de rjjrdadev. 
J'espère que cela ne vous a pas fait perdre de temps. 

Bonne journée 

Vincent

NicolasBernaerts, 2020-05-13:
Bonjour Vincent,
La version de ce fork est effectivement une vieille version qui date de 2019 et qui ne gérait que les compteurs monophasés.
Votre compteur étant un compteur triphasé, il vous faut utiliser la dernière version téléchargeable sur ce repo. J'ai publié ce matin une mise à jour qui règle certains problèmes spécifiques au triphasé.
Cette version détecte les compteurs triphasés et propose un graph de consommation des 3 phases sur 24h.
Tenez moi au courant.
Nicolas

Vincent835, 2020-05-13:
Bonsoir,

Nickel, ça marche. Je vous remercie pour ce correctif rapide.

Je viens de m'apercevoir que le menu Domoticz n'est pas activé.
Puis-je le réactiver ou n'est-il pas compatible avec les fonctions que vous avez ajoutées?

Bonne soirée

Vincent

NicolasBernaerts, 2020-05-13:
Bonsoir,

Je l'ai désactivé pour rester en dessous des 500ko.

Mais Domoticz devrait pouvoir être réactivé sans problème majeur car l'extension Teleinfo respecte les callback standards de Tasmota.

closed 2020-05-15 by NicolasBernaerts