NicolasBernaerts/tasmota #2: SetOption19 Autodiscovery HomeAssistant inopérant

Yannonyme, 2020-12-17:
Bonjour Nicolas,

Un grand merci pour cette contribution sur Tasmota.
Malheureusement je constate un problème sur l'option SetOption19 pour activer l'auto discovery sur Home Assistant en utilisant ton tasmota.bin
L'option bien que validée en console (SetOption19 1 > SetOption On) est inopérante (comme expurgée du firmware)
En faisant l'update en OTA vers TASMOTA 9.2.0 cette fonction est immédiatement opérationnelle.

Question subsidiaire, est il possible de rajouter un comptage d'impulsion pour un compteur gaz classique, les deux étant souvent l'un à coté de l'autre ?

Merci

NicolasBernaerts, 2020-12-19:
Bonsoir Yannonyme,

Cela parait normal car j'ai désactivé les options USE_DISCOVERY et MQTT_HOST_DISCOVERY sur les différents buils pour minimiser la taille du firmware. 
Sur quelle version en avez vous besoin ?
Suivant la taille je pourrais réactiver ces options.

Yannonyme, 2020-12-19:
Bonjour Nicolas,

Merci pour votre réponse.
C'est pour un ESP8266 (NodeMCU ESP-12E).

Cordialement,
Yann

NicolasBernaerts, 2020-12-19:
Je viens de publier une nouvelle version activant les options USE_DISCOVERY, MQTT_HOST_DISCOVERY et USE_HOME_ASSISTANT.
Est-ce que cela règle l'auto-discovery dans Home Assistant ?

Yannonyme, 2020-12-20:
Super ! Ca fonctionne !
Pensez vous qu'il soit possible d'y intégrer un compteur d'impulsion pour le compteur gaz afin d'en faire un compteur d'énergies complet ?

Encore merci.
Yann

NicolasBernaerts, 2020-12-20:
Désolé, j'y ai bien pensé, mais je n'ai actuellement ni compteur de gaz, ni compteur d'eau à impulsion.
Dès que l'on change l'un des 2 compteurs, je regarde.
Nicolas

closed 2020-12-20 by NicolasBernaerts