NicolasBernaerts/tasmota #3: TIC standard et suivi des tensions

Vincent835, 2021-03-07:
Bonjour Nicolas,

J'ai des soucis de chutes de tension et j'aimerais bien pouvoir suivre les tensions des 3 phases de mon alimentation.
J'ai vu que le mode standard le permet.
Pour le moment mon compteur est en TIC historique, mais si votre application permet le suivi des tension, je demanderai le changement de mode.
Est-ce le cas?

Merci

Vincent

Vincent835, 2021-03-12:
Bonjour,

Je suis passé au TIC standard et suis arrivé à faire ce que je voulais avec la version officielle de Tasmota.

Vincent

closed 2021-03-12 by Vincent835

NicolasBernaerts, 2021-03-22:
Bonsoir Vincent,
Ayant finalement basculé mon Linky en mode standard, je viens de publier une nouvelle version v7.5 testée en TIC Standard. Cette version exploite les étiquettes URMSx et affiche le voltage en temps réel.
Nicolas