NicolasBernaerts/tasmota #5: Téléinfo - messages MQTT

Vincent835, 2021-04-10:
Bonjour Nicolas,

C'est encore moi.

J'utilisais sur Tasmota 9.3.1.1 l'option 108 pour disposer des messages MQTT RESULT à partir desquels je pouvais traiter les données avec des règles et transmettre sur DOMOTICZ les infos que je souhaitais.
`19:53:50.015 MQT: tele/tasmota_DF20CB/RESULT = { "ADSC":"031776013513","VTIC":2,"NGTF":"      BASE      ","LTARF":"      BASE      ","EAST":27416857,"EASF01":15506794,"EASF02":9861893,"EASF03":826037,"EASF04":514699,"EASF05":442412,"EASF06":265022,"EASF07":0,"EASF08":0,"EASF09":0,"EASF10":0,"EASD01":9975618,"EASD02":8227473,"EASD03":2960408,"EASD04":6253358,"IRMS1":2,"IRMS2":2,"IRMS3":2,"URMS1":225,"URMS2":237,"URMS3":245,"PREF":12,"PCOUP":12,"SINSTS":1305,"SINSTS1":482,"SINSTS2":390,"SINSTS3":433,"SMAXSN":6243,"SMAXSN1":3094,"SMAXSN2":3214,"SMAXSN3":2158,"SMAXSN-1":5714,"SMAXSN1-1":3559,"SMAXSN2-1":1954,"SMAXSN3-1":2064,"CCASN":790,"CCASN-1":1550,"UMOY1":229,"UMOY2":240,"UMOY3":239,"STGE":"003A4001 ...`

Je n'arrive pas à obtenir ce message sur ta version et mes règles ne marchent pas sur le message SENSOR que produit ta version.
`19:49:06.821 MQT: tele/tasmota_DF20CB/SENSOR = {"Time":"2021-04-10T19:49:06","TIC":{"ADSC":"031776013513","VTIC":"02","DATE":"E210410194907","NGTF":"BASE","LTARF":"BASE","EAST":"027416815","EASF01":"015506752","EASF02":"009861893","EASF03":"000826037","EASF04":"000514699","EASF05":"000442412","EASF06":"000265022","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"009975618","EASD02":"008227431","EASD03":"002960408","EASD04":"006253358","IRMS1":"002","IRMS2":"002","IRMS3":"002","URMS1":"223","URMS2":"242","URMS3":"243","PREF":"12","PCOUP":"12","SINSTS":"01370","SINSTS1":"00462","SINSTS2":"00480","SINSTS3":"00428","SMAXSN":"06243","SMAXSN1":"03094","SMAXSN2": ...`

Pour info, un extrait de la règle utilisée :
`rule1 on URMS1 do var4 %value% endon`

Y a-t-il un moyen d'avoir le message d'origine sans reprendre le code?

Je te remercie

Vincent

NicolasBernaerts, 2021-04-10:
Bonsoir Vincent,

Dans le menu "configure teleinfo" tu peux demander le json teleinfo pour
tous les messages.

Est ce que cela répond à ta question ?

Vincent835, 2021-04-10:
Bonsoir Nicolas,

C'est bien ce que j'ai fait mais le message JSON n'est pas le même que sur la version originale de Tasmota.
Du coup ma règle ne marche pas.

J'ai mis les 2 versions de messages JSON dans mon premier message.

Vincent835, 2021-04-10:
Après réflexion, ce n'est peut-être pas la forme du message JSON qui bloque mais la version de Tasmota.
Ce n'est peut-être qu'à partir de la v9.3.1.1 que les triggers sur les messages JSON fonctionnent.

Vincent835, 2021-04-17:
En fait cela marche bien.
Il y a des champs imbriqués. En utilisant la règle suivante, j'obtiens bien la tension 1 dans la variable var4.
`rule1 on TIC#URMS1 do var4 %value% endon`

J'ai passé des heures à faire des essais, sans succès.
Maintenant que cela marche, j'ai réessayé sur toutes les dernières version 7.7.1, 7.7.2, 7.8. tout est ok.

Je ne comprends pas, ce que je faisais pour que cela ne marche pas du premier coup.

Du coup je ferme l'issue.

Merci, je devrais maintenant arriver à utiliser ta version avec Domoticz.

closed 2021-04-17 by Vincent835