NicolasBernaerts/tasmota #6: support du mode Producteur

sublimz, 2022-06-05:
Bonjour,

@NicolasBernaerts ; Tout d'abord, merci beaucoup pour cette release et le travail de qualité.
Je dispose de deux compteurs : un linky en mode historique dont je relève la téléinfo sans problème avec la version teleinfo 9.7. 
et un second : linky en mode standard (protocole teleinfo Producteur) car revente totale de la production; sur ce dernier, j'obtiens le résultat suivant :
20:39:55.759 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:55","METER":_{"PH":1,"ISUB":0,"PSUB":0,"PMAX":0,"U1":230,"P1":0,"W1":0,"I1":0.0,"C1":1.00}}
20:39:56.011 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:56","TIC":{}}
20:39:56.768 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:56","METER":{"PH":1,"ISUB":0,"PSUB":0,"PMAX":0,"U1":230,"P1":0,"W1":0,"I1":0.0,"C1":1.00}}
20:39:57.014 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:57","TIC":{"Kr":"]00000rc\|'"}}
20:39:57.759 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:57","METER":{"PH":1,"ISUB":0,"PSUB":0,"PMAX":0,"U1":230,"P1":0,"W1":0,"I1":0.0,"C1":1.00}}
20:39:58.022 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:58","TIC":{}}
20:39:58.759 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:58","METER":{"PH":1,"ISUB":0,"PSUB":0,"PMAX":0,"U1":230,"P1":0,"W1":0,"I1":0.0,"C1":1.00}}
20:39:59.012 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:59","TIC":{}}
20:39:59.759 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:39:59","METER":{"PH":1,"ISUB":0,"PSUB":0,"PMAX":0,"U1":230,"P1":0,"W1":0,"I1":0.0,"C1":1.00}}
20:40:00.023 MQT: tele/Production_solaire/SENSOR = {"Time":"2022-06-05T20:40:00","TIC":{"K}:K00p[2":"Hi"}}_

Je n'arrive pas interpréter s'il s'agit d'un problème de mon circuit, ou si les balises Producteurs sont inconnus et provoque ce résultat? Merci

NicolasBernaerts, 2022-06-05:
Hello @sublimz,
 A priori le problème doit venir de votre interface.
En mode historique la résistance de l'interface TIC peut être assez élevée (4.7k). En mode standard, elle doit être souvent beaucoup plus faible (1k). Essayez de changer la résistance jusqu'à ce que la section TIC contienne des clés propres et lisibles. Ensuite, il faudra peut être adaptée le décodage des informations en mode producteur (je n'ai que des linky en mode conso sous la main.

sublimz, 2022-06-06:
Bonjour,
Merci pour cette précieuse info, j'ai essayé sans succès 1K, 2K (1K+1K), 1.2K ..
Je vais me procurer d'autres résistances
Dans la mesure où le problème provient de mon interface, je clos ce ticket; merci

closed 2022-06-06 by sublimz

sublimz, 2022-06-06:
Edit..
J'ai finalement enlevé la résistance pull-up et j'obtiens à présent des trames complètes,
exemple :

ADSC | "supprimé"| '
-- | -- | --
VTIC | 02 | J
DATE | E220606134759 | K
NGTF | PRODUCTEUR | .
LTARF | INDEX NON CONSO | 0
EAST | 000001600 | V
EASF01 | 000001600 | )
EASF02 | 000000000 | #
EASF03 | 000000000 | $
EASF04 | 000000000 | %
EASF05 | 000000000 | &
EASF06 | 000000000 | '
EASF07 | 000000000 | (
EASF08 | 000000000 | )
EASF09 | 000000000 | *
EASF10 | 000000000 | "
EASD01 | 000001600 | '
EASD02 | 000000000 | !
EASD03 | 000000000 | "
EASD04 | 000000000 | #
EAIT | 001966489 | 0
ERQ1 | 000000236 | F
ERQ2 | 000000000 | <
ERQ3 | 000640591 | V
ERQ4 | 000053400 | J
IRMS1 | 006 | 4
URMS1 | 238 | G
PREF | 02 | A
PCOUP | 02 | [
SINSTS | 00000 | F
SMAXSN | E220606062108 00180 | 4
SMAXSN-1 | E220605135907 00190 | Z
SINSTI | 01376 | M
SMAXIN | E220606134533 01370 | .
SMAXIN-1 | E220605141926 01800 | M
UMOY1 | E220606134000 238 | 0
STGE | 003A0301 | =
MSG1 | PAS DE MESSAGE | <
PRM | "supprimé" | =
RELAIS | 000 | B
NTARF | 01 | N
NJOURF | 00 | &
NJOURF+1 | 00 | B
PJOURF+1 | 00008001 NONUTILE NONUTILE NONU | 9

Mon objectif est à présent de récupérer EAIT dans Home assistant pour obtenir ma balance conso/production

NicolasBernaerts, 2022-06-09:
Bonjour,
Les trames semblent beaucoup mieux.
J'ai repéré un bug de traitement de EAIT.
Je vais publier le correctif.
Mais n'ayant aucun compteur en mode producteur je ne pourrais le tester.
Serait il possible d'enregistrer le flux de sortie pendant 10mn et de me l'envoyer pour le permettre de le rejouer ?

sublimz, 2022-06-09:
Bonjour,

Merci, j'ai fait la màj ; je ne vois pas de différence sur EAIT mais je n'avais pas non plus relevé de bug.
J'ai tenté une capture en activant tcp_start et en relevant la sortie en mode RAW d'un client putty.
Je ne sais si c'est rejouable en l'état (?) De plus, à cette heure, plus de soleil, ni de production ..
Je peux réitérer la capture avec un peu plus de précision sur le format attendu si il faut à un autre moment.
[putty.log](https://github.com/NicolasBernaerts/tasmota/files/8873093/putty.log)

NicolasBernaerts, 2022-06-09:
Bonsoir,

Je n'ai pas republié les binaires, seulement les sources. Il faut
recompiler la dernière version.

Pour le tcp_start, cela devrait être sans problème.

Avec une version de log, cela sera beaucoup plus simple de valider le mode
producteur.

Bonne soirée.