NicolasBernaerts/tasmota #7: tasmota compatibility

ObiYoann, 2022-07-12:
Hi,
I was looking to compile this for my ESP32S2, as this target is not available in the binary format (and my WEMOS D1 MINI ESP32 is not compatible with the teleinfo32.bin as this binary was compiled with PSRAM, and the D1 provided by Charles does not support PSRAM, bootloop with error initializing PSRAM), and i tried to compile it by following the instructions, however i am facing many build errors, dur to i think incompatibility between the tasmota release i was using and the one this source code is based upon.
I was trying to build using tasmota v12.0.2 as base, which is the latest and the current "stable" build.
At first i had issue regarding xdrv_01_webserver.ino, it complained that ./html_compressed/HTTP_HEADER1_NOES6.h is missing, indeed it is. I then compared the content of xdrv_01_webserver.ino from the one provided here in your repo to the stock one, i couldn't find any meaningfull difference that would be specific to your version, it appeared to be an older revision of the same file. Is it the case?

What would be the best version of tasmota to build your version upon?
In v12.0.2, xdrv* and xsns* files are supposed to go in tasmota/tasmota_xdrv_driver and tasmota/tasmota_xsns_sensor dubfolder, not directly in tasmota folder, i was believing, so i'm not sure if it is worth it to continue trying to build this using v12.0.2 as the base, or if i should find a more suitable version, what do you think?

Thanks for your help and have a nice day!

NicolasBernaerts, 2022-07-17:
Hello ObiYoann,
I've ported **teleinfo** to Tasmota 12, but it was wtill not published.
Under Tasmota 12, compilation is a little bit different as ESP8266 16M and ESP32 partitioning are now standard and because of the new file structure.
I've uploaded latest files and updated compilation instructions.
It should now compile easily.
Let me know ...

NicolasBernaerts, 2022-07-20:
I've just ordered some esp32-s2 boards from Ali express to check proper compilation parameters.

ObiYoann, 2022-07-21:
So i imported your updates (again thanks!).
I also added a new target in platformio_override.ini :
`[env:tasmota32s2-tic-fs]
extends                     = env:tasmota32s2
; build_flags                 = ${env:tasmota32s2.build_flags} -DMY_LANGUAGE=fr_FR -DUSE_FTPSERVER
build_flags                 = ${env:tasmota32s2.build_flags} -DMY_LANGUAGE=fr_FR
board                  = esp32s2_FS
board_build.filesystem = littlefs`
I also added a new file in boards folder, named "esp32s2_FS.json" and mimicking the changes between "esp32_4M.json" and "esp32_4M_FS.json" (so changing the partitions file to "partitions/esp32_partition_app1856k_fs1344k.csv").
Other than that, user_config_override.h is your file, no other thing were enabled in it.
I compiled successfully then flashed my board using the "factory" bin file, but the board only boots if i compile without the FTPSERVER. I don't quite know how to debug this board as it uses usb only when booted in flash mode, by pressing the flash button before resetting it. If booted in normal mode there is nothing visible computer side... :(
I'm happy to report that the partition layout worked fine :
[image]
Anyway it does appear to work, without the FTP Server at least! I don't think i would use it so i'm fine right now with this solution... Thanks A LOT!

NicolasBernaerts, 2022-07-22:
Thanks for the feedback.
I'll try to check what is the trouble with FTPSERVER option as soon as I receive my esp32s2.
Cheers

NicolasBernaerts, 2022-08-04:
Hello,
I just received my ESP32S2.
I updated and uploaded files to handle it for teleinfo.
About FTP server, boot was hanging because of Init function. I changed it to start the service manually thru ftp_start command. I now works perfectly.

closed 2022-08-04 by NicolasBernaerts