NicolasBernaerts/tasmota #10: on Console, uploading an "xxx5.csv" file throught web, it duplicate it (day, week...), I'm on 10.3 release...

chf-6309, 2023-03-29:
Hello Nicolas,
first, thank you for that great development: never I saw an TeleInfo on ESP that's so simple to use, configure, and a lot of embodiment.
Only ONE thing, for ME, is the programming of the target ESP32 done under Linux, that will create and format the LittleFS partition.
I did program the ESP32 on my Windows 11 laptop, so, I couldn't find out how to upload the binaries with the tasmota-flash (downloaded from your tool directory), even after installing Python on Windows 11...
So, programming under Windows, there is a "small" LittleFS with a size of 320kB on the ESP32 4M, that's why on a 10 days base, I upload all .csv files, and delete the xxx02.csv to xxxx1.csv on the ESP32. Then I have room for the yearly graphics!
When uploading the file using Firefox 111.01 (right click the file, upload to a directory), all files are uploaded, but only the xxx05.csv create an empty file (teleinfo-day-5) and the tele-info-5(1).csv that contain the data. Same for -week05.
I'm not sure that the ftp reproduce that issue (I'm not at home).

Maybe a command that will create the LitteFS from TeleInfo program, on the "Console" could be great (at least for me!)
Kind Regards
-=CHF=-

NicolasBernaerts, 2023-03-29:
Hello,
Thanks for the feed-back !
As ESP is very low on resources, the embedded FTP server accepts only one simultaneous connexion.
So you can't use a browser to upload files as it will typically use 2 connexions.
You need to use a client like filezilla and to configure it for 1 simultaneous connexion.
It should work.
Cheers

chf-6309, 2023-04-02:
Hello Nicolas,
thanks for you (to) fast answer ;o)
But "_So you can't use a browser to upload files as it will typically use 2 connexions._", yes, à home, (local connection) I use FileZilla, and set it up for only one connection as recommended on the readme, and it work.
Here (far away, +34°C), I used gave a try to FireFox, to download the files: and it works (FireFox clever?). All files could be downloaded to my laptop, bur only that with the final "5" download, BUT, the original name, size 0, and the second try (with an (1) after the filename), was OK. So, I delete the empty one, and rename the "(1)" one.
When I'm back, I will upgrade to the 11.0 release.

Is there any information about programming the ESP with your tool under Windows 11?
I still couldn't get that 1.2MB LittleFS partition when using an other programming tool.

Again, bravo for your development!
Kind Regards.
-=CHF=-

NicolasBernaerts, 2023-04-06:
Thank CHF !
When downloading with firefox, you may download thru http, not ftp.
About programming under windows 11, I'm sorry but I abandoned Microsoft OS 10 years back. So I really won't know how to help.
Try to install Ubuntu on an old laptop ... you'll be amazed 😀
Cheers

closed 2023-05-31 by NicolasBernaerts