NicolasBernaerts/tasmota #12: Erreur sur valeur transmise à home assistant

McRMX, 2023-05-31:
Bonjour, 

Je viens tout juste d'installer votre version de Tasmota [tasmota-teleinfo.bin.gz] sur un D1mini.
Je recupere mes infos.

[image]

Seulement de façon assez aléatoire je récupère une valeur erroné dans la journée.  (environ 9000 kWh).

Ayant ajouter la variable "energy today" dans la panneaux "Energy" de Home Assistant, le graph s'adapte à la valeur la plus grande et la courbe n'est plus bonne.

Avez vous déjà rencontré ce problème ?

[image]
[image][image]

NicolasBernaerts, 2023-05-31:
Bonsoir @McRMX,
Je viens de déposer une nouvelle version basée sur Tasmota 12.5.0.
Pourriez-vous vérifier que le problème persiste avec cette version ?

McRMX, 2023-06-01:
Je test et reviens vers vous.
Petite remarque : ce lien ne fonctionne pas ou plus.
These projects are explained in detail under http://www.bernaerts-nicolas.fr/iot/

NicolasBernaerts, 2023-06-01:
Hello,
Les graphs sont interessants à 2 niveaux :

 - 382 messages : c'est le nombre de messages reçus depuis le boot. c'est très faible. est-ce que votre tasmota ne reboot pas à cause d'une alimentation un peu faible ?
 
 - pics à 9000kWh entre 9h et 11h totalement plat ... est ce que vous envoyez vraiment la bonne donnée ?

Pour avoir la puissance instantanée en Wh, il faut activer l'option METER, clé W1 :
 2:44:40.456 MQT: turenne/compteur/tele/SENSOR = {"Time":"2023-06-01T22:44:40","METER":{"PH":1,"ISUB":45,"PSUB":9000,"PMAX":9000,"U1":232,"P1":1238,"W1":965,"I1":5.0,"C1":0.78}}

Pour récupérer le total en kWh, il faut activer l'option "TIC" et repérer la clé servie par votre compteur.

McRMX, 2023-06-01:
En effet il semblerait que Energy today reparte régulièrement de 0. Un reboot ferait cela ?
Je vais changé l'alimentation pour m'assurer de ça.

Les infos envoyé a Home assistant sont ajouté a l'aide d'un add-on ( https://www.home-assistant.io/integrations/tasmota)
Il récupère toutes les infos en théories.
[image]

Concernant la configuration, j'ai effectivement pas tout compris. J'ai cherché un peu de doc mais j'ai pas trouvé.(d'ou l'indication d'un lien HS qui correspondait peut être a des infos supplémentaires).

Apres votre dernier message j'ai changé les paramètres comme ceci.
[image]

NicolasBernaerts, 2023-06-02:
Hello,
D'après le site du plugin, il doit s'appuyer sur les messages ENERGY

tele/SENSOR = {"Time":"2023-06-02T07:03:16","ENERGY":{"TotalStartTime":"2023-02-24T23:45:20","Total":26305.301,"Yesterday":16.881,"Today":4.221,"Period":0,"Power":500,"ApparentPower":625,"ReactivePower":375,"Factor":0.81,"Voltage":234,"Current":2.671},

Vous pouvez donc configurer "Message Data" sur "None" car ENERGY est publié en standard.

Vérifiez en mode console que ENERGY soit publié régulièrement et que la valeur "Power" soit cohérente.
Si c'est le cas, le problème doit venir du traitement fait par le plugin Home Assistant.

N'utilisant pas Home Assistant mais OpenHab, j'aurais du mal à vous aider sur ce sujet.

McRMX, 2023-06-02:
La valeur power est coherente. 
[image]

Effectivement "Messages / Cos Phi" la valeur Message repasse régulièrement a zero. --> Et du coup le Energy Today est remis a zero également.
[image]

D'ou j'imagine l'erreur de calcul. 

J'imagine qu'elle devrait le faire seulement a minuit. Qu'est ce qui pourrait produire ça ?

NicolasBernaerts, 2023-06-02:
Le plus courant est une faiblesse de l'alimentation. Cela provoque des reboots intempestifs. Votre courbe de puissance montre des parties sans données. Il faudrait chercher dans ce sens.

McRMX, 2023-06-02:
Je n'ai pas compris votre 1er phrase, sans doute une erreur de frappe.
J'ai changé toutes l'alimentation sur le D1mini Cable / chargeur, j'ai également remplacé le D1mini. Ce qui explique les interruptions. Les coupures correspondent au different test que j'ai fait.
Hier soir tard 00h30 et ce matin après 9h.

Je n'y touche pas ce soir pour voir la différence avec la courbe que je vous ai envoyé.
y a t'il un moyen de savoir le nombre de reboot / un log un peu plus bavard ?

Je vois quelques fois apparaitre une ligne rebott / erreur qui ne s'affiche pas tout le temps.

NicolasBernaerts, 2023-06-02:
Ces éléments sont dans "Information"

**Bout count** présente le nombre total de reboot depuis le début
**Restart Reason** indique la cause du reboot (**Software/System restart** signale par exemple un redémarrage à la demande de l'utilisateur)

NicolasBernaerts, 2023-06-03:
Je viens de publier une version 11.1 où le compteur **Energy today** est sauvegardé entre les reboot.
Cela devrait aider à régler le problème des incohérences d'historique.

McRMX, 2023-06-03:
J'ai pas eu trop le temps aujourd'hui d'y jeter un coup d'oeil.

J'ai pu voir tout de meme bcp de reboot.
[image]

Merci pour la modif. Je test rapidement

NicolasBernaerts, 2023-06-03:
Cela ressemble a un problème connu pour les cartes WeMos non originales.
https://www.letscontrolit.com/forum/viewtopic.php?t=6720

McRMX, 2023-06-04:
J'ai pu mettre la dernière MAJ 11.1.

J'ai eu bon espoir ce matin avec 1000 messages affich, l'energy today semblait cohérente.

Mais quelques minutes plus tard sans doute après un reboot.(J'ai reset le compteur "bount count" et ce matin je suis déja a 27 sans doute comme vous l'avez dit avec une carte WeMos non originales). 
[image]
Il semblerait qu'il n'y ai plus d'erreur sur le calcul, mais la valeur energy today repasse a zero.

Il arrive quelques fois que mon Active power soit à zero (Panneaux solaire) pourrait t'il y avoir une erreur de calcul a ce moment la ?

Ou trouve t'on un D1mini ou ESP32 officiel (J'avoue être un peu perdu avec ces modèles de cartes).

NicolasBernaerts, 2023-06-05:
Bonjour,

La valeur Energy Today en enregistrée en cas de reboot planifié, de mise à jour, ... mais pas en cas de reboot intempestif.

closed 2023-06-05 by NicolasBernaerts