NicolasBernaerts/tasmota #13: Support Autoconsommation et TIC standard

sublimz, 2023-08-29:
Bonjour,

J'ai une petite installation en autoconsommation monophasé sans revente,
Sur l'afficheur de mon Linky, j'ai un index d'injection qui s'incrémente lorsque ma production est supérieure à ma conso.

je ne retrouve pas cet index dans la TIC mode standard consommateur(?)
La valeur de l'index injection ne serait pas transmis par la TIC, ni l'étiquette SMAXIN (Puissance app. max. injectée n) ?

Si quelqu'un qui fait aussi de l'autoconsommation peut confirmer, Merci

Une des trames

  "Time": "2023-08-29T10:12:52",
  "TIC": {
    "ADSC": "*************",
    "VTIC": "02",
    "DATE": "E230829101254",
    "NGTF": "H PLEINE/CREUSE",
    "LTARF": "HEURE PLEINE",
    "EAST": "019290594",
    "EASF01": "009649880",
    "EASF02": "009640714",
    "EASF03": "000000000",
    "EASF04": "000000000",
    "EASF05": "000000000",
    "EASF06": "000000000",
    "EASF07": "000000000",
    "EASF08": "000000000",
    "EASF09": "000000000",
    "EASF10": "000000000",
    "EASD01": "009649882",
    "EASD02": "009640712",
    "EASD03": "000000000",
    "EASD04": "000000000",
    "IRMS1": "002",
    "IRMS2": "000",
    "IRMS3": "001",
    "URMS1": "239",
    "URMS2": "242",
    "URMS3": "244",
    "PREF": "12",
    "PCOUP": "12",
    "SINSTS": "00208",
    "SINSTS1": "00399",
    "SINSTS2": "00115",
    "SINSTS3": "00000",
    "SMAXSN": "E230829005334 02998",
    "SMAXSN1": "E230829083157 01954",
    "SMAXSN2": "E230829073932 01491",
    "SMAXSN3": "E230829005132 02777",
    "SMAXSN-1": "E230828174225 03054",
    "SMAXSN1-1": "E230828075850 02133",
    "SMAXSN2-1": "E230828182211 01814",
    "SMAXSN3-1": "E230828230529 02751",
    "CCASN": "E230829100000 00183",
    "CCASN-1": "E230829090000 01406",
    "UMOY1": "E230829101000 241",
    "UMOY2": "E230829101000 242",
    "UMOY3": "E230829101000 244",
    "STGE": "003A4401",
    "MSG1": "PAS DE MESSAGE",
    "PRM": "19776266264308",
    "RELAIS": "000",
    "NTARF": "02",
    "NJOURF": "00",
    "NJOURF+1": "00",
    "PJOURF+1": "0000C001 06308002 2230C001 NONU"
  }
}

localhost61, 2023-08-29:
Je confirme. Mon frère est dans le même cas.
Installation PV 4kWc avec injection sur une seule phase et Linky triphasé. Mais tarif de base chez Total Energies.
Il va bientôt remplacer son onduleur pour injecter sur les 3 phases.
Le compteur d'injection s'affiche sur le Linky mais n'est pas transmis via la TIC.
Voici ce qu'il obtient en redirigeant la TIC avec la commande Tasmota "tcp_start 9000" et la capture sous Windows avec l'outil Curl (Champs ADSC et PRM obfusqués) :

ADSC    123456789012    @
VTIC    02      J
DATE    E230826181249           L
NGTF          BASE              <
LTARF         BASE              F
EAST    040769595       <
EASF01  040769595       O
EASF02  000000000       #
EASF03  000000000       $
EASF04  000000000       %
EASF05  000000000       &
EASF06  000000000       '
EASF07  000000000       (
EASF08  000000000       )
EASF09  000000000       *
EASF10  000000000       "
EASD01  040475072       =
EASD02  000294523       :
EASD03  000000000       "
EASD04  000000000       #
IRMS1   001     /
IRMS2   001     0
IRMS3   000     0
URMS1   242     B
URMS2   240     A
URMS3   240     B
PREF    12      B
PCOUP   12      \
SINSTS  00000   F
SINSTS1 00000   7
SINSTS2 00189   J
SINSTS3 00079   I
SMAXSN  E230826173138   03300   <
SMAXSN1 E230826084854   01680   <
SMAXSN2 E230826173137   01240   .
SMAXSN3 E230826133506   02650   1
SMAXSN-1        E230820091934   01200   T
SMAXSN1-1       E230820084840   00610   G
SMAXSN2-1       E230820104026   01180   @
SMAXSN3-1       E230820091935   01380   Q
CCASN   E230826180000   00486   H
CCASN-1 E230826173000   00032   [
UMOY1   E230826181000   241     1
UMOY2   E230826181000   240     1
UMOY3   E230826181000   239     :
STGE    003A4201        @
MSG1         PAS DE          MESSAGE            <
PRM     12345678901234  D
RELAIS  000     B
NTARF   01      N
NJOURF  00      &
NJOURF+1        00      B
PJOURF+1        00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE

localhost61, 2023-08-29:
(Noter l'incohérence sur SINST=0)

sublimz, 2023-08-29:
Ok, c'est donc pas un problème de mon installation.
Perso j'ai des micro-onduleurs, je pourrais injecter sur les 3 phases mais ça n'optimise pas vraiment ma conso par rapport à mon équilibrage. Je tente une autre approche : récupérer la phase qui soutire le plus en temps-réel et aiguiller la production sur cette phase (grâce à un esp8266 et des relais SSR) , en projet ...
@localhost61, Merci beaucoup pour ta réponse

closed 2023-08-29 by sublimz