NicolasBernaerts/tasmota #16: ECO: Signal - Error 403 ()

nyok92, 2023-12-06:
Got a valid token but then "ECO: Signal - Error 403 ()" after eco_enable 1

Deleted Eco AP appliication, created a new one, updated eco key, enabled again eco but still issue.
Could you please advise ?
Thanks

NicolasBernaerts, 2023-12-07:
Quel sont les logs de la console ?
Est-ce que le token a bien été récupéré ?
Ce message pourrait signaler que vous générez un token depuis un compte RTE sur lequel l'application Ecowatt n'a pas été autorisée.

NicolasBernaerts, 2023-12-07:
To debug what is happening, best is to try to connect thru **curl** first.

`
curl --header "Content-Type: application/x-www-form-urlencoded" --header "Authorization: Basic YourBase64RTEKey" https://digital.iservices.rte-france.com/token/oauth/
{
  "access_token" : "4vJO4XEhBjFYi5aPEfBukv0116Avb3N3qtr3UHBToTKecnNHO4yp",
  "token_type" : "Bearer",
  "expires_in" : 7199
}

curl -verbose --header "Content-Type: application/x-www-form-urlencoded" --header "Authorization: Bearer 4vJO4XEhBjFYi5aPEfBukv0116Avb3N3qtr3UHBToTKecnNHO4yp"  https://digital.iservices.rte-france.com/open_api/ecowatt/v4/signals
*   Trying 185.30.132.138:443...
* Connected to digital.iservices.rte-france.com (185.30.132.138) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS header, Finished (20):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS header, Finished (20):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES128-GCM-SHA256
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: C=FR; ST=�le-de-France; L=Paris; O=RTE RESEAU DE TRANSPORT D' ELECTRICITE; CN=digital.iservices.rte-france.com
*  start date: Aug  2 07:46:02 2023 GMT
*  expire date: Sep  2 07:46:01 2024 GMT
*  subjectAltName: host "digital.iservices.rte-france.com" matched cert's "digital.iservices.rte-france.com"
*  issuer: C=BE; O=GlobalSign nv-sa; CN=GlobalSign RSA OV SSL CA 2018
*  SSL certificate verify ok.
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
> GET /open_api/ecowatt/v4/signals HTTP/1.1
> Host: digital.iservices.rte-france.com
> User-Agent: curl/7.81.0
> Accept: */*
> Referer: rbose
> Content-Type: application/x-www-form-urlencoded
> Authorization: Bearer 4vJO4XEhBjFYi5aPEfBukv0116Avb3N3qtr3UHBToTKecnNHO4ypB7
> 
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Date: Thu, 07 Dec 2023 09:47:42 GMT
< Server: Apache-Coyote/1.1
< X-Rate-Limit: [{"window":737,"type":"throttle","remaining":0}]
< Max-Forwards: 19
< Via: 1.1 vespaprdif005vm (), 1.1 vespaprdif005vm ()
< X-CorrelationID: Id-be947165d60e996b8fec41af 0; Id-be947165a177ce50b190a116 0; Id-be947165d80e47e32eb44e37 0
< Cache-Control: private
< Expires: Thu, 01 Jan 1970 01:00:00 GMT
< Content-Type: application/json;charset=utf-8
< Transfer-Encoding: chunked
< 
* TLSv1.2 (IN), TLS header, Supplemental data (23):
{"signals":[{"GenerationFichier":"2023-12-06T23:00:00+01:00","jour":"2023-12-07T00:00:00+01:00","dvalue":1,"message":"Pas d\u2019alerte.","values":[{"pas":0,"hvalue":1},{"pas":1,"hvalue":1},{"pas":2,"hvalue":1},{"pas":3,"hvalue":1},{"pas":4,"hvalue":1},{"pas":5,"hvalue":1},{"pas":6,"hvalue":1},{"pas":7,"hvalue":1},{"pas":8,"hvalue":1},{"pas":9,"hvalue":1},{"pas":10,"hvalue":1},{"pas":11,"hvalue":1},{"pas":12,"hvalue":1},{"pas":13,"hvalue":1},{"pas":14,"hvalue":1},{"pas":15,"hvalue":1},{"pas":16,"hvalue":1},{"pas":17,"hvalue":1},{"pas":18,"hvalue":1},{"pas":19,"hvalue":1},{"pas":20,"hvalue":1},{"pas":21,"hvalue":1},{"pas":22,"hvalue":1},{"pas":23,"hvalue":1}]},{"GenerationFichier":"2023-12-06T23:00:00+01:00","jour":"2023-12-08T00:00:00+01:00","dvalue":1,"message":"Pas d\u2019alerte.","values":[{"pas":0,"hvalue":1},{"pas":1,"hvalue":1},{"pas":2,"hvalue":1},{"pas":3,"hvalue":1},{"pas":4,"hvalue":1},{"pas":5,"hvalue":1},{"pas":6,"hvalue":1},{"pas":7,"hvalue":1},{"pas":8,"hvalue":1},{"pas":9,"hvalue":1},{"pas":10,"hvalue":1},{"pas":11,"hvalue":1},{"pas":12,"hvalue":1},{"pas":13,"hvalue":1},{"pas":14,"hvalue":1},{"pas":15,"hvalue":1},{"pas":16,"hvalue":1},{"pas":17,"hvalue":1},{"pas":18,"hvalue":1},{"pas":19,"hvalue":1},{"pas":20,"hvalue":1},{"pas":21,"hvalue":1},{"pas":22,"hvalue":1},{"pas":23,"hvalue":1}]}
`

NicolasBernaerts, 2023-12-07:
Not so easy ...
I tested URL v5 and I get error 403 !
It seems that the URL you have to use depends on when you created your account with RTE :-(
This is just ridiculous.
So we have to be able to select which version you want to use according to when you created your account !
I'll add this option **eco_version** as a console command.

nyok92, 2023-12-07:
Yes, you should delete your existing app and create a new app with the ECOWatt app API v5.
Then and you should get the v5 link working correctly
but indeed if they update to a new version v6 , v7... we will get the same issue so if we can select the version it's far better :)
thanks

NicolasBernaerts, 2023-12-07:
Yep !
By default it will be v5 but you should be able to change it to be futur ... and past ... proof.
Thanks for your investigations.