NicolasBernaerts/tasmota #18: Give access to old build

Wendigogo, 2024-01-16:
Hello,

I flashed my Denky-D4 with latest build yesterday and it seems to freeze after some time.

Everything was fine with previous build so could you add an "archive" folder with previous builds please? At least one version.

Thanks for your support.

localhost61, 2024-01-16:
@Wendigogo Github keeps an history of previous builds, it's one of its purpose.
- In the [master](https://github.com/NicolasBernaerts/tasmota) Code page, under the Green "Code" button you have a commits counter.
- Click on it, it leads you to the [Commits](https://github.com/NicolasBernaerts/tasmota/commits/master/) page.
- Then click on the title of the build you are interested in, for instance [v13.3.1](https://github.com/NicolasBernaerts/tasmota/commit/bfc608aae842e00277d1cb3bb52a3b9bd66c9b03)
- Here you find the files you are looking for, but less specifically, you may want to "browse files" (the button is located on the right side of the build title bar)
- It leads you to a snapshot of the Code page, where "master" is replaced by the last seven chars of the build id.
- Click on the "Code" button and you can download a Zip of the source code for that [build](https://codeload.github.com/NicolasBernaerts/tasmota/zip/bfc608aae842e00277d1cb3bb52a3b9bd66c9b03).
- You'll find those binaries in he folder teleinfo/binary

NicolasBernaerts, 2024-01-16:
Hello,
Thank to @localhost61 for the history feature explanation.
You are right, there is a compatibility problem with Berry which gives instability on latest build.
I've just published a new 13.3.2 binary version with Berry disabled. It should solve the instability.
I'll try to find out what is the problem with Berry to re-enable it in future.

closed 2024-01-16 by NicolasBernaerts

Wendigogo, 2024-01-17:
> @Wendigogo Github keeps an history of previous builds, it's one of its purpose.

You're right and I finally found the way to go back in time. But it was not really easy for someone who wad used to find a "release" page on github projects.

Best regards