NicolasBernaerts/tasmota #20: Messages Erreur TIC / absence remontées MQTT / Denky D4

jsapede, 2024-01-24:
Bonjour,
J'ai installé hier votre firmware sur un Denky D4 mais j'ai quelques problèmes queje n'arrive pas à a résoudre 

je suis sous contrat TEMPO
J'essaie d'integrer a homeassistant via l'integration TASMOTA 

Je suis sur la derniere version du firmware 
[image]

Mon compteur linky est en mode historique, et il me semble que le energyconfig est correctmenet paramétré 
[image]

J'ai rajouté RTE Tempo et visiblement j'ai acces dans l'interface a toutes les données possibles 
[image]

[image]

je fais toutefois face a un certain nombre de problèmes / interrogations 

Dans la console je vois systematiquement des erreurs TIC remonter alors que ça semble OK dans les "messages" et dans les rendus sur l'affichage principal 

[image]

et ces infos ne remontent pas en MQTT  ni sur le tasmota. Si j'acrtive enegeryconfig tic=1 ça ne change rien du tout
Les infos TEMPO ne remontent pas non plus alors que pourtant on les voit bien passer dans la console

On voit bien ici que j'ai des données de suivi ENERGY, équivalentes à ce qui s'affiche sur la page web du denky + des données METER, mais impossible d'avoir les infos TIC ou ALERT (TEMPO ?)

[image]

C'est coherent avec ce qu'on voit sur le MQTT

> {"sn":{"Time":"2024-01-24T09:52:32","ENERGY":{"TotalStartTime":"2024-01-23T23:15:19","Total":4.331,"Yesterday":0.206,"Today":4.125,"Power":256,"ApparentPower":320,"ReactivePower":0,"Factor":0.80,"Voltage":230,"Current":1.4},"METER":{"PH":1,"ISUB":30,"PSUB":6000,"PMAX":6600,"U1":230,"P1":340,"W1":272,"I1":1.478,"C1":0.80,"P":340,"W":272,"I":1.478}},"ver":1}

si je relance energyconfig tic=1 ça ne change rien aux infos qui remontent 

> {"sn":{"Time":"2024-01-24T10:25:35","ENERGY":{"TotalStartTime":"2024-01-23T23:15:19","Total":4.471,"Yesterday":0.206,"Today":4.265,"Power":275,"ApparentPower":320,"ReactivePower":0,"Factor":0.86,"Voltage":230,"Current":1.4},"METER":{"PH":1,"ISUB":30,"PSUB":6000,"PMAX":6600,"U1":230,"P1":320,"W1":272,"I1":1.391,"C1":0.85,"P":320,"W":272,"I":1.391}},"ver":1}

Autre sujet, qui semble lié, dans la console on voit bien la maj des infos tempo 

[image]

Mais qui semble etre dans une section "TEMPO" et pas "ALERT"

Rien ne remonte dans MQTT/TASMOTA

Je suis débutant dans le TASMOTA mais j'ai creusé toute la journée hier et je coince !

jsapede, 2024-01-24:
Je progresse 
en mettant setoption19 à 1, ça a changé les choses. toutefois la modification de semble pas persistente.
en rebootant le Denky, on voit a nouveau apparaitre l'integraton tasmota dans les flux mqtt (setoption19 0) mais il semblerait que cette fois les TIC soeint bien transmis, mais il semblerait que les maj de valeurs se fassent sur le canal MQTT et pas de la meme façon sur la canal dédié tasmota 
[image]
Cette fois l'integration tasmota fait bien remonter les TIC mais ne remonte pas les METER

jsapede, 2024-01-25:
Bonjour,

je peux essayer de vous enregistrer des données mais mon compteur va passer en mode standard ce jour.
par contre j'ai affiné la problématique.

Mon soucis de remontée des infos dans homeassistant vient plutot, j'ai l'impression, du topic mqtt "tasmota/discovery"qui est utilisé par homeassistant pour découvrir et intégrer les équipements de type tasmota. 

https://www.home-assistant.io/integrations/tasmota/

Il semblerait que dans le firmware Tasmota on ait une option "setoption19" qui est directement dédiée à la communication mqtt sous format "integratrion tasmota" et qui est activée par defaut dans votre build (Setoption19 0) mais que les trames mqtt qui sont envoyées sous ce topic ne sont pas completes (abscence de tempo, relay, ...) et que certaines sont envoyées en doublon dans un autre topic (celui defini dans l'interface web).

Je ne sais pas si je suis clair

NicolasBernaerts, 2024-01-25:
Bonsoir,
Effectivement j'avais désactivé l'option de compilation auto-discovery car je ne l'utilise pas.
De plus, je n'utilise pas HomeAssistant mais Openhab.
J'ai modifié les publications teleperiod afin de publier en groupé les données METER, PROD, COUNTER et ALERT.
Quel ESP utilisez vous pour que je publie une version de test ?

jsapede, 2024-01-25:
Super je peux faire un essai demain je pense

jsapede, 2024-01-26:
je vois que vous avez fait les mises a jour, j'essaie ce soir si je peux sinon demain

Wendigogo, 2024-01-26:
Bonjour,
Je découvre ce fil et j'aurai quelques questions (@jsapede) :
- Peux-tu partager un lien sur la façon dont tu as installé MQTT Explorer dans ton Home Assistant stp ?
- Je viens de mettre la V13.3.5 sur mon Denky et après reboot, je ne vois rien de plus dans l'intégration Tasmota de mon HA.
    - Il y a une manipulation à faire ? Supprimer/ajouter le Denky dans Tasmota ?
    - Ou il faut ajouter un sensor pour qu'il apparaisse dans HA ?
Merci

PS : Je suis en HPHC avec un Linky en mode Historique et tout semble fonctionner correctement pour l'instant.

jsapede, 2024-01-26:
pareil ici, je ne vois rien de plus dans l'integration tasmota, il y a toujours 2 topics qui cohabitent dont que l'essentiel sur le topic tasmota, pas de TEMPO, pas de TIC ou autres

jsapede, 2024-01-26:
https://github.com/adamoutler/mqtt-explorer

NicolasBernaerts, 2024-01-27:
Est-ce que vous savez si l'auto-discovery Home assistant se base sur une émission de topic type teleperiod ?
Pourriez-vous afficher le topic publié au moment de l'auto-discovery ?
Je vais essayer de trouver coté Tasmota où cette publication est gérée.

jsapede, 2024-01-27:
Si vous me donnez les instructions pour vous obtenir ce qu'il vous faut je prendrai le temps de le faire 

Le teleperiod intervient dans la mise a jour des valeurs de l'intégration tasmota, quand on change le timer dans les réglages ça change la fréquence de mise a jour

Je vais voir pour vous sortir le contenu du topic discovery tasmota.

NicolasBernaerts, 2024-01-27:
Les données peuvent être récupérées directement en mode console.

Remarque importante : dans mon build, le teleperiod publie soit les données TIC (très verbeuses, dépendant du compteur et du contrat) soit les données METER, PROD, COUNTER, CAL, ALERT, ... qui sont normalisées et indépendantes du contrat. Les deux sont possibles, mais logiquement vous n'avez pas besoin des 2 en même temps.
Donc logiquement vous ne devriez pas avoir besoin de publier TIC.
Dans le cadre de vos tests, je vous invite à décocher cette option.

jsapede, 2024-01-27:
Je vais regarder 
Sachent que les infos tic m'intéresse pour la gestion du tempo et des délestages qui vont avec

Wendigogo, 2024-01-27:
> Je vais regarder Sachent que les infos tic m'intéresse pour la gestion du tempo et des délestages qui vont avec

Pareil pour moi. J'aimerai récupérer l'information de passage HC / HP pour gérer certains éléments via Home Assistant. Et cette information se trouve dans la "variable" TIC.

NicolasBernaerts, 2024-01-27:
La section CAL est là pour cela.
lv donne la periode actuelle (1 bleu, 2 blanc, 3 rouge)
hp donne le type d'heure (0 creuse, 1 pleine)

Ces données sont alimentées par les infos STGE du compteur et par RTE Tempo si elle est activée.

NicolasBernaerts, 2024-01-27:
La section RELAY donne également l'état des 8 relais virtuels gérés par le compteur (le relai 1 est celui de l'eau chaude sanitaire). Je vais développer très prochainement une version tasmota permettant de récupérer ce signal et de gérer le ou les relais de l'ESP en conséquence. Cela permettra d'exploiter ces signaux même loin du compteur.

jsapede, 2024-01-27:
{"ip":"192.168.0.131","dn":"Teleinfo","fn":["Teleinfo",null,null,null,null,null,null,null],"hn":"teleinfo-7820","mac":"4C7525F5BE8C","md":"Denky D4 (v1.1)","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"13.2.0","t":"teleinfo","ft":"%topic%/%prefix%/","tp":["cmnd","stat","tele"],"rl":[2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":3,"bat":0,"dslp":0,"sho":[],"sht":[],"ver":1}

jsapede, 2024-01-27:
topic tasmota/discovery/MAC/sensors 

{"sn":{"Time":"2024-01-27T12:20:18","ENERGY":{"TotalStartTime":"2024-01-26T17:46:57","Total":11.103,"Yesterday":7.002,"Today":4.100,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":230,"Current":0.0}},"ver":1}

Wendigogo, 2024-01-27:
> La section CAL est là pour cela. lv donne la periode actuelle (1 bleu, 2 blanc, 3 rouge) hp donne le type d'heure (0 creuse, 1 pleine)

Je vais activer cette option, merci.
Je pensai passer par la valeur de "OPTARIF" disponible dans le message du "TIC".

Enjoy !

Wendigogo, 2024-01-29:
Bonjour,

J'ai bien activé l'option "CAL" et désactivé le "TIC" sur mon Denky-D4. J'ai conservé les données "METER & PROD" qui me sont utiles.
Je n'ai rien de plus dans mon Tasmota que ce qui était déjà disponible avant.

Je remarque qu'effectivement, il y a 2 messages successifs qui sont envoyés via MQTT (merci @jsapede pour le lien) à quelques secondes d'intervalles :
    {"Time":"2024-01-29T11:45:20","CAL":{"lv":1,"hp":1,"day":{"lv0":0,"hp0":1,...
    {"Time":"2024-01-29T11:45:15","ENERGY":{"TotalStartTime":"2023-03-12T01:30:15",...
    {"Time":"2024-01-29T11:40:04","CAL":{"lv":1,"hp":1,"day":{"lv0":0,"hp0":1,...
    {"Time":"2024-01-29T11:40:00","ENERGY":{"TotalStartTime":"2023-03-12T01:30:15",...

Je remarque aussi que le message "CAL" est interprété différemment dans MQTT Explorer. Le JSON n'est pas mis en forme :
[image]

Alors que c'est bien le cas dans la section STATE :
[image]
Peut-être parce que les champs sont différents (limitation de MQTT Explorer ?).

Et au redémarrage du Denky-D4, le champs "tele" de MQTT est davantage peuplé, avec notamment un champ SENSOR et un autre "STATE" :
[image]
qui disparaissent après quelque temps :
[image]

Bonne journée

NicolasBernaerts, 2024-01-30:
En analysant la publication Home Assistant, nous sommes sur une trame spécifique. Je ne suis pas sur qu'en ajoutant une donnée elle sera prise en compte par Home Assistant. Mais c'est possible.
Est ce que quelqu'un sait comment Home Assistant interprétarait de nouvelles données dans la trame mqtt ?

jsapede, 2024-01-31:
Je ne suis poas du tout familier de mqtt ni de tasmota mais je vais essayer de vous retranscrire ce que j'en comprends.

Mais pour repondre a votre question oui HA sera capable d'auto integrer de nouvelles données si elles apparaissent 

[image]

De ce que j'en comprends, pour etre autodecouvrable soit en tant que mqtt pur (setoption19 1) il faut que le topic respecte un certain formalisme ici decrit : 

https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery

A priori pour rendre les equipements autedecouvrables en tant que tasmota (setoption19 0), c'est une variante dans le formalisme mqtt avec des topics qui respectent un autre formalisme : 

https://tasmota.github.io/docs/Home-Assistant/

Je pense que dans vos recherches / modificaitons vous avez deja touché à ça. Je m'explique. Les premieres versions de votre firmware que j'ai essayé (1 semaine en arriere), integraient de façon "aleatoire" des infos complémentaire dans la section tasmota. L'autodecouverte qui etait faite integrait alors les parametre generiques "energy" mais aussi parfois TIC ou parfois CALENDAR.

Depuis, les modifications que vous avez faites, ont fait disparaitre tout ça. On ne recupere plus que les données "energy" dans l'integration tasmota. Donc je pense que vous avez du modifier la façon dont ces infos se diffusaient avant et qui parfois etaient correctment capturées par l'integration tasmota.

L'autre point que je vois c'est que dans la trame tasmota que je vous ai transmis, on voit qu'on a tout un tas de paramètres qui sont a "null" ou "0" ou "-1". Est-ce que ce ne sont pas des diffusions "tasmota compatibles" automatiques (setoption19 0) qui ne sont aujourd'hui pas utilisées dans votre firmware ?

NicolasBernaerts, 2024-01-31:
Hello,
N'ayant pas Home Assistant, le débuggage est difficile.
Je vais publier une nouvelle version 13.3.6 qui intègre un certain nombre de données de base dans les trames appelées "Teleperiod" qui sont émises périodiquement et sont sensées contenir toutes les données de type Sensor. J'espère que ce sont ces trames qui sont interprétées par Home Assistant.
Vous pourrez activer toutes les données sauf TIC et regarder si elles apparaissent coté Home Assistant.
Si ce n'est pas le cas, il faudra analyser plus en détail le code de l'auto-discovery ...

jsapede, 2024-01-31:
On fera les beta testeurs 👍

Wendigogo, 2024-02-01:
Hello,
J'ai installé la version 13.3.6 (taggée 13.3.7 ici) et voici mes remarques.
Dans la configuration Teleinfo du Denky-D4, tout est coché sauf TIC et RELAY.

Dans la console, je vois ça (mis en forme par mes soins donc pas forcément aux normes) : 
```
17:18:14.909 MQT: tele/tasmota_EB1F5C/SENSOR = {"Time":"2024-02-01T17:18:14",
    "METER":{"PH":1,"ISUB":45,"PSUB":9000,"PMAX":9000,"U1":230,"P1":350,"W1":203,"I1":1.521,"C1":0.58,"P":350,"W":203,"I":1.521}
    }
17:23:09.916 MQT: tele/tasmota_EB1F5C/STATE = {"Time":"2024-02-01T17:23:09",
    "Uptime":"0T03:38:34","UptimeSec":13114,"Heap":146,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,
    "Berry":{"HeapUsed":4,"Objects":52},
    "POWER":"ON","Dimmer":45,"Color":"417073","HSBColor":"184,43,45","Channel":[25,44,45],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON",
    "Wifi":{"AP":1,"SSId":"XXXXXXx","BSSId":"XXXXXXXXXXX","Channel":2,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:02"}
    }
17:23:13.849 MQT: tele/tasmota_EB1F5C/SENSOR = {"Time":"2024-02-01T17:23:13",
    "ENERGY":{"TotalStartTime":"2023-03-12T01:30:15","Total":1290.817,"Yesterday":22.949,"Today":11.107,"Period":22,"Power":288,"ApparentPower":390,"ReactivePower":0,"Factor":0.74,"Voltage":230,"Current":1.7},
    "ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""},
    "METER":{"PH":1,"ISUB":45,"PSUB":9000,"PMAX":9000,"U1":230,"P1":390,"W1":288,"I1":1.695,"C1":0.74,"P":390,"W":288,"I":1.695},
    "COUNTER":{"HC..":18115522,"HP..":475058},
    "CAL":{"lv":1,"hp":1}
    }
17:23:17.907 TIC: Error [OPTARIF HB]
17:23:18.931 MQT: tele/tasmota_EB1F5C/SENSOR = {"Time":"2024-02-01T17:23:18",
    "CAL":{"lv":1,"hp":1,
        "day":{"lv0":0,"hp0":1,"lv1":0,"hp1":1,"lv2":0,"hp2":1,"lv3":0,"hp3":1,"lv4":0,"hp4":1,"lv5":0,"hp5":1,"lv6":0,"hp6":1,"lv7":0,"hp7":1,"lv8":0,"hp8":1,"lv9":0,"hp9":1,"lv10":0,"hp10":1,"lv11":0,"hp11":1,"lv12":0,"hp12":1,"lv13":0,"hp13":1,"lv14":0,"hp14":1,"lv15":0,"hp15":1,"lv16":0,"hp16":1,"lv17":0,"hp17":1,"lv18":0,"hp18":1,"lv19":0,"hp19":1,"lv20":0,"hp20":1,"lv21":0,"hp21":1,"lv22":0,"hp22":1,"lv23":0,"hp23":1},"next":{"lv0":0,"hp0":1,"lv1":0,"hp1":1,"lv2":0,"hp2":1,"lv3":0,"hp3":1,"lv4":0,"hp4":1,"lv5":0,"hp5":1,"lv6":0,"hp6":1,"lv7":0,"hp7":1,"lv8":0,"hp8":1,"lv9":0,"hp9":1,"lv10":0,"hp10":1,"lv11":0,"hp11":1,"lv12":0,"hp12":1,"lv13":0,"hp13":1,"lv14":0,"hp14":1,"lv15":0,"hp15":1,"lv16":0,"hp16":1,"lv17":0,"hp17":1,"lv18":0,"hp18":1,"lv19":0,"hp19":1,"lv20":0,"hp20":1,"lv21":0,"hp21":1,"lv22":0,"hp22":1,"lv23":0,"hp23":1}
        }
    }
17:23:25.839 TIC: Error [HCHP 00047]
17:23:26.831 MQT: tele/tasmota_EB1F5C/SENSOR = {"Time":"2024-02-01T17:23:26",
    "COUNTER":{"HC..":18115522,"HP..":475059}
    }
```
Avec la coloration c'est plus parlant : 
[image]

On remarque qu'il y a des doublons : la trame de 17:23:13 contient les infos METER, CAL et COUNTER qui sont publiées seules dans d'autres trames.

Je pense que c'est la partie ENERGY qui est interprétée par Home Assistant (désolé, je n'ai pas lu le lien sur Tasmota de @jsapede ) car dans l'interface, on retrouve bien ces éléments : 
[image]
avec les dénominations correspondantes aux champs du JSON "ENERGY + nom du champs + valeur"

Il suffirait peut-être d'intégrer les champs COUNTER et CAL (METER est un doublon de champs présents dans ENERGY dans mon cas) directement dans le champs ENERGY, et nom pas dans des champs séparés pour qu'ils soient vus par Home Assistant.

En gros, d'arriver à quelque chose comme ça : 
```
17:23:13.849 MQT: tele/tasmota_EB1F5C/SENSOR = {"Time":"2024-02-01T17:23:13",
    "ENERGY":{"TotalStartTime":"2023-03-12T01:30:15","Total":1290.817,"Yesterday":22.949,"Today":11.107,"Period":22,"Power":288,"ApparentPower":390,"ReactivePower":0,"Factor":0.74,"Voltage":230,"Current":1.7,
        "HC..":18115522,"HP..":475058,
        "lv":1,"hp":1},
    }
```
En couleurs : 
[image]

En espérant avoir aidé.

NicolasBernaerts, 2024-02-03:
Merci pour cette analyse très pertinente !
Je viens de publier une version 13.3.8 qui met en oeuvre une section ENERGY complétée avec les nouvelles données.
Les hooks de Tasmota ne sont clairement pas fait pour cela, mais cela devrait être ok.
En complément, toutes les sections sont toujours publiées en propre sous tele/SENSOR .
Indiquez moi si l'auto-détection Home Assistant fonctionne maintenant.

jsapede, 2024-02-05:
dsl pas eu le temps ce weekend de me repencher la dessus. je regarde demain

jsapede, 2024-02-06:
Bonjour,
je suis passé en 1.3.3.8 ce matin et j'ai activé totu sauf les RELAY et TIC
entre temps je suis passé en mode standard.

Je ne vois pas de difference sur l'affichage HA : 

[image]

dans la console on voit bien qu'il y a bcp de choses qu isont transmises au MQTT 

[image]

Mais avec toujours le meme problème de doublon de topic MQTT : 

on a le topic MQTT du teleinfo qu'on peut voir sur la capture d'ecran : teleinfo/tele/SENSOR, teleinfo/tele/STATE qui est bien reçu par le HA puisque on le retrouve dans MQTT explorer : 

[image]

le contenu semble OK : 

`{"Time":"2024-02-06T10:29:50","COUNTER":{"HC BLEU":17960872,"HP BLEU":18454345,"HC BLANC":235879,"HP BLANC":198778,"HC ROUGE":112288,"HP ROUGE":77521}}`

mais le topic tasmota (tasmota/discovery/adresse_MAC_du_DENKY/) lui reste désespéréménet vide, (hormis la partie ENERGY, et c'est je pense ce qui empeche l'autodécouverte des complements d'information : 

[image]

si on regarde le contenu de tasmota/discovery/adresse_MAC_du_DENKY/sensors, on retrovue bien la section ENERGY et les infos génériques energie de tasmota qui sont lues et intégrées par HA : 

`{"sn":{"Time":"2024-02-06T10:19:47","ENERGY":{"TotalStartTime":"2024-01-26T17:46:57","Total":162.343,"Yesterday":17.272,"Today":5.059,"Power":387,"ApparentPower":387,"ReactivePower":0,"Factor":1.00,"Voltage":239,"Current":1.6}},"ver":1}`

On voit bien que les sensors qui sont sous le topic dédié tasmota n'incluent pas les autres données de type TEMPO ou TIC alors qu'on les retrouve sur le topic teleinfo

Les autres sous-sections (ici tasmota/discovery/adresse_MAC_du_DENKY/config) restente vides : 

`{"ip":"192.168.0.183","dn":"Teleinfo","fn":["Teleinfo",null,null,null,null,null,null,null],"hn":"teleinfo-7820","mac":"4C7525F5BE8C","md":"Denky D4 (v1.1)","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"13.2.0","t":"teleinfo","ft":"%topic%/%prefix%/","tp":["cmnd","stat","tele"],"rl":[2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":3,"bat":0,"dslp":0,"sho":[],"sht":[],"ver":1}`

en soi j'ai l'impression que la diffusion des infos par MQTT se fait sans pb mais que les valeurs sont simplement adressées dans des topics qui ne sont pas exploitables en integration / autodecouverte.

NicolasBernaerts, 2024-02-06:
Donc l'ajout des étiquettes dans ENERGY ne suffit pas à déclencher l'autodiscovery. 
Si vous faites un ajout manuel, les nouvelles étiquettes présentes dans ENERGY sont utilisables ?

jsapede, 2024-02-06:
pour moi les ajouts n'aterissent pas dans le bon topic mqtt pour etre autodecouvrables. Ce n'est pas un probleme d'etiquette, c'est sans nul doute un probleme de topic de publication.

le topic mqtt qui est definit dans l'interface web (chez moi ça devient teleinfo/tele) et dans lequel arrivent les infos que vous avez rajouté, n'est pas integrable en autodecouverte.

Pour que ça marche il faudrait que ça soit publié dans un topic conforme "autodecouverte tasmota" soit , de ce que j'en comprends, de la forme :  tasmota/discovery/adresse_mac_du_denky

je suppose que comme çà atterit dans un topic mqtt, meme non autodecouvrable, ça doit pouvoir s'integrer manuellement. Je pourrais essayer de faire un essai en manuel des que je trouve 5 minutes

A la limite si ça vous facilite la vie et que vous voulez vous lancer là dedans, je pourrais vous faire une instance en machine virtuelle de homeassistant accessible depuis l'exterieur

jsapede, 2024-02-06:
Si je mets ça dans ma config homeassistant, par exemple, ça marche : 

`mqtt:
  sensor:
    - name: "Puissance Apparente"
      unique_id: "sensor.denky_puissance_apparente"
      unit_of_measurement: "VA"
      state_class: "measurement"
      device_class: "apparent_power"
      state_topic: "teleinfo/tele/SENSOR"
      value_template: "{{ value_json.ENERGY.ApparentPower }}"
`

Je lis bien la valeur qui est dans le JSON, dans le topic teleinfo, dans la rubrique ENERGY, au bon item.

sur le principe pas de pb, si ce n'est que je ne maitrise pas les extractions json avec des espaces mais je vais trouver !

mais bon c'est dommage de ne pas pouvoir beneficier de l'integration tasmota qui devrait permettre de découvrir automatiquement toutes les valeurs

NicolasBernaerts, 2024-02-06:
Effectivement, si l'auto discovery ne fonctionne pas, le plus simple est de revenir sur des étiquettes de base simples à interpréter.

jsapede, 2024-02-06:
La partie energy de base fonctionne bien en autodiscovery. Suffit d'aller récupérer les autres données dans le flux mqtt. J'ai pas besoin de tout, surtout les totalisateurs par couleur et les basculements de couleur pour éviter de dépendre de l'API de RTE

jsapede, 2024-02-07:
Vous sera-t-il possible de remettre les données triées par sections détaillées (comme dans la version .6 il me semble) ? 
J'adapterai mon sensor mqtt en conséquence

jsapede, 2024-02-07:
Si ça peut aider quelqu'un voici comment j'ai fait pour integrer certaines données manquantes a mon integration tasmota (les items ENERGY sont autodecouverts) : 

> mqtt:
  sensor:
    - name: "Denky TEMPO code Couleur"
      unique_id: "sensor.denky_tempo_couleur"
      object_id: "Denky TEMPO Code Couleur"
      device:
        connections: [["mac", "4c:75:25:f5:be:8c"]]
        model: "Denky D4 (v1.1)"
        manufacturer: "Tasmota"
        name: "Linky"
        sw_version: "13.2.0"
      unit_of_measurement: null
      state_class: measurement
      suggested_display_precision: 0
      state_topic: "teleinfo/tele/SENSOR"
      value_template: "{{ value_json.TEMPO.lv }}"

en l'etat, avec les infos dans "devices" qui recoupent celles du tasmota, la lecture directe des flux mqtt permet de completer les éléments autodecouverts.

Dans la listre des équipements mqtt on voit bien notre tasmota qui est apparu et qui est donc bien lié : 

[image]

et si on va dans le détail du device tasmota on voit bien que la valeur complmentaire que j'ai ajouté (ici le code tempo) s'est bien intégrée : 

[image]

Si jamais vous corrigez le parsing des diffrentes sections de données, ça pourrait être interessant de le mettre dans doc pour qu'on puisse piocher ce qui interesse dans le flux MQTT

Seul point que j'ai pas encore resolu c'est l'extraction des JSON avec des espaces dans les noms genre "HC BLEU"

NicolasBernaerts, 2024-02-07:
Je vais publier une nouvelle version avec des étiquettes séparées pour les différents types de données (tel que précédemment).
Concernant les espaces dans les étiquettes JSON, je vais les remplacer par des -. Cela devrait régler les problème de parsing.

localhost61, 2024-02-07:
> Seul point que j'ai pas encore resolu c'est l'extraction des JSON avec des espaces dans les noms genre "HC BLEU"

Tu as essayé avec un niveau supplémentaire de brakets: ["HC BLEU"] ?

jsapede, 2024-02-07:
Il me semble que oui mais j'ai peut être pas tout essayé

jsapede, 2024-02-08:
@tutruie63 
pour avoir les infos en detail du tasmota il faut ouvrir les flux mqtt, j'utilise mqtt explorer : 
https://forum.hacf.fr/t/mqtt-explorer-et-home-assistant/21090

une fois que c'est connecté au mqtt tu peux voir tout le détail des message suivant les topics : 

[image]

on voit ici le mac, le md (model), etc ... faut regarder dans le doc mqtt de ha a quoi correspondent les abreviations, puis y'a plus qu'a reprendre ces éléments dans le descriptif du "device" de ton sensot mqtt. J'ai tatonné un peu mais ça a marché assez vite.

pour en revenir au debut de ton message, j'ai eu ça aussi dans les premieres versions que j'ai testé, ça devait etre la 1.3.3.4 ou 1.3.3.5, avant qu'on commence a essayer de debuger. Effectivement certains YIC remontaient parfois à la decouverte du tasmota (et donc devaient bien tre diffusés soous le bon topic) mais c'etait aléatoire, parfois j'avais les TIC et parfois les CAL.

jsapede, 2024-02-08:
En effet les topics tasmota ne remontent pas dans la console c'est assez deconcertant.

jsapede, 2024-02-09:
@NicolasBernaerts 
ne vous embetez pas a changer les etiquettes ça marche comme ça : 
      value_template: "{{ value_json.COUNTER['HC BLEU'] }}"
il suffira juste de tout repositionner correctement dans les bonnes sections d'origine et je me derai mon template pour recuperer ce qui m'interesse
merci pour votre aide !

NicolasBernaerts, 2024-02-10:
Je viens de publier une version 13.3.9 qui remet en place les différentes sections autonomes.
Le compteur gère désormais les sections METER, TOTAL, CAL, RELAY et ALERT.
En complément, les calendriers RTE suivants sont gérés : ECOWATT, POINTE et TEMPO.
Toutes les sections sont publiées avec un décalage d'une seconde pour éviter les erreurs de réception TIC.

jsapede, 2024-02-10:
Merci
J'essaie d'implémenter ça demain
Je publierai le code ca aidera peut être quelqu'un

jsapede, 2024-02-11:
@tutruie63 
Vous êtes sur que le topic mqtt est bon dans votre cas? Il est calé sur le mien la, et est directement dépendant du nom que vous avez mis dans les réglages du tasmota

NicolasBernaerts, 2024-02-11:
Il manque une virgule après l'étiquette "I=4,956"
Vous devez utiliser une ancienne version de mon firmware, car le problème a été corrigé depuis.
De plus avec la version 13.3.9, toutes les sections sont publiées séparément.
Une mise à jour s'impose ...

jsapede, 2024-02-11:
@tutruie63 
Votre Indentation n'est pas bonne

mqtt:
  Sensor:
    - name :

jsapede, 2024-02-11:
[image]

jsapede, 2024-02-12:
[image]

NicolasBernaerts, 2024-02-13:
N'ayant pas Home Assistant, je ne peux tester la réponse de Copilot.
Mais sur le papier la réponse est impressionnante !

jsapede, 2024-02-14:
@NicolasBernaerts 
C'est surtout plus élégant puisque on va chercher et passer automatiquement le contenu du json dans le topic et en faire les "attributs" d'un seul capteur.
Ça évite quelque dizaines de ligne de configuration

Wendigogo, 2024-02-14:
> @NicolasBernaerts C'est surtout plus élégant puisque on va chercher et parser automatiquement le contenu du json dans le topic et en faire les "attributs" d'un seul capteur. Ça évite quelque dizaines de ligne de configuration, pour chaque valeur qu'on veut remonter. Suffit à la fin de désactiver celles dont on a pas besoin

J'ai essayé cette méthode mais je ne trouve pas le capteur dans la liste... Il faut l'ajouter par la suite ?

Merci

jsapede, 2024-02-14:
@Wendigogo la formulation sensor: - platform: mqtt est obsolete, il faut revoir et il faut surement rajouter un unique_id mais sur le principe le json_attributes_topic qui distribue les valeurs du json dans les attributs du capteur reste valide

ShagoY, 2024-03-20:
Quel dommage d'avoir abandonné l'autodiscovery avec Home Assistant 😭

Dans tous les cas, ce topic est grandement intéressant car je viens de recevoir mon Denky et souhaite aussi l'intégrer complètement dans HA.

@jsapede As tu trouvé 5 minutes depuis.... 😇

@NicolasBernaerts D'abord un grand merci pour ton firmware, je suis plus expérimentés par Esphome du coup je découvre aussi Tasmota mais ton intégration RTE Tempo est au top 👍
J'ai utilisé pendant des années le modules teleinfokit : https://github.com/342apps/teleinfokit
Il permettait de recevoir les trames MQTT dans HA avec l'autodiscovery, je ne sais pas si cela peut aider à relancer ce topic 😊

jsapede, 2024-03-20:
Non j'ai pas fait plus. La première formulation que j'avais fait fonctionne j'ai pas pris le temps de creuser plus loin

NicolasBernaerts, 2024-03-21:
> Quel dommage d'avoir abandonné l'autodiscovery avec Home Assistant 😭
>
> Dans tous les cas, ce topic est grandement intéressant car je viens de recevoir mon Denky et souhaite aussi l'intégrer [...]

Effectivement ce projet est très intéressant ... Je vais regarder comment publier des topics auto discovery sur la base des trames émises.

jsapede, 2024-03-21:
@NicolasBernaerts une machine virtuelle a relier au broker mqtt et ça devrait marcher 
mais je fera volontier le beta testeur

NicolasBernaerts, 2024-03-21:
@jsapede  Si vous pouvez me mettre a disposition une machine virtuelle accessible pour les tests, merci !

jsapede, 2024-03-21:
Ça peut peut être se faire mais il faut que je voie comment, il vous faudrait l'accès aussi au denky
Je regarde si j'arrive à faire quelque chose mais sans garantie

jsapede, 2024-03-22:
je viens juste de voir que le sujet a en fait deja ete abordé ici il y a 4 ans ...

https://github.com/NicolasBernaerts/tasmota/issues/2

ShagoY, 2024-03-22:
En faisant mes recherches et sauf erreur, je suis tombé sur des articles indiquant que SetOption19 pourrait être deprecated.

jsapede, 2024-03-22:
mais c'était déjà le même sujet

[38 more replies]

closed 2024-04-02 by NicolasBernaerts