NicolasBernaerts/tasmota #21: Message METER - Px et Cx a zero

Bidouille28, 2024-01-28:
bonjour 

je viens d'installer votre version teleinfo TASMOTA sur un Denky D4

y a t il une activation du calcul du facteur de puissance (cos φ) ?

je récupère dans jeedom via MQTT ces infos:

{"Time":"2024-01-28T23:52:18","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":241,"P1":2387,"W1":0,"I1":10.000,"C1":0.00,"U2":244,"P2":2092,"W2":0,"I2":9.000,"C2":0.00,"U3":242,"P3":3561,"W3":0,"I3":15.000,"C3":0.00,"P":8040,"W":0,"I":34.000}}

les valeurs W1 W2 W3 C1 C2 C3 à zéro.
Cela est il normal ?

Merci de votre aide

NicolasBernaerts, 2024-01-29:
Bonjour,
Quel type de compteur, quel mode et quel contrat avez vous ?
Suivant les compteurs, il arrive de temps en temps qu'au démarrage, les trames envoyées par le compteur ne permettent pas de lancer le calcul du cosphi. La plupart du temps, il suffit de redémarrer le tasmota et cela démarre.

Bidouille28, 2024-01-29:
merci de ta réponse 
j'ai donc redémarrer le denky et le calcul est opérationnel.
j'ai cependant pas mal d'erreurs .
[image]
aurais tu des conseils pour diminuer le nombre d'erreurs?

merci

NicolasBernaerts, 2024-01-30:
Une recherche Google ou la lecture de l'autre ticket ouvert actuellement répondra à la question.

closed 2024-01-30 by NicolasBernaerts