NicolasBernaerts/tasmota #22: Teleinfo esp8266 v13.3.4 Puisqance réactive non calculée

Vincent835, 2024-02-07:
Bonjour Nicolas, 

Je vous remercie pour ces mises à jour régulières.

J’ai noté sur la version ESP8266-1m que la puissance réactive n’est plus calculée.
Cela marchait sur la v9.7, mais si je ne me trompe pas, pas sur les v11.x, 12.x et 13.x.

Config : LINKY TRI / TEMPO / TIC STANDARD
Installé sur SONOFF BASIC ESP8285

[image]

> 06:46:38.438 MQT: compteur/tele/SENSOR = {"Time":"2024-02-07T06:46:38","ENERGY":{"TotalStartTime":"2022-06-17T01:03:30","Total":8.116,"Yesterday":15.425,"Today":7.620,"Period":[2,2,2],"Power":[252,261,223],"ApparentPower":[414,429,366],"**ReactivePower":[0,0,0]**,"Factor":[0.61,0.61,0.61],"Voltage":[228,231,228],"Current":[1.8,1.9,1.6]}}

Bonne journée 

Vincent

NicolasBernaerts, 2024-02-10:
C'était effectivement un effet de bord de la prise en compte des compteurs Emeraude 2 quadrants qui calculent la puissance réactive. Cela devrait être corrigé dans la version 13.3.9.

Vincent835, 2024-02-11:
Essai v13.3.10.
Puissance réactive OK.
Merci

closed 2024-02-11 by Vincent835