NicolasBernaerts/tasmota #25: Teleinfo - Pas d'informations Bornes TIC avec ESP 8266 et PITInfo v1.5

SpecialOOps, 2024-02-15:
Bonjour, je tatonne à travers tout les forums mais je n'ai toujours pas réussi à bien configurer ces deux appareils.

J'utilise le firmware pré-compilé sur un esp 8266 : tasmota-teleinfo.bin 
J'arrive bien à me connecter dessus et le configurer.
Mais malheureusement je n'ai pas d'informations qui s'affichent : les valeurs reste à 0
[image]

Je suis en mode Historique / Vérifié sur le compteur Linky.
[image]

Et j'ai également la sortie du module PitInfo v1.5 qui va dans le port Rx de l'esp 8266 et bien configuré également...
[image]

Voici le log du démarrage

`00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Chargé de la flash à F9, Compte 66
00:00:00.726 SNS: Hardware Serial
00:00:00.735 NRG: Init driver 15
00:00:00.738 Projet tasmota - Tasmota-TeleInfo Version 13.2.0(tasmota)-2_7_5(2024-02-11T13:33:53)
00:00:00.739 HLP: tcp_help to get help on TCP Server commands
00:00:00.740 HLP: EnergyConfig to get help on all Teleinfo commands
00:00:00.741 HLP: tz_help to get help on Timezone commands
00:00:01.276 WIF: Connexion à l'AP1 xxx Channel 6 BSSId 9C:C9:EB:2A:15:CB en mode 11n comme ESP-Linky...
00:00:02.754 WIF: Connecté
00:00:03.007 HTP: Serveur web actif sur ESP-Linky avec l'adresse IP 192.168.1.246
10:33:15.044 MQT: Tentative de connexion...
10:33:15.094 MQT: Connecté
10:33:15.097 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/LWT = Online (persistant)
10:33:15.100 MQT: maison/tasmota/Wemos_Teleinfo_Linky/cmnd/POWER = 
10:33:15.106 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/INFO1 = {"Info1":{"Module":"Generic","Version":"13.2.0(tasmota)","FallbackTopic":"cmnd/ESP-Linky_fb/","GroupTopic":"maison/tasmota/tasmotas/cmnd/"}}
10:33:15.108 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"ESP-Linky","IPAddress":"192.168.1.246"}}
10:33:15.111 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":26}}
10:33:19.636 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/STATE = {"Time":"2024-02-15T10:33:19","Uptime":"0T00:00:09","UptimeSec":9,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"9C:C9:EB:2A:15:CB","Channel":6,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:04"}}
10:33:19.646 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/SENSOR = {"Time":"2024-02-15T10:33:19","ENERGY":{"TotalStartTime":"2024-02-14T19:20:05","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.0}}
10:33:20.639 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/SENSOR = {"Time":"2024-02-15T10:33:20","ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""}}
10:33:21.642 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/SENSOR = {"Time":"2024-02-15T10:33:21","METER":{"PH":1,"ISUB":0,"PSUB":0,"PMAX":0,"U1":230,"P1":0,"W1":0,"I1":0.000,"C1":1.00,"P":0,"W":0,"I":0.000}}
10:33:22.644 MQT: maison/tasmota/Wemos_Teleinfo_Linky/tele/SENSOR = {"Time":"2024-02-15T10:33:22","TOTAL":{}}`

Voici les infos quand j'execute EnergyConfig
`10:39:13.122 CMD: EnergyConfig
10:39:13.126 EnergyConfig Teleinfo parameters :
10:39:13.127   historique   set historique mode at 1200 bauds (needs restart)
10:39:13.129   standard     set Standard mode at 9600 bauds (needs restart)
10:39:13.131   stats        display reception statistics
10:39:13.133   percent=100  maximum acceptable contract (%)
10:39:13.135   policy=0     message policy : 0=Telemetrie seulement, 1=± 5% Evolution puissance, 2=Tous les messages TIC
10:39:13.137   meter=1      publish METER & PROD data
10:39:13.139   tic=0        publish TIC data
10:39:13.140   calendar=0   publish CALENDAR data
10:39:13.143   relay=0      publish RELAY data
10:39:13.144   counter=1    publish COUNTER data
10:39:13.146   maxv=235     graph max voltage (V)
10:39:13.148   maxva=3000  graph max power (VA or W)
10:39:13.150   nbday=8      number of daily logs
10:39:13.151   nbweek=4     number of weekly logs
10:39:13.153   maxhour=2    graph max total per hour (Wh)
10:39:13.155   maxday=10    graph max total per day (Wh)
10:39:13.157   maxmonth=100 graph max total per month (Wh)
10:39:13.161 MQT: maison/tasmota/Wemos_Teleinfo_Linky/stat/RESULT = {"EnergyConfig":"Done"}`

Je suis donc coincé là et je donne ma langue au chat. Si une petit aide est possible cela m'aiderais bien merci !
ps : j'ai commencé le projet en suivant ce lien : https://ghost.canaletto.fr/home-assistant-esp-teleinfo/

NicolasBernaerts, 2024-02-15:
Bonjour,
Si vous souhaitez de l'assistance au démarrage, je vous conseille d'aller sur le forum https://community.ch2i.eu/ qui dispose d'une section particuliere pour pitinfo.

SpecialOOps, 2024-02-15:
Bonsoir, j'ai effectivement déjà épluché l'entièreté du forum et étant donné que j'utilise la version compilé que vous avez fourni j'ai d'abord posé la question ici :)

NicolasBernaerts, 2024-02-15:
Désolé je n'utilise pas PIT info.

SpecialOOps, 2024-02-16:
D'accord merci encore !

closed 2024-02-16 by SpecialOOps