NicolasBernaerts/tasmota #26: "Energie active injectée totale" EAIT

Yourry, 2024-02-15:
Bonjour à tous,
Je viens tout juste faire le changement de TIC en historique vers standard (sous contrat tempo) car j'ai quelques panneaux solaires et je voulais savoir si mon installation rivalise avec une centrale nucléaire :)
Je pensais récupérer la valeur de suproduction, genre "Energie active injectée totale EAIT" mais à priori non.
N'étant pas un expert, est-ce qu'il y a un truc à faire, car dans les messages reçu j'ai :
[image]

Il n'y a pas d'erreur de trame d'après le menu principale. Un truc à cocher/renseigner ?
Pour info ma demande d'autoconso a été faite officilement. 
@++

NicolasBernaerts, 2024-02-15:
Votre compteur ne génère pas l'étiquette EAIT. Le problème est donc chez votre fournisseur d'accès qui n'a pas finalisé sa configuration.
Quand l'étiquette apparaîtra, une section production devrait apparaître automatiquement sur la page d'accueil.

Yourry, 2024-02-15:
C'est noté, je verais avec EDF alors pour leur demander une mise à jour de la TIC.
Merci pour votre réponse.

Yourry, 2024-02-15:
Merci pour l'aide.

closed 2024-02-15 by Yourry