NicolasBernaerts/tasmota #27: Erreur à gogo

Yourry, 2024-02-15:
Bonjour,
Je viens de m'apercevoir d'un petit problème.
Lorsque la teleInfo est dans le mode suivant :

[image]

 J'ai pas mal d'erreur : 

```
22:47:09.806 TIC: Error [EASF	J]
22:47:10.154 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:10","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224703","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044578","EASF01":"015302158","EASF02":"001638092","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","CCASN":"H240215220000 00430","DATE":"H240215224702","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044578","EASF01":"015302158","EASF02":"001638092","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"}}
22:47:10.904 TIC: Error [IRMSE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	.]
22:47:11.705 TIC: Error [PCOUS	001	C]
22:47:15.555 TIC: Error [MSG1	PAS DE          MEONUTILE NONUTILE NONUTILE NONUTILE	.]
22:47:16.122 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:16","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224708","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044581","EASF01":"015302161","EASF02":"001638092","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809977","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02240","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","EASF01":"015302160"}}
22:47:16.868 TIC: Error [PJOURF+1	00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE ND02	002653599	H]
22:47:18.139 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:18","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224710","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044583","EASF01":"015302163","EASF02":"001638092","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809979","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02238","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","EASD03":"001809978","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02232","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"}}
22:47:19.500 TIC: Error [EAST	01804458092	@]
22:47:20.113 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:20","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224711","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044583","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809980","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02246","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"}}
22:47:21.076 TIC: Error [EASF06	0001367OURF+1	00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	.]
22:47:21.232 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:21","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224713","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044585","EASF01":"015302165","EASF02":"001638092","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809980","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02246","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"}}
22:47:22.126 TIC: Error [NGTF	     T]
22:47:23.129 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:23","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224716","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044586","EASF01":"015302166","EASF02":"001638092","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809982","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02237","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"}}
22:47:24.094 TIC: Error [PREF	09	36	;]
22:47:25.088 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:25","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224718","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044588","EASF01":"015302168","EASF02":"001638092","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809982","EASD04":"001698575","IRMS1":"010","URMS1":"232","PCOUP":"09","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809983","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02238","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"}}
22:47:27.115 MQT: teleinfo/SENSOR = {"Time":"2024-02-15T22:47:27","TIC":{"ADSC":"123456","VTIC":"02","DATE":"H240215224720","NGTF":"TEMPO","LTARF":"HC BLEU","EAST":"018044589","EASF01":"015302169","EASF02":"001638092","EASF03":"000438706","EASF04":"000273881","EASF05":"000255036","EASF06":"000136705","EASF07":"000000000","EASF08":"000000000","EASF09":"000000000","EASF10":"000000000","EASD01":"011882430","EASD02":"002653599","EASD03":"001809985","EASD04":"001698575","IRMS1":"010","URMS1":"232","PREF":"09","PCOUP":"09","SINSTS":"02217","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","SMAXSN":"H240215001741 02432","SMAXSN-1":"H240214225418 04393","CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE N [... cut ...] ,"CCASN":"H240215220000 00430","CCASN-1":"H240215210000 00414","UMOY1":"H240215224000 233","STGE":"013A8000","MSG1":"PAS DE MESSAGE","PRM":"19582344356813","RELAIS":"001","NTARF":"01","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"}}
````

Ce qui fait que le module ne remonte pas les autres infos TIC.
Avec les deux options cochés je devrais pas avoir les lignes MQTT : TIC+METER ?

Merci pour votre aide.

NicolasBernaerts, 2024-02-15:
Très simple : avec un message TIC par seconde, l' ESP n'arrive pas à suivre les publications MQTT.
Changez vos options.

Yourry, 2024-02-16:
Merci pour votre réponse. Quand j'ai l'option `msgtype=3` et `msgpol=1` j'ai un message à toutes les deux minutes :
```
18:07:26.093 MQT: teleinfo/SENSOR = {"Time":"2024-02-16T18:07:26","TIC".......
18:09:36.113 MQT: teleinfo/SENSOR = {"Time":"2024-02-16T18:09:36","METER".......
```
Est-ce qu'il y a un entre-deux, entre toutes les secondes et 120sec ?
Genre toutes les 30sec par exemples ?
Merci par avance,

NicolasBernaerts, 2024-02-17:
Il faut simplement définir un teleperiod à 30 sec. C'est la base de tasmota.

closed 2024-02-17 by NicolasBernaerts

Yourry, 2024-02-17:
Merci pour votre retour.
En effet, j'ai reglé l'option sur "Télémétrie seulement" et via la console envoyé la variable suivante :
`TelePeriod 30`
Cela permet donc d'avoir une trame MQTT toutes les 30sec.