NicolasBernaerts/tasmota #28: ESP32 Olimex POE

nicolas-r, 2024-04-08:
bonjour

j'ai voulu remplacer le firmware par défaut avec le votre. J'ai pris la version tasmota32-teleinfo.bin mais après le flash, je n'ai plus d'interface réseau cablé. Et je ne retrouve pas dans la configuration les entrées ETH Pwr et ETH MDIO/MDC

Est-ce que j'ai pris la mauvaise image ?

ESP Chip Id | 396192 (ESP32-D0WD-V3 v3.0)

Merci

Nicolas

NicolasBernaerts, 2024-04-08:
Bonjour Nicolas, 
Effectivement le support Ethernet n'est pas compilé par défaut car l'immense majorité des ESP32 n'ont que du wifi. 
Mais l'option USE_ETHERNET est effectivement la solution.

nicolas-r, 2024-04-08:
Bonsoir

Merci pour la confirmation, je vais regarder comment compiler une version avec cette option

Bonne soirée

NicolasBernaerts, 2024-04-08:
Je viens de déposer un binaire ESP32 ethernet.

nicolas-r, 2024-04-08:
Merci beaucoup, je suis en train de la flasher.

Je venais juste de merger vos fichiers avec les sources de Tasmota avant de tenter une compilation :)

nicolas-r, 2024-04-08:
Une fois la ligne de conf suivante saisie j'ai récupéré une IP via le port Ethernet
{"NAME":"Olimex ESP32-PoE","GPIO":[1,1,1,1,1,1,0,0,5536,1,1,1,1,0,5600,0,0,0,0,5568,0,0,0,0,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

Je testerai demain si le POE est ok. D'ici la fin de la semaine je devrais aussi pouvoir le connecter à mon Linky, le temps de tirer le câble entre le compteur et ma maison.

Merci beaucoup pour l'image faite rapidement

closed 2024-04-08 by nicolas-r

nicolas-r, 2024-04-09:
Bonjour 

est-ce que vous pensez pouvoir intégrer cette version dans votre chaîne de build pour les prochaines versions ?

Si je souhaite compiler cette version, est-ce que ce repo est à jour ?

Merci