NicolasBernaerts/tasmota #30: Teleinfo today repart à zéro plusieurs fois par jour

sj29fr, 2024-04-26:
Bonjour @NicolasBernaerts,
j'ai flashé le dernier tasmota-teleinfo-4m2m.bin sur un wemos d1 mini.
Mon compteur est un Linky monophasé en 12KVA.
Je rencontre un problème avec le compteur de conso journalisé.
Dans HA, sensor conso_aujourdhui repart à 0 plusieurs fois par jour sans explication.
Après vérification, je viens de me connecter et je vois que l'ESP8266 a fait un reset.
Est-ce qu'il vaudrait mieux utiliser le firmware de base tasmota-teleinfo.bin peut être plus stable ?
Mon besoin est juste de remonter les infos en MQTT vers HA. Je n'ai pas besoin de stocker d'historique.
Je rajoute des screenshots pour montrer l'incohérence entre le total journalier web et le total journalier MQTT.
Cordialement,
Sébastien

sj29fr, 2024-04-26:
[image]
[image]
[image]

sj29fr, 2024-04-26:
En fait, MQTT export "Consommation Aujourd'hui = 38Wh" au lieu de "Energie aujourd'hui 13.358kWh"
Est-ce possible de modifier l'export MQTT ?
A quoi correspond du coup "Consommation Aujourd'hui" car ça me parait faible ?
Et pourquoi repart t-il à 0 pendant la période de 24h ?

NicolasBernaerts, 2024-04-26:
A minuit Conso hier est initialisé à conso aujourd'hui et conso aujourd'hui initialisé à zéro. On parle ici d'un total quotidien en Wh. 
Dans le mqtt,  CONSO.TDAY doit augmenter tout au long de la journée comme conso aujourd'hui. Si elle repasse à zéro dans la journée c'est qu'il y a un problème ... sans doute des redémarrage intempestif de l'ESP sans sauvegarde du compteur.
Pour TeleInfo, je conseille fortement un esp32 car l'esp8266 est tres limité en mémoire et cela peut être la cause de plantages s'il est trop sollicité.
Si vous préférez utiliser la clé du total ENERGY c'est tout a fait possible mais elle est sans doute beaucoup moins précise car issue d'un calcul tasmota et non pas des totaux du linky.
Pour ma part, je n'utilise jamais les données ENERGY, je me contente de les alimenter.

NicolasBernaerts, 2024-04-26:
Pour info, s'il n'y a pas eu de reboot programmé, 1082 trames à 13:06 signifie clairement un reboot intempestif.

sj29fr, 2024-04-26:
Merci pour ces infos.
J'ai un esp32 de côté. Je vais modifier ma carte pour voir si c'est plus stable

NicolasBernaerts, 2024-04-26:
[image]

Voici mon TeleInfo esp8266 en cours d'utilisation. Aucun reboot depuis la dernière mise a jour il y a plus de 4 jours.

sj29fr, 2024-04-30:
Je confirme qu'avec un ESP32, le firmware est OK.
Je vais peut être prendre un ESP32-C3 mini, ça sera plus pratique à intégrer dans mon tableau.

closed 2024-04-30 by sj29fr