NicolasBernaerts/tasmota #31: ESP32 C3 zero board

amrx06, 2024-05-01:
Binary for ESP32-C3 does not boot on ESP32-C3 zero board. This board is very close to Supermini [https://templates.blakadder.com/SuperMini-ESP32-C3.html](url)

Official Tasmota binary for supermini is tasmota32c3-cdc and works fine on ESP32-C3 zero
Can you build this variant?

NicolasBernaerts, 2024-05-01:
Please ?

amrx06, 2024-05-01:
> Please ?

yes please :) and if you have a "buy me a coffee" it will be with pleasure

NicolasBernaerts, 2024-05-01:
Le lien vers la carte ne fonctionne pas.
Où peut on se la procurer ?

amrx06, 2024-05-01:
lien édité.
elle est vendue 3€ sur aliexpress

NicolasBernaerts, 2024-05-03:
> lien édité. elle est vendue 3€ sur aliexpress

Quelle procédure de flash avez vous mis en œuvre pour flasher le firmware esp32c3 ?

amrx06, 2024-05-03:
esptools sous macOS via homebrew et https://tasmota.github.io/install/
chacune avec le firmware .factory ou OTA

Firmware factory : 
`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42045a2c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x48
load:0x403cc710,len:0x644
load:0x403ce710,len:0x2294
SHA-256 comparison failed:
Calculated: 5ece9d5bf1312efcde90bc5d93121ced775d5ffc4c0b0d076c26b33285a599d7
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403cc710`

et ensuite l'écran reste noir

Firmware OTA : 
`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40047ed2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c110020,len:0x46740
load:0x3fc8aa00,len:0x2d78
load:0x40380000,len:0x6b30
load:0x42000020,len:0x106de8
Invalid image block, can't boot.
ets_main.c 333 
`

Chipset : 
`esptool.py v4.7.0
Serial port /dev/cu.usbmodem1201
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.4)`

amrx06, 2024-05-03:
wait… 
après avoir regardé les release infos je vois que la différence entre le firmware cdc et le non cdc est la liaison série via usbc.
**en fait votre firmware fonctionne mais je ne le vois pas sur la console !** 
comme je configurais le réseau wifi via la console je n'étais pas allé voir dans la liste des réseaux dispos, et le serveur wifi n'est pas Tasmota-xxx mais Teleinfo !

NicolasBernaerts, 2024-05-03:
Donc tout est ok ?
Cela va se finir par une bière ... :-)

amrx06, 2024-05-03:
c'est ok oui merci ! 
La console sur le port série c'est quand même pratique, surtout pour voir quand ça ne fonctionne pas.

closed 2024-05-03 by NicolasBernaerts