NicolasBernaerts/tasmota #32: Graphiques sur Conso et Courbes mal formater

Flub1, 2024-05-03:
Bonjour,
je teste actuellement votre Firmware sur un Denky-D4 de Mr Hallard (esp32 8Mb Flash 2Mb PSRAM) car je suis impressionné par le rendu des graphiques que peut produire un si petit microcontrolleur (et le travail fournis pour y parvenir !).
J'ai malheureusement un affichage sur différent navigateurs qui ne correspond pas au résultat attendu:
[image]
le graphique semble décaler vers la gauche pour la partie Consommation
[image]
mais plus gênant, pour la page COURBE , le résultat ressemble a une ligne bleu qui aurait toute ses abscisses a zéro...
[image]
la variation de l'echelle avec les icones + ou - font bien varier la taille de la partie bleu
[image]
la courbe live ne fonctionne pas non plus mais pour la courbes des jours j'ai bien le fichier .csv correspondant avec des valeurs cohérentes
[teleinfo-day-01.csv](https://github.com/NicolasBernaerts/tasmota/files/15200271/teleinfo-day-01.csv)
en regardant le code HTML envoyer par l' esp32 dans la console de firefox on voit que l'objet path de SVG est bien cette ligne bleu
qui devrait etre la courbe ? (la sortie "afficher le source de la page" produit par contre un objet path vide ???)
je n'ais aucune erreur de connection et les fichiers de log semble être ceux annoncés:
[image]
avec un espace dispo de 5Mb, le firmware est celui prévue pour le Denki D4 soit : tasmota32-teleinfo-denkyd4.factory.bin
j'ai cru comprendre que vous etes l'auteur de la partie graphique de ce firmware et c'est pourquoi je me tourne vers vous pour m'éclairer sur ces dysfonctionnement.
La partie serveur TCP fonctionne pour moi mais le FTP fait probleme également, la connection teleinfo/teleinfo fonctionne mais sous linux la commande LS retourne un message lié au mode passif et fillezilla sous windows10 me dit au même endroit:
__Statut :	Connecté
Statut :	Récupération du contenu du dossier…
Commande :	PWD
Réponse :	257 "/" is the current directory.
Commande :	TYPE I
Réponse :	200 TYPE is now 8-bit Binary.
Commande :	PASV
Réponse :	227 Entering Passive Mode (0,0,0,0,195,89).
Commande :	MLSD_
Erreur :	Les données de connexion ne peuvent pas être établies : WSAEADDRNOTAVAIL - Impossible d'assigner la requête d'adresse_
Mais la récupération des fichiers .csv se fait sans probleme dans la page syteme de fichier de Tools.
La remonté MQTT est désactiver chez moi donc le ESP32 n'est pas trop stresser.
Encore bravo pour cette intégration dans Tasmota/Teleinfo

NicolasBernaerts, 2024-05-03:
Concernant les graphs c'est la première fois que j'ai ce retour. Je l'utilise sur Chromium et Firefox sous Ubuntu 22.04 et sur Firefox sous Android 14 sans aucun soucis.

Quelle version de navigateur utilisez vous ? Avez vous essayé de vider le cache ?

NicolasBernaerts, 2024-05-03:
Pour le serveur FTP, le problème doit logiquement venir du fait que vous n'avez pas configuré votre client pour n'utiliser qu'une seule connexion simultanée tel que précisé dans la doc.

NicolasBernaerts, 2024-05-03:
> Pour le serveur FTP, après vérification, il y a effectivement un problème sur la dernière version. Le code n'a pas évolué depuis longtemps et j'avoue ne pas beaucoup utiliser cette fonctionnalité. Je vais regarder s'il n'y a pas eu une régression suite au passage sur les nouvelles librairies Arduino 3.0.

En fait le problème vient du fait que Tasmota 13.4 a introduit une gestion native du FTP se basant sur une version légèrement différente de la librairie que j'utilisait. Les 2 étaient en conflit.
J'ai donc activé la nouvelle version native de Tasmota en remplacement de celle de mon fork.
De ce fait les nouvelles commandes sont les suivantes : 

  # ufsftp 0 : arrêt du serveur FTP
  # ufsftp 2 : démarrage du serveur FTP

Les limitations sont les mêmes : 
  - mode passif
  - 1 seule connexion simultanée

La version 14.3.2 amènera cette évolution.

NicolasBernaerts, 2024-05-03:
Les binaires v14.3.2 viennent d'être publiés.

Flub1, 2024-05-07:
Bonjour,
merci de ce retour rapide, le firmware v14.3.2 résoud tous les problèmes soulevés.
La mise à jour OTA (une première pour moi) conserve les fichiers ET les réglages précedents, elle necessite juste deux lancements pour fonctionner chez moi...
Seul l'utilisation du FTP reste un peux "rugueuse", FileZilla veut une connection authentifiée de base et la réponse du serveur n'est pas suffisante pour lui faire changer d'avis (passage par le mode manuel obligatoire) et le client FTP sous Linux demande des réflexes affutés pour éviter le timeout a la frappe du mot de passe :-) mais la connection, la visualisations et le rapatriement des fichiers fonctionne désormais !
encore bravo pour la résolution express des différents points de dysfonctionnement.

closed 2024-05-07 by Flub1