NicolasBernaerts/tasmota #33: 3 phase problem

reguiny, 2024-05-14:
I am using teleinfo and when the current used is below 1 amp the display shows all three phases as the same value.
[image]

NicolasBernaerts, 2024-05-14:
Hello,
Votre compteur est en mode historique.
Il ne transmet donc pas la puissance par phase mais :
- une puissance globale
- un courant par phase arrondi à l'ampère (sans aucune décimale)

Avec 133 VA, logiquement le compteur doit transmettre 0 A sur chaque phase (à vérifier via le menu Messages).
Dans ce cas, nous n'avons donc aucun moyen de savoir quelle phase consomme. 
D'où la répartition arbitraire entre les 3 phases de la puissance globale annoncée.

reguiny, 2024-05-15:
Merci, je vais changer le compteur au standard.

closed 2024-05-15 by reguiny