NicolasBernaerts/tasmota #34: HA : Template variable error: 'dict object' has no attribute 'METER' when rendering

sj29fr, 2024-05-17:
Bonjour,
j'ai remarqué que HA n'était pas content de ce qu'il reçoit.
J'ai bien configuré MQTT et Teleinfo :
[image]

[image]

Comment corriger ce problème pour éviter ces logs inutiles ?

`
17/05/2024
20:43:44
2024-05-17 20:43:44.075 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.YDAY}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.076 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.TDAY}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.076 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'CONTRACT' when rendering '{{value_json.CONTRACT.name}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.076 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.U}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.077 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.I}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.077 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.P}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.077 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.W}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.079 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.C}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.079 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.YDAY}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.079 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.TDAY}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.573 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'CONTRACT' when rendering '{{value_json.CONTRACT.name}}'
17/05/2024
20:43:44
2024-05-17 20:43:44.574 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute 'CONTRACT' when rendering '{{value_json.CONTRACT.name}}'
`

NicolasBernaerts, 2024-05-20:
Effectivement, le topic SENSOR ne contient pas toujours toutes les données. Même si ce n'est pas gênant, je vais regarder comment publier une version de SENSOR contenant toutes les données de base afin d'éviter ce problème.

sj29fr, 2024-05-20:
En effet, ça ce gène pas au fonctionnement mais ça pollue les logs de HA.
Pour l'instant, je suis en mode +/-5% pour réduire la quantité de logs.

NicolasBernaerts, 2024-05-22:
Hello, 
J'ai préparé une nouvelle version avec trame SENSOR unique qui devrait régler le problème côté Home Assistant ...
Quelle version utilisez vous ?
Je publierai une version de test en 14.4.

NicolasBernaerts, 2024-05-22:
Quel type d'esp utilisez vous ?

sj29fr, 2024-05-22:
esp32 mais je pense migrer plus tard sur esp32-c3 ou esp32-s3

NicolasBernaerts, 2024-05-22:
Je viens de publier une v14.4 pour ESP32 OTA qui devrait régler le problème des logs Home Assistant.
En complément cette version minimise grandement les erreurs de réception.
Si tout est ok, je publierai les autres versions.
Merci pour le retour.

jsapede, 2024-05-23:
[image]

j'ai la meme chose sur denky, j'attendais d'avoir vérifié que ça venait pas d'anciens topics persistents mais j'ai effacé les db de mqtt et fait du propre dans les topics et ça réapparait.

quand on desactive une des entités qui pose pb la partie correspondante du message disparait mais revient des qu'on la réactive

NicolasBernaerts, 2024-05-23:
La version publiée 14.4 est la version esp32 pas la version denky. Vous avez un denky ?

jsapede, 2024-05-23:
oui je suis en denky et version 14.3.2

sj29fr, 2024-05-23:
J'ai fait la mise à jour en 14.4 mais je ne vois pas d'amélioration :
[image]

sj29fr, 2024-05-23:
Non, j'ai zappé ce paramètre.
Je vais réessayer sans les données brutes

sj29fr, 2024-05-23:
J'ai modifié la config ainsi et HA ne remonte pas d'erreur dans le journal
[image]

NicolasBernaerts, 2024-05-23:
> oui je suis en denky et version 14.3.2

Je vais publier une version Denky. Tenez moi au courant.

NicolasBernaerts, 2024-05-23:
> J'ai modifié la config ainsi et HA ne remonte plus d'erreur dans le journal ![image](https://private-user-images.githubusercontent.com/79538834/333208139-4726c98d-f731-4530-8a18-7d58ce9ffd6d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0OTIxNTksIm5iZiI6MTcxNjQ5MTg1OSwicGF0aCI6Ii83OTUzODgzNC8zMzMyMDgxMzktNDcyNmM5OGQtZjczMS00NTMwLThhMTgtN2Q1OGNlOWZmZDZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDE5MTczOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhNzg5MzEyNTUzODRlMzRhMDQwM2Q0MTEyYTEyZjg4YTViMGNhOTA2ODA3MmM5MjVlYzRhM2FiMjVjOTU3YzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.O9D6hcPqvDoMhch7LiE9Zd2x8yghyaoLIhepd25OGyo)

Très bonne nouvelle !

sj29fr, 2024-05-23:
> > J'ai modifié la config ainsi et HA ne remonte plus d'erreur dans le journal
> 
> Très bonne nouvelle !

Merci pour votre support !

jsapede, 2024-05-24:
bon alors ... mise a jour 14.4 OK
settings preconisés ok 
[image]

par contre je n'ai plus les infos de l'etat du tempo 
+ les noms des entités des compteurs tempo HP/HC bleu/blan/rouge ont changé ce qui m'a obligé a refaire mes totalisateurs :
[image]

Quand je regarde dans les topics mqtt je vois la définitio ndes sensors qui correspond a ce que j'ai d'affiché et donc sans les etats TEMPO : 
[image]

hors dans les trames de la console je les vois bien comme etant diffusées : 
[image]

j'ai Setoption19 à zero : 
[image]

si dans les reglages du denky je mets "toutes les infos TIC" , je vois bien que tout est diffusé dans le topic téléinfo : 
[image]

[image]

j'avoue que je suis un peu perdu là ...

NicolasBernaerts, 2024-05-24:
Pour avoir la couleur du moment, il faut logiquement activer l'option de calendrier.

NicolasBernaerts, 2024-05-24:
Réponse : les deux 😀
Cela lit les données STGE envoyées par le compteur. Et si RTE est activé, elles sont complétées par les données publiées par RTE. 
Les mauvaises nouvelles sont prioritaires. Et RTE est très souvent en avance sur les mauvaises nouvelles : on le sait plusieurs heures avant que linky l'annonce 😔

closed 2024-05-25 by NicolasBernaerts