NicolasBernaerts/tasmota #35: Reboot en boucle

Lutty76, 2024-06-24:
Bonjour,

Mon esp 8266 reboot en boucle dès que je le branche à mon compteur. 

Avant, j'utilisais ce [firmware](https://github.com/Jason2866/Tasmota-specials/tree/firmware/firmware/tasmota/other) sans souci (en version 12 quelque chose).

Donc mes branchements sont corrects.

J'utilise ce [circuit](https://github.com/hallard/WeMos-TIC) comment interface entre mon ESP8266 et mon compteur Linky.

Je suis en monophasé sans contrat particulier.
J'ai des panneaux solaires qui parfois produisent plus que je consomme (courant injecté dans les réseaux, mais sans revente).

Voici un exemple d'info remonté par mon ancien firmware :

````

NicolasBernaerts, 2024-06-25:
Bonjour,
La dernière version publiée sur Github fonctionne parfaitement avec un ESP32 mais pose problème sur un ESP8266.
Je pense avoir trouvé la source du problème (différence de comportement sur PSTR () entre ESP32 et ESP8266.
Je devrais pouvoir publier un nouveau binaire avec le correctif d'ici 1 ou 2 jours.
Quelle version de ROM utilisez-vous (1Mb, 4Mb ou 16Mb) ?

Lutty76, 2024-06-26:
Bonjour,

J'ai essayé la 1M et la 4M, normalement la board est compatible 4M (WeMos D1 Mini)

NicolasBernaerts, 2024-06-27:
En fait j'ai fait pas mal d'évolutions alors que mon ESP de test est un ESP32. En vérifiant, certaines évolutions sont trop lourdes pour un ESP8266 en terme de RAM.
Je vais faire une version light compatible ESP8266.

NicolasBernaerts, 2024-06-28:
Je viens de publier une version ESP8266 qui devrait corriger le problème.
Est-ce ok ?

Lutty76, 2024-07-03:
Désolé je n'ai pas pris le temps de regarder ça (et je n'ai pas les notif de réponse sur l'issue), je regarde ça dans la soirée, Merci !

NicolasBernaerts, 2024-07-04:
Toutes les valeurs sont à zéro, la page Configuration / TeleInfo nécessite sans doute une verification.

Lutty76, 2024-07-04:
Pourtant je réutilise celle de mon firmware d'avant basé sur tasmota qui fonctionne correctement, je vais étudier ça quand même

Lutty76, 2024-07-12:
Je suis passé sur un ESP32 pour ne plus etre limité niveau performance. Je n'ai aps approfondi pour l'ESP8266, mais ça a fonctionné directement avec l'ESP32. J'espère que c'était juste un soucis de configuration.

NicolasBernaerts, 2024-07-12:
Ticket fermé faute d'éléments.

closed 2024-07-12 by NicolasBernaerts