NicolasBernaerts/tasmota #36: Teleinfo - Pb remonté vers Domoticz

Vincent835, 2024-07-01:
Bonjour Nicolas,

J’utilise Teleinfo 13.3.10 associé à Domoticz 2022.1. J’ai aussi utilisé les versions précédentes de votre Teleinfo. Jusqu'à récemment cela marchait très bien.
D’un coup,  sur Domoticz, au lieu d’avoir le suivi de la puissance en continu,  il n’y a plus que des 0 ou pics.
J’ai redémarré Tasmota et Domoticz,  cela n’a pas réglé le problème. 
Je ne comprends pas ce qu'il s’est passé. 
Auriez-vous une idée ? 
Je vous remercie 

Vincent 

Ci-dessous un extrait du log en console, ainsi que des captures de Domoticz et Teleinfo. 
> 21:02:37.641 MQT: domoticz/in = {"idx":205,"nvalue":0,"svalue":"0.0;4290786048.0;0.0;65536.0;892;0","Battery":100,"RSSI":7}
> 21:02:37.647 MQT: compteur/tele/SENSOR = {"Time":"2024-07-01T21:02:37","ENERGY":{"TotalStartTime":"2022-06-17T01:03:30","Total":12.475,"Yesterday":14.467,"Today":11.958,"Period":[4,1,2],"Power":[473,175,244],"ApparentPower":[714,265,369],"ReactivePower":[535,199,277],"Factor":[0.66,0.66,0.66],"Voltage":[228,233,233],"Current":[3.1,1.1,1.6]}}
> 21:02:38.628 MQT: compteur/tele/SENSOR = {"Time":"2024-07-01T21:02:38","ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""}}
> 21:03:07.625 MQT: compteur/tele/STATE = {"Time":"2024-07-01T21:03:07","Uptime":"8T21:22:00","UptimeSec":768120,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","BSSId":"xx:xx:xx:xx","Channel":11,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}

[image]
[image]

[image]

NicolasBernaerts, 2024-07-01:
Bonjour Vincent,
Il faudrait fournir plus de renseignements :
- version du firmware
- modèle de l'ESP
- enregistrements des trames MQTT Domoticz
Sans ces éléments, il est difficile de diagnostiquer.
D'autant plus que je n'ai pas de contrat Tempo.

NicolasBernaerts, 2024-07-01:
En rejouant un enregistrement de contrat Tempo, un test sur la dernière version 14.7 (non encore publiée, mais Domoticz n'a pas évolué depuis plusieurs versions) semble fonctionner correctement :

[image]

[image]

Quel est exactement le problème ?

Vincent835, 2024-07-02:
Bonjour Nicolas, 

Voici des complètements d'info :
Version du programme | 13.2.0(tasmota)
2024-02-11T13:33:53
2_7_5/2.2.2-dev(38a443e)
Teleinfo 13.3.10
ESP8285N08

Pour les trames MQTT, il y en a une dans mon message initial.
Je vais essayer d'en récupérer sur une durée plus longue.

Je ne sais pas comment détailler plus ce que j'ai déjà indiqué dans mon premier message.
Ça a marché, ca ne marche plus sans aucunes modifications de ma part.
J'espèrai que tu aies une piste pour m'aider à trouver l'origine du problème.

NicolasBernaerts, 2024-07-02:
La dernière version publiée pour ESP8266 est la v14.6.
Sachant qu'il y a eu énormément d'évolutions entre votre version 13.3 et la dernière, merci de tester la dernière version publiée. Tenez moi au courant.

Vincent835, 2024-07-02:
J’ai finalement fait un reset 6, rechargé la config, et ça marche à nouveau. 
Merci de m’avoir répondu.
On ne saura pas quel était le problème.

closed 2024-07-02 by Vincent835

NicolasBernaerts, 2024-07-02:
J'ai publié une v14.7 qui intègre de nouveau Domoticz, Hass et Homie pour les ESP8266.