NicolasBernaerts/tasmota #38: Linky Triphasée PH2=PH3

CTPA, 2024-09-08:
Bonjour et merci pour le travail Nicolas,

J'ai installé sur une carte Olimex ESP32-POE, le firmware "tasmota32-teleinfo.bin" en version 14.8 Beta.
J'ai un compteur Linky 3 phases en mode historique (demande pour le mode standard faite hier). 

Petit bug report : la Ph2 et Ph3 sont identiques en U / I / W / VA

[image]

Cordialement

NicolasBernaerts, 2024-09-08:
Hello,
En mode historique le courant est publié en A sans aucune décimale. Cela signifie qu'il n'y a aucune précision.
Le problème vient sans doute de là.
Regardez dans le menu "Message" pour voir si les courants des phases 2 et 3 ne sont pas à 1A.
Cela expliquerait la similitude entre les 2 phases.
En passant en mode standard, la précision par phase n'est pas du tout la même.

CTPA, 2024-09-09:
Bonjour, effectivement en regardant de plus prêt cela n'arrive que quand les valeurs sont basse.
Je vais attendre sagement la TIC en mode standard :-)

Merci pour la réponse

closed 2024-09-09 by CTPA

NicolasBernaerts, 2024-09-09:
Avant que vous ne passiez en mode standard je serai très intéressé si vous pouviez enregistrer 5 ou 10 mn de flux teleinfo. En effet je n'ai aucun jeu de données triphasé mode historique pour mes tests de non regression. Merci d'avance.

CTPA, 2024-09-09:
Avec plaisir,
Pouvez vous me préciser le format/méthode.
Pensez vous au messages dans la console, avec le format suivant : 

22:56:58.004 MQT: Compteur_Maison_Linky/tele/TIC = {"Time":"2024-09-09T22:56:57","TIC":{"ADCO":"031876067693","OPTARIF":"BBR(","ISOUSC":"30","BBRHCJB":"050664130","BBRHPJB":"044197790","BBRHCJW":"002332363","BBRHPJW":"000978604","BBRHCJR":"000632297","BBRHPJR":"000317331","PTEC":"HCJB","DEMAIN":"----","IINST1":"013","IINST2":"014","IINST3":"011","IMAX1":"060","IMAX2":"060","IMAX3":"060","PMAX":"20947","PAPP":"08940","HHPHC":"A","MOTDETAT":"000000","PPOT":"00"}}

22:57:00.007 MQT: Compteur_Maison_Linky/tele/SENSOR = {"Time":"2024-09-09T22:56:59","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":230,"I1":13.28,"P1":3055,"W1":3018,"C1":0.98,"U2":230,"I2":14.30,"P2":3290,"W2":3250,"C2":0.98,"U3":230,"I3":11.23,"P3":2585,"W3":2553,"C3":0.98,"U":230,"I":38.82,"P":8930,"W":8821,"C":0.98,"YDAY":42602,"TDAY":25335},"CONTRACT":{"serial":31876067693,"name":"Tempo","period":"Creuses Bleu","color":"Bleu","hour":"Creuse","CONSO":99122522,"HCJB":50664137,"HPJB":44197790,"HCJW":2332363,"HPJW":978604,"HCJR":632297,"HPJR":317331},"CAL":{"level":1,"hp":0},"RELAY":{"R1":0,"R2":0,"R3":0,"R4":0,"R5":0,"R6":0,"R7":0,"R8":0},"ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""}}

CTPA, 2024-09-09:
Ce n'est pas le cas dans le .txt en PJ mais comme vous pouvez le voir ci dessous, quand je branche 1 ou 2 VE  j'arrive a avoir un cos phi ~=0, ce qui donne du coup des puissances nulles :-)

[image]

[image]

[20240909_10MinFluxTIC.txt](https://github.com/user-attachments/files/16936688/20240909_10MinFluxTIC.txt)

NicolasBernaerts, 2024-09-09:
C'est étonnant car dans les logs le cosphi est toujours supérieur à 0.75.
Pour l'enregistrement, il faudrait démarrer le serveur TCP et enregistrer les flux directement depuis le port TCP ouvert. La procédure est décrite dans la section concernant le serveur TCP.
Avec cet enregistrement je pourrais essayer de reproduire le problème de cosphi.
Merci

CTPA, 2024-09-11:
Les logs transmis n'ont pas été fait en même temps que les graphs. C'est des impressions écran que j'ai fait plus tôt.

Je n'ai pas de PC sous linux, j'ai tenté l'opération avec nmap sous windows, mais je dois mal m'y prendre car je n'y arrive pas.
Avez vous une solution avec un PC windows?

NicolasBernaerts, 2024-09-11:
Vous devriez pouvoir le faire avec putty
https://documentation.help/PuTTY/using-rawprot.html
En activant le logging, le flux devrait être enregistré en live.
A tester ...

CTPA, 2024-09-12:
Pas de soucis pour le tcp_start avec le # de port.
Dans Putty, j'ai mis l'IP, le port et la connexion en RAW.
Et essayé toutes les différentes options de loggin session, mais cela ne semble pas fonctionner correctement.

[image]

localhost61, 2024-09-19:
Essayez plutôt avec  [Curl pour Windows](https://curl.se/windows/).
-> [mon post](https://community.ch2i.eu/post/5217) sur le site de Charles Hallard

NicolasBernaerts, 2024-09-19:
Merci pour le tip !
Étant sous linux depuis plus de 10 ans, je deviens un handicapé de windows :-(

CTPA, 2024-09-19:
Merci.
Malheureusement, je n'ais pas eu plus de chance...

cmd => curl ip:port
>curl: (1) Received HTTP/0.9 when not allowed

cmd => curl --http0.9 ip:port
> Warning: Binary output can mess up your terminal. Use "--output -" to tell curl to output it to your terminal anyway,
Warning: or consider "--output <FILE>" to save to a file.

cmd => curl --http0.9 --output toto.txt ip:port
> $  (    !`@*H	 !!~ EC(4!  8%	 
>   )  = $  	  -!R     4 PA$% $! $!$!    % 2	  (    -$    
>   !@ P@H !!~ E$
>  
>  
>  
>   @   5 		 ; 1 (- @@  t @)  @ @
>   @!%   % ! %$ (    -$      !@ P@H !!~ E
>  ( 
>   )    - !	  =!Q   5 4$ (P@$%$`$!$$!$)    % 2	  (    -$      !@ P@H !!~ E
>   ( 
>   % $    -  !P 9   5  P)@!     	%!! $ % ) %$ (   -$  (4  
>   !1(@=@pH !!~ E$A! 
> (
>    )  3      
> 1 % $  )  )-$  ! P)P  !!~ E4 
>  
>  
>    (@
>    % !!$
>  % A%-"A  P @) @	 @  @!%@$ % ! %$    -$  @  % !`RPH	 !!~ EC % 0 '$  9% A
>   e=  
>    -!! 
> !!0$ A %!2$  1P @  @+ @+  @!   %$!( !%   $!   !      P)P5 !!~ E$ 
>  '(! 
>  
>   e=     ,	  &4   AP @) @ @)  @!  % %$ % $   @  	,5 I   !  !$` E4 
>  
> (8 
>  -  a  
>    -!)	!! $ 	a$%2    T @  @+ 
> @
>   @!%  %$!  !     (@ $    !1@=@pH !!~ E +  
>    
>  -  @= $  	  	  $ @- @@ 0 P @	  @ @  @!$ % ! %$ (  ( -$      !@ P@H !!~ E 
>  (! 
>  
>   e=  
>    -!!
> !!0$ 
> A %!2$  ET @  @) @  @!%   % ) !$ (   -$  @ 
>  !@ P@H !!~ E$A! 
> (
>  -  a  
>    (  	(%P ,@4   	  `$!$ $!$)  @ % 	     -$   @ 
>  !@ P@H !!~ E$ 
>  ()$, 9 
>   e=  
>    	!  	(%Q %,@$   P @) @ @  @!%   % ) !$    -$   @   !`P@H )!~ E$%! 
>    
> 
>  -  a  0  	 (%Q%,B4 (  P @) @	 @
>   @!%   % ! %$    -$   @  % !`P@H )~ M$<!  H8
>  -  a  
>    
> !! 	  @Q  %%"  P @) @	 @  @!%  	 !$! P !4 $! 1 ($  $  % !(`H	 !!~ EC(4! ' (-	$  9- 
>   m       
> 	  -!P   5   At @  @ @)  @!$    !!  1  $    !(@XPH	 !!~ MC 0! ' 8-	$  9- 
>   @    
>   66 I) @aAa)Q4 J $  $!$!$)  4 % 1&	   $ ),  %
> (% !(@5XH !!~ MB<! '$  9 =  a  
> H !!~ EC(4! ' 
>   % $   = 
>   	!q  9 A4  () !  1       @!%   %$!( !   % (  $    !(@#H	 !!~ MC  0 '	P 
>  
>   @ 5 $  -  	  @)) @@ %-  `$!$$! $)    % 2	  (   -$   @   !@P@ !!~ E$	! (! 
>  
>   e=  
>    -!! (4!I$H(@ 4  ( ET @  @) 
> @  @!%   % ) !$    -$   @   !@P@H )!~ E$ 
>  (!$% 9 
>   @  3!  9P  @r  aAQ)B!  !$ $!$!$)    % 2	     -$   @   !`R@H )!~ E$ 
>  (!$% 9 
>   @  3!
>   9P  @p  aAQ	PP) ) 1   	!!!%   % ! %$    -$   @  % !`P@H )~ E4 
>  (!$%	 
>   @  3!  9P  @p  aAQ)PQ) )    	!!!%   % ! %$   ( - !$ % % !`PPH !!~ E$%	 !  
>  
>   a  
> H !!~ EC(4! ' (%	$! 9=   @  ;!	  =!P    5 P) @!   ! %    @! $ ! ! %$   ( ),  %  % !(`H	 !!~ E< 
>   
>    
>   ;   !
>  	 	$0$ A$!2   At @)  @ @+  @! $ P !    ( $  $ %4  p)P  !!~ E, 
>   
>      ;  
>   	 	$
> E$% !  
>   5 $    
>   	!y  @q   	  ) !     	%!! $ ! ! % 	   ( -$  F %
>  !`PPH	 !!~ MC  ! '$! 9%   @    
>   	!y  @q  4`4    $!$`$!$ $! $  % 2&    ( -$   % % !`PPH	 !!~ E  
> 0 '$! 9% 
>   @    
>   	!y  @q  04$)( ) !   1     @! $ ! ! % 	   ( -$  % % !`PPH	 !!~ E,
>   '(! 

localhost61, 2024-09-19:
@CTPA, on est bien d'accord :

- Que vous avez d'abord démarré le serveur sur votre carte Olimex avec :
>`tcp_start 8888`
- Que la commande a été acceptée
- Que sous Windows, dans une boîte DOS vous avez simplement tapé
 >`curl IP:Port`
 où `IP` est l'adresse IP de votre carte Olimex et `Port` est l'argument de la commande `tcp_start`, soit 8888 dans mon exemple ce qui donne par exemple: `curl 192.168.1.10:8888`

CTPA, 2024-09-20:
Oui précisément.

NicolasBernaerts, 2024-09-21:
Il semble que nc / netcat soit disponible sous windows 
https://nmap.org/download.html#windows

CTPA, 2024-09-24:
J'aurai vraiment essayé mais à mon retour de wkd le compteur été en mode standard. 

Par curiosité j'ai refait la manipulation et elle semble avoir fonctionné, peut être un bug du coup, voici 4-5 secondes de résultat : 

> CCASN	E240924223000	03720	@
> CCASN-1	E240924220000	00716	]
> UMOY1	E240924225000	237	5
> UMOY2	E240924225000	239	8
> UMOY3	E240924225000	241	2
> STGE	013A0000	:
> MSG1	PAS DE          MESSAGE         	<
> PRM	21198552806784	C
> RELAIS	001	C
> NTARF	01	N
> NJOURF	00	&
> NJOURF+1	00	B
> PJOURF+1	00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	.
> 
> ADSC	031876067693	E
> VTIC	02	J
> DATE	E240924225642		H
> NGTF	     TEMPO      	F
> LTARF	    HC  BLEU    	^
> EAST	099765794	G
> EASF01	051134735	?
> EASF02	044370464	C
> EASF03	002332363	:
> EASF04	000978604	G
> EASF05	000632297	C
> EASF06	000317331	9
> EASF07	000000000	(
> EASF08	000000000	)
> EASF09	000000000	*
> EASF10	000000000	"
> EASD01	038701095	A
> EASD02	034316976	H
> EASD03	015398300	?
> EASD04	011349423	>
> IRMS1	007	5
> IRMS2	005	4
> IRMS3	005	5
> URMS1	239	H
> URMS2	241	B
> URMS3	242	D
> PREF	18	H
> PCOUP	18	"
> SINSTS	03834	X
> SINSTS1	01557	I
> SINSTS2	01185	G
> SINSTS3	01092	E
> SMAXSN	E240924022201	05125	3
> SMAXSN1	E240924082102	01910	(
> SMAXSN2	E240924185705	02772	=
> SMAXSN3	E240924022200	03077	)
> SMAXSN-1	E240923012353	11673	\
> SMAXSN1-1	E240923012345	03571	L
> SMAXSN2-1	E240923012349	04544	R
> SMAXSN3-1	E240923014754	04651	T
> CCASN	E240924223000	03720	@
> CCASN-1	E240924220000	00716	]
> UMOY1	E240924225000	237	5
> UMOY2	E240924225000	239	8
> UMOY3	E240924225000	241	2
> STGE	013A0000	:
> MSG1	PAS DE          MESSAGE         	<
> PRM	21198552806784	C
> RELAIS	001	C
> NTARF	01	N
> NJOURF	00	&
> NJOURF+1	00	B
[... 233 lines cut ...]
> EASF10	000000000	"
> EASD01	038701101	5
> EASD02	034316976	H
> EASD03	015398300	?
> EASD04	011349423	>
> IRMS1	007	5
> IRMS2	005	4
> IRMS3	005	5
> URMS1	239	H
> URMS2	241	B
> URMS3	242	D
> PREF	18	H
> PCOUP	18	"
> SINSTS	03849	^
> SINSTS1	01572	F
> SINSTS2	01185	G
> SINSTS3	01092	E
> SMAXSN	E240924022201	05125	3
> SMAXSN1	E240924082102	01910	(
> SMAXSN2	E240924185705	02772	=
> SMAXSN3	E240924022200	03077	)
> SMAXSN-1	E240923012353	11673	\
> SMAXSN1-1	E240923012345	03571	L
> SMAXSN2-1	E240923012349	04544	R
>

CTPA, 2024-09-24:
curl --http0.9 --output toto.txt ip:port