NicolasBernaerts/tasmota #39: [Teleinfo] Questions ESP8266

ysard, 2024-09-13:
Bonjour, Merci pour cet énorme travail !

Je développe actuellement ma propre carte TeleInfo basée sur ESP8266 4M + Super condensateur.

J'ai quelques questions auxquelles vous pourrez répondre plus facilement que moi 
grâce à votre maitrise de ce projet (Je ne suis pas du tout familier avec Tasmota).

- Deepsleep:
    Mon ESP a une autonomie de plusieurs minutes sur son condensateur, mais le mode deepsleep reste nécessaire.
    
    J'aimerais savoir s'il serait possible d'intégrer une option deepsleep
    similaire à ce qu'on trouve sur le Winky sur cette plateforme.

    Le flag USE_WINKY active une multitude de choses dont je n'ai pas l'utilité sur ma carte 
    (mesures de tensions, etc.)

- Horloge:
    Toujours dans un but d'autonomie, est-il légitime de vouloir faire tourner ce code avec une horloge de 80MHz au lieu de 160MHz ou cela pose des problèmes qui m'échappent (en dehors d'une négociation SSL plus longue par ex) ?
    
- Courbes/historique:
    Vu que mon projet peut être couplé à un serveur comme HomeAssistant + MQTT, 
    est-il possible de désactiver 
    totalement les courbes et l'historique afin de ne pas user la flash en vain ?
    
    Compiler sans -DUSE_TELEINFO_CURVE -DUSE_TELEINFO_HISTO suffirait-il ?

D'avance merci !

NicolasBernaerts, 2024-09-14:
Bonjour,

Concernant le deepsleep, tout est possibe.
Vous pouvez activer le fonctionnement standard de Tasmota via la commande
deepsleeptime xxx

Avec une horloge à 80MHz cela devrait fonctionner, mais le mode standard demande beaucoup plus que le mode historique.
Le jeu n'en vaut sans doute pas la chandelle.

Pour les courbes, elles sont désactivable via les 2 options de compilation. C'est leur but.

ysard, 2024-09-14:
Merci pour votre réponse cela me fait gagner beaucoup de temps.

J'ai testé la fonctionnalité deepsleeptime sur 2 cartes :

- la mienne (broche RST reliée au GPIO16, le tout avec une résistance de pullup de 10K comme recommandé partout)
- une copie de wemos D1 mini d'aliexpress banale :
    https://fr.aliexpress.com/item/32968761711.html
        
J'obtiens 2 comportements sensiblement similaires dont voici les observations par ordre chronologique :
    
- les 2 cartes rentrent en deepsleep durant le temps configuré
- les 2 cartes se réveillent néanmoins dans un mode "intermédiaire"
non fonctionnel: aucune connectivité réseau, 30mA de conso au lieu des 70mA
observés en fonctionnement normal.
- ma carte requiert 1 connexion GND-RST pour se réveiller correctement
- le wemos requiert 2 pressions sur le bouton RST pour se réveiller correctement
- les 2 cartes sont observées sur le réseau 8-12s (le temps d'une requête MQTT)
puis repassent en deepsleep. Ce n'est donc pas un reboot de type PowerOn
qui passerait en deepsleep au bout de ~65s (countdown affiché dans la console tasmota).

    
Ces observations sont très reproductibles. Cela vous dit-il quelquechose ?

NicolasBernaerts, 2024-09-15:
Bonjour,
Je ferme ce thread car il est plutôt destiné à un forum de discussion.

closed 2024-09-15 by NicolasBernaerts

ysard, 2024-09-15:
À vrai dire mes ESP8266 ont un problème hardware bien référencé ici : https://github.com/esp8266/Arduino/issues/6007
Ce problème est solutionné chez certains, ces clones défaillants sont encore en circulation 5 ans après.
Bref, Tasmota n'est pas en cause, et j'ai reproduit la comportement sur MicroPython également.

Merci à vous :p