NicolasBernaerts/tasmota #40: données manquantes maj 14.ç-beta9

jsapede, 2024-09-18:
Bonjour,

j'ai perdu un peu le fil du sujet mais apres mise à jour a l'instant je me retrouveavec une interface dépourvue de toute mesure "instantanée" 

[image]

Les données ne sont egalement plus transmises au homeasisstant

j'ai essayé de trifouiller un peu dans les reglages mais je ne trouve pas comment les reafficher ni les faire transiter vers le mqtt.

ai-je loupé une étape ?

jsapede, 2024-09-18:
j'ai ce message dans le log du homeassistant : 

Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.C}}'
Template variable error: 'dict object' has no attribute 'METER' when rendering '{{value_json.METER.YDAY}}'
Template variable warning: 'dict object' has no attribute 'tomorrow' when rendering '{{value_json.CONTRACT.tomorrow}}'
Template variable warning: 'dict object' has no attribute 'today' when rendering '{{value_json.CONTRACT.today}}'
Template variable warning: 'dict object' has no attribute 'C' when rendering '{{value_json.METER.C}}'

visiblement un certain nombre de topics ont disparu

NicolasBernaerts, 2024-09-19:
Bonjour, 
Dans Configure Teleinfo, est ce que : 
-  le mode standard est bien sélectionné ?
- les données Consommation et production sont cochées ?

jsapede, 2024-09-19:
Bonjour,

a priori oui 

[image]

j'ai supprimé et reinstallé le Denky sur HA mais j'ai toujours les données tempo manquantes : 

[image]

on le constate sur les remontéers mqtt : 

[image]

De plus les entites des index des compteurs ont changé dans MQTT ce qui a obligé a refaire tous les compteurs journaliers : 

[image]

jsapede, 2024-09-19:
par contre je viens de constater ce matin que la maj a fait passer le tempo_enable à 0 alors qu'il etait à 1

[image]

NicolasBernaerts, 2024-09-19:
Vous pouvez décocher "calendrier" dans les options. C'est le calendrier destiné aux radiateurs fil pilote.
Vous pouvez également décocher Énergie Tasmota car HA n'utilise pas ces données.
Il est dangereux de publier a chaque Télémétrie car l'ESP a du mal à suivre la publication d'autant de données toutes les secondes. Je devrais enlever cette option qui est plutôt du debug. Cochez plutôt évolution de +-100w

Pour savoir ce qui est vraiment publié regardez la console et vérifiez que les sections METER et CONTRACT soient bien publiées dans SENSOR.

jsapede, 2024-09-19:
j'ai corrigé suivant vos préconisatoins, et voila ce qui est publié dans SENSOR : 

{"Time":"2024-09-19T09:02:39""ESP32":{"Temperature":56.7},"METER":{"PH":1,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":238,"I1":2.00,"P1":359,"W1":288,"C1":0.80,"U":238,"I":2.00,"P":359,"W":288,"C":0.80,"YDAY":3000,"TDAY":10827},"CONTRACT":{"serial":841775199524,"name":"Tempo","period":"Pleines Bleu","color":"Bleu","hour":"Pleine","CONSO":39820484,"HC BLEU":19385293,"HP BLEU":19440106,"HC BLANC":384447,"HP BLANC":285578,"HC ROUGE":197089,"HP ROUGE":127971},"ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""},"TempUnit":"C"}

ce qui est etonnant ausis c'est que la visu dans mqttexplorer n'est pas du tout sous la meme forme que STATE : 

[image]

alors que STATE 

[image]

ou RTE : 

[image]

NicolasBernaerts, 2024-09-19:
Ok, il y a un bug sur la publication de la température ESP32, il manque une virgule. Dans Configure Options, decochez Température ESP32. Le TOPIC sensor devrait être mieux.

jsapede, 2024-09-19:
c'est effectivement mieux merci : 

[image]

NicolasBernaerts, 2024-09-19:
J'ai trouvé le bug.
Il sera corrigé dans la prochaine version.
Merci !

NicolasBernaerts, 2024-09-24:
J'ai publié un binaire pour le DenkyD4 en v14.8beta16 qui devrait corriger le problème.

jsapede, 2024-09-26:
Bonsoir 
Mise a jour beta16 mais je n'arrive toujours pas a avoir les couleurs tempo dans homeassistant 

On voit bien les messages mqtt

[image]

Mais rien ne remonte dans homeassistant 

[image]

La config est calée ainsi 

[image]

localhost61, 2024-09-26:
La TIC ne permet pas de connaître le tarif qui sera appliqué à J+1. Pour cela il faut utiliser l'API de RTE qui nécessite une souscription (gratuite). Avez-vous fait la démarche?

jsapede, 2024-09-26:
Oui j'ai bien l'API rte
Le PB c'est sur les infos dont diffusées en mqtt mais ne sont visiblement pas auto découvertes

localhost61, 2024-09-26:
La TIC ne permet pas de connaître le tarif qui sera appliqué à J+1. Pour cela il faut utiliser l'API de RTE qui nécessite une souscription (gratuite). Avez-vous fait la démarche? -> c'est indiqué i[ci](https://github.com/NicolasBernaerts/tasmota/tree/master/teleinfo#calendriers-rte--tempo-pointe--ecowatt)

localhost61, 2024-09-26:
Je ne sais pas si Nicolas a limité le nombre d'appels à l'API, il y a un quota à respecter sous peine de bannissement. Pour ma part je ne l'interroge qu'à 10h20 et toutes les 10mn jusqu'à 11h en cas d'échec. Si les requêtes sont lancées aussi au boot, le quota risque d'être atteint en phase de test.

jsapede, 2024-09-26:
Je ne pense pas que ce soit un pb d'api, j'ai bien des choses qui sont diffusés en mqtt sous le topic RTE, je les vois avec mqtt explorer. J'ai plutôt l'impression que l'auto découverte homeassistant n'intègre pas les topics rte

NicolasBernaerts, 2024-09-27:
> La TIC ne permet pas de connaître le tarif qui sera appliqué à J+1. Pour cela il faut utiliser l'API de RTE qui nécessite une souscription (gratuite). Avez-vous fait la démarche?

Essayez de cocher "calendrier' dans les données émises.

NicolasBernaerts, 2024-09-27:
> Je ne sais pas si Nicolas a limité le nombre d'appels à l'API, il y a un quota à respecter sous peine de bannissement. Pour ma part je ne l'interroge qu'à 10h20 et toutes les 10mn jusqu'à 11h en cas [...]

Le firmware interroge les API RTE toutes les 3 heures (de mémoire) pour éviter de dépasser le quota. En cas d'erreur par dépassement d equota un test est fait toutes les 15mn jusqu'à ce que les API répondent de nouveau

NicolasBernaerts, 2024-09-27:
> Je ne pense pas que ce soit un pb d'api, j'ai bien des choses qui sont diffusés en mqtt sous le topic RTE, je les vois avec mqtt explorer. J'ai plutôt l'impression que l'auto découverte homeassistant n'intègre pas les topics rte

Je confirme que le topic RTE n'est pas en auto découverte dans HA.
C'est le topic SENSOR qui est en auto découverte.
Les jours tempo étant liés au calendrier il faut cocher l'option 'calendrier' pour que la section CAL soit publiée en complément dans SENSOR. Et l'auto découverte HA doit l'afficher.

jsapede, 2024-09-27:
Ça redonne des éléments mais les topics aujourd'hui et demain ne remontent pas 

[image]

NicolasBernaerts, 2024-09-27:
N'ayant pas de contrat Tempo, je n'exploite pas ces données. Je vais faire le test avec un jeu de données Tempo pour voir s'il y a un problème.

jsapede, 2024-09-27:
On voit dans le topic RTE que l'index couleur du jour et du lendemain est bien prévu mais ça ne remonte pas dans le topic mqtt sensor 
Ça marchait pourtant sur les versions plus anciennes

NicolasBernaerts, 2024-09-27:
J'ai réorganisé les données entre CONTRAT et CAL et je pense que côté intégration HA quelque chose n'a pas suivi.

NicolasBernaerts, 2024-09-28:
En fait la publication des données de la couleur du jour et du lendemain avait disparue suite à une réorganisation de code.
Elle est maintenant de nouveau disposnible sous CONTRACT/tday et CONTRACT/tmrw
L'intégration HASS pour ces données est à jour.
Je viens de publier le binaire v14.8beta17 pour ESP32, DenkyD4 et ESP32S3.
Dites moi si tout est rentré dans l'ordre.

jsapede, 2024-09-28:
c'est ok merci !

[image]

jsapede, 2024-09-28:
du coup y-a-til interet a continuer a diffuser les calendriers pour avoir Tempo ?

NicolasBernaerts, 2024-09-28:
Effectivement, les données étant dans CONTRACT, cela ne devrait pas être nécessaire.
Je viens de republier une beta17 avec laquelle ces données sont publiées sans besoin d'avoir coché "Calendrier".
Si vous faites la mise à jour, vous devriez avoir les données disponibles systématiquement.

NicolasBernaerts, 2024-09-30:
Tout est ok ?
Le ticket peut être clôturé ?

jsapede, 2024-09-30:
Ça a l'air ok de mon côté

closed 2024-09-30 by NicolasBernaerts