NicolasBernaerts/tasmota #41: Homeassistant dashboard energyg - Données de consommation "retournée" apparemment erronée

perefouras, 2024-09-25:
Bonjour,

J'utilise depuis une semaine sur un Denky votre firmware qui est venu remplacer avantageusement (!) l'original teleinfo.
Tout fonctionne correctement mais dans le dashboard Energie de HomeAssistant je n'arrive à expliquer le "total retourné" qui apparait chaque jour sur le graphe pour la période minuit/1 heure :

[image]

Je précise que j'ai un kit solaire plug&play en production mais en autoconsommation (pas de revente au réseau)
La timezone est bien configurée sur tasmota (GMT+2 en ce moment) et sur Homeassistant (Paris).
J'ai installé la beta16 qui ne change rien.

Auriez-vous une idée ?
Merci.

NicolasBernaerts, 2024-09-25:
Vous utilisez sans doute le total quotidien METER/TDAY au lieu du total general CONTRACT/CONSO.
Le total quotidien revenant à 0 à minuit, vous avez cette différence.
Utilisez CONTRACT/CONSO cela devrait être mieux.

perefouras, 2024-09-25:
En effet, merci.  J'étais allé l'explication un peut trop loin !

closed 2024-09-27 by perefouras