NicolasBernaerts/tasmota #42: Utilisation date et heure TIC

Vincent835, 2024-10-01:
Esp8266, TASMOTA Teleinfo 13.3.10

Bonjour Nicolas, 
Ayant une panne internet depuis quelques jours, je viens de me rendre compte que Tasmota, après un reboot, attend une mise à jour de l'heure pour démarrer la lecture des données TIC. Donc absence de relevé pendant plusieurs heures, le temps que je m'en rende compte et fasse une mise à l'heure manuelle. 
N'y aurait-il pas un moyen d'utiliser l'heure issue du compteur pour synchroniser l'heure de TASMOTA automatiquement ? 
Je te remercie 

Vincent

NicolasBernaerts, 2024-10-01:
Bonjour Vincent,
En fait j'ai mis cette sécurité car certaines publications nécessitaient un timestamp.
Je vais regarder, mais il doit être possible de publier même si la synchro RTC n'est pas faite.

Vincent835, 2024-10-01:
Bonjour Nicolas,
En fait, je me demandais si, à condition que les trames TIC soient lues, cela serait possible de faire la synchro dd l’heure avec une "rule” tasmota sans modifier le firmware.

Vincent835, 2024-10-01:
Bonjour Nicolas,
En fait, je me demandais si, à condition que les trames TIC soient lues, cela serait possible de faire la synchro dd l’heure avec une "rule” tasmota sans modifier le firmware.
Nota : Après vérification, la date et l’heure ne sont diffusées qu’en mode TIC STANDARD. Donc pas applicable à toutes les configurations.

NicolasBernaerts, 2024-10-02:
Je vais préparer une évolution pour mettre à jour l'heure RTC manuellement depuis la première trame reçue du compteur. Cela permettra d'avoir une heure validé en attendant une synchro RTC.
Cela fonctionnera pour les linky en mode standard, les compteurs PME/PMI et les compteurs émeraude. Seuls les linky en mode historique devront attendre la synchro RTC avant de pouvoir émettre.

NicolasBernaerts, 2024-10-12:
Je viens de publier une v14.9beta1 pour esp8266 qui se synchronise sur le linky puis sur le serveur NTP si accessible. Cela devrait régler le problème de publication sans accès internet.

Vincent835, 2024-10-13:
Bonjour Nicolas, 
La gestion des publications vers Domoticz est faite différemment sur Teleinfo v14 par rapport à la v13.
Du coup, je n’ai plus l'enregistrement de ma conso elec avec cette version beta.
Je suis revenu à la version 13, le temps de trouver le temps pour voir comment récupérer la conso elec v14 sur mon compteur Domoticz.
Je te tiendrai au courant. 

Petit hors sujet : en revenant sur la v13, je n’ai pas pu restaurer la sauvegarde. Erreur mémoire insuffisante ! L’ESP avec 1Mb de mémoire n’est plus suffisant pour cette version. J’imagine que de doit être pareil avec la 14.

NicolasBernaerts, 2024-10-13:
L'ESP8266 1Mb est super limite pour teleinfo en terme de ROM et RAM.
J'ai encore un compteur que sonoff basic R2. Avant de le flasher je reboote l'ESP pour libérer le maximum de RAM (utilisée par l'upload OTA).
En cas de problème, le plus simple est de passer par un flash Tasmota minimal en flash intermédiaire. Cela doit passer sans problème.

NicolasBernaerts, 2024-10-13:
> Bonjour Nicolas, La gestion des publications vers Domoticz est faite différemment sur Teleinfo v14 par rapport à la v13. Du coup, je n’ai plus l'enregistrement de ma conso elec avec cette version [...]

Pour Domoticz, je stocke peut être les index à un endroit différent en ROM. Le plus simple et est noter les index en v13 et de les reposrter dans la page de configuration Teleinfo v14. 
Le code n'a pas changé. J'ai juste ajouté des données supplémentaires et utilisé les variables Domoticz natives de Tasmota pour stocker les index.

Vincent835, 2024-10-13:
Pas de problème pour flasher. C'est pour récupérer la sauvegarde de la configuration que cela ne marche pas.

Je viens de refaire un essai.
Coté mise à l'heure automatique sans connexion internet sur la base de l'heure TIC, c'est parfait.
Merci beaucoup.
`00:00:01.001 WIF: Connexion à l'AP1 Freebox-XXXXXX Channel 6 BSSId xx:xx:xx:xx:xx en mode 11n comme compteur-0203...
00:00:01.603 TIC: Mode Standard
00:00:01.604 TIC: Contrat n° 0123456789
22:09:40.654 RTC: Date set from meter (2024/10/13 22:09:40)
22:09:40.753 WIF: Connecté
22:09:40.004 HTP: Serveur web actif sur compteur-0203 avec l'adresse IP 192.168.1.95
22:09:40.356 TIC: Nouveau contrat TEMPO, 6 périodes connues
22:09:40.357 TIC: Configuration saved
22:09:40.357 TIC: Contrat en kVA
22:09:40.358 TIC: Changement de période HC BLEU (index 1, level 1, hchp 0)
22:09:41.111 MQT: Tentative de connexion...
22:09:41.252 MQT: Connecté
22:09:41.255 MQT: compteur/tele/LWT = Online (persistant)
22:09:41.256 MQT: compteur/cmnd/POWER = 
22:09:41.263 MQT: compteur/tele/INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"14.1.0(tasmota)","FallbackTopic":"cmnd/DVES_DF20CB_fb/","GroupTopic":"tasmotas/cmnd/"}}
22:09:41.267 MQT: compteur/tele/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"compteur-0203","IPAddress":"192.168.1.95"}}
22:09:41.270 MQT: compteur/tele/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":15}}
22:09:41.273 MQT: compteur/stat/RESULT = {"POWER":"OFF"}
22:09:41.275 MQT: compteur/stat/POWER = OFF
22:09:45.244 MQT: compteur/tele/STATE = {"Time":"2024-10-13T22:09:45","Uptime":"0T00:00:08","UptimeSec":8,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Freebox-A496DF","BSSId":"14:0C:76:30:71:B7","Channel":6,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:09:45.255 MQT: compteur/tele/SENSOR = {"Time":"2024-10-13T22:09:45","METER":{"PH":3,"ISUB":20,"PSUB":4000,"PMAX":4000,"U1":235,"I1":2.15,"P1":506,"W1":430,"C1":0.85,"U2":235,"I2":1.26,"P2":298,"W2":253,"C2":0.85,"U3":234,"I3":1.71,"P3":401,"W3":340,"C3":0.85,"U":234,"I":5.13,"P":1205,"W":1023,"C":0.85,"YDAY":0,"TDAY":0},"CONTRACT":{"serial":31776013513,"name":"Tempo","period":"Creuses Bleu","color":"Bleu","hour":"Creuse","tday":"Bleu","tmrw":"Bleu","CONSO":53038918,"HC BLEU":38629708,"HP BLEU":11904545,"HC BLANC":913333,"HP BLANC":717807,"HC ROUGE":504577,"HP ROUGE":368948},"ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""}}
22:09:45.363 MQT: domoticz/in = {"idx":205,"nvalue":0,"svalue":"38629708;11904545;0;0;1023;0","Battery":100,"RSSI":8}
22:09:45.613 MQT: domoticz/in = {"idx":207,"nvalue":0,"svalue":"2.0;1.0;2.0","Battery":100,"RSSI":8}`

closed 2024-10-13 by Vincent835