NicolasBernaerts/tasmota #43: Compatibilité avec compteur divisionnaire

sinclairfr, 2024-10-04:
Bonjour Nicolas,

Tout d'abord bravo et merci pour votre excellent travail sur ce projet ! 
Je n'ai pas de linky, mais un compteur divisionnaire à l'ancienne de type "divisionnaire" avec télétransmission.
Exemple : [https://voltebox.com/collections/compteurs-electriques/products/zcd126](url)

Je me demandais s'il y avait une moyen de faire fonctionner l'extension avec ce type de protocole de données ? 

En vous remerciant par avance,
M.

NicolasBernaerts, 2024-10-04:
Bonsoir sinclairfr,
Parlez vous de ce compteur ?
https://www.polier.fr/_files/ugd/511b90_df1b5c3ef93d4da2b5bbd6ead161261e.pdf

sinclairfr, 2024-10-04:
Merci pour la réponse !
Oui en effet, tous les compteurs avec la téléinfo ancienne version.
Je me demandais si le protocole linky était le même.

NicolasBernaerts, 2024-10-04:
Hello,
Bonne nouvelle !
Ces compteurs utilisent le protocole TeleInfo historique qui devrait être parfaitement géré par mon firmware. 
Vous devez simplement configurer la vitesse à 1200 bps et tout devrait être reconnu.
Quel contrat avec vous ?

sinclairfr, 2024-10-05:
J'utilise ce module :
[image]
et la configuration suivante:
[image]
[image]
Mon contrat est un EDF Zen Online avec HP/HC.

sinclairfr, 2024-10-05:
Je suis en train de me demande si tout bêtement le pin-out n'est peut être pas le bon dans la configuration Wemos Teleinfo.

sinclairfr, 2024-10-05:
Problème réglé, j'avais le mauvais GPIO. Pour information, gpio 37 = teleinfo RX sur ce modèle, on ne sait jamais, cela peut aider quelqu'un d'autre.
Merci en tout cas !

NicolasBernaerts, 2024-10-05:
Bonjour Medhi,
Très bonne nouvelle.
Concernant les options cochées, je vous conseille de décocher :
- données temps réel
- données Teleinfo brutes
Elles ne sont d'aucune utilisée pour Home assistant.
Ayant un contrat historique, vous pouvez décocher les données "Relai virtuel".
L'option "A chaque message reçu" est dans la plupart des cas inutiles.
Si vous cochez "Tous les 100 VA", vous aurez le même résultat et votre ESP sera beaucoup moins stressé.
Pour Home Assistant, aucun besoin de "Energie Tasmota".

closed 2024-10-05 by NicolasBernaerts