NicolasBernaerts/tasmota #44: Tasmota Teleinfo v14.9béta1 - Domoticz - Total global conso.

Vincent835, 2024-10-13:
Bonjour Nicolas,

Merci beaucoup pour le travail que tu as fait et que tu continues de faire avec cette application Tasmota Téléinfo.
Je suis impressionné par ta réactivité pour proposer des correctifs ou améliorations.

Le changement de gestion des données envoyées vers Domoticz depuis la version v14 m'a perturbé.
Sur la v13, il n'y avait qu'un seul compteur sans distinction des différents tarifs. Maintenant, la version v14 propose des compteurs différents selon les périodes tarifaires et un compteur global.

J'ai donc choisi le compteur "Total global conso." pour faire la publication vers Domoticz. Mais cela ne marche pas car le message de la publication n'est pas sous le même format.

Par contre les compteurs par tranche horaire ont le bon format (celui qui était sur la v13 : P1 Smart Meter).
Publications Tasmota Teleinfo v14.9béta1 :

Compteur "Totaux conso. Bleu"
`22:24:18.324 MQT: domoticz/in = {"idx":205,"nvalue":0,"svalue":"38629798;11904545;0;0;341;0","Battery":100,"RSSI":8}`
Compteur "Total global conso".
`22:24:18.573 MQT: domoticz/in = {"idx":300,"nvalue":0,"svalue":"147;53039008","Battery":100,"RSSI":8}`

Est-ce une erreur de format pour le compteur "Total global conso" ou est-ce un format pour un dispositif Domoticz spécifique? Le cas échéant, pourrais-tu me préciser lequel?

Je trouve intéressant cette évolution avec la distinction des consommations par tranche tarifaire, cependant je regrette que la puissance instantanée ne soit pas aussi au niveau du compteur "Total global conso".

Merci d'avance pour ta réponse.

NicolasBernaerts, 2024-10-14:
Bonjour Vincent,
Tout d'abord merci pour les compliments 😀
Concernant le total, je l'ai déclaré comme une valeur de type puissance car il n'y a pas de distinction HP/HC car nous parlons d'un total général.
Mais effectivement, on pourrait le déclarer également comme smartmeter avec un seul total et la puissance instantanée associée.
N'utilisant pas moi même Domoticz, j'ai besoin de ce type de retours :-)

NicolasBernaerts, 2024-10-14:
Je viens d'uploader la v14.9beta2 qui a changé le total général de conso en P1SmartMeter.
Indique moi si tout est ok ?

Vincent835, 2024-10-14:
Bonsoir Nicolas,
Zut, je suis en retard. Je suis tjs sur la béta1
J'ai fait des essais en changeant le type de compteur dans Domoticz.
C'est bon, j'ai ce que je veux à une coquille près. Il ne remonte que la puissance de la phase 1 (triphasé).
[image]

Je vais essayé la béta2.

Vincent835, 2024-10-14:
Téléinfo v14.9beta2
Même problème de non cumul de la puissance des 3 phases.
Par contre pour le type de compteur cela correspond à celui qui était sur la v13 : P1 Smart Meter. Donc pour moi, avec cette beta2 le passage de teleinfov13 à v14.9beta2 est transparent sur Domoticz.
Reste "plus qu'à" corriger le calcul de la puissance totale.
Merci Nicolas.

NicolasBernaerts, 2024-10-14:
Cela devrait être corrigé dans la beta3.

Vincent835, 2024-10-14:
Nicolas,
Cela me semble tout bon.
Je vais me coucher, je regarderai mieux demain.
Merci beaucoup.

Vincent835, 2024-10-15:
Bonjour Nicolas, 
Je confirme que ça marche bien.
Concernant la différence de configuration des compteurs entre v13 et v14, lors de mes essais je me suis rendu compte que j’avais choisi le P1 smart meter, mais que j’aurais pu choisir le compteur normal. La transition avec la v14 aurait été plus simple.
Sur la capture d’écran les 2 types de compteur sont proposés dans tasmota.
[image]

Autre constat, pour le moment, je n’ai pas d’erreur de lecture des trames TIC. Alors qu’avec la v13 j’en avais tous les jours quelques unes. Et systématiquement lorsque je me connectais à la page Web de tasmota teleinfo. 

Merci beaucoup 

Vincent

NicolasBernaerts, 2024-10-16:
Bonjour Vincent, 
Effectivement j'ai revu la gestion de la réception série et cela limite très fortement le nombre d'erreurs de réception, en particulier quand l' esp est fortement sollicité.

closed 2024-10-17 by NicolasBernaerts