NicolasBernaerts/tasmota #46: remontées infos dans home assistant

keanoekai, 2024-10-18:
Bonjour Nicolas,
Tout d'abord je veux vous féliciter pour l'excellent travail que vous faites sur ce projet.

En utilisant le firmware Charles, toutes les infos de mon compteur sont publiés via l'intégration Tasmota dans mon Home assistant :
[image]

Lorsque j'utilise votre firmware, les infos sont publiés via l'intégraton MQTT et je n'ai rien dans l'intégration Tasmota :
[image]

Est-ce bien normal? ou est-ce un problème de configuration de mon côté?
D'avance, merci pour votre retour.

NicolasBernaerts, 2024-10-19:
Bonjour keanoekai,
Cela semble normal car j'ai écrit l'intégration from scratch  en me basant sur la documentation du projet Home Assistant pour l'intégration via MQTT.
L'intégration déclare tout une série de sensors spécifiques au compteur. Vous pouvez le voir en mode console via la publication des topics homeassistant/sensor au démarrage.
Mais n'utilisant pas moi même home assistant j'aurais du mal à vous aider quand à la configuration dans l'outil lui même.

keanoekai, 2024-10-19:
Merci pour cette précision. Cela me rassure. Je commençais à penser qu’il y avait un souci dans la config de mon Home assistant. Le fait que  votre firmware (comme celui de Charles) soit fondé sur Tasmota peut amener à penser que la publication des données TIC dans Home assistant  se fasse via l’intégration Tasmota. Maintenant je n’ai plus aucun doute. Encore merci et bonne continuation.

NicolasBernaerts, 2024-10-19:
Je vais préciser dans le readme sur GitHub que l'intégration est différente de celle par défaut de Tasmota.