NicolasBernaerts/tasmota #47: Denky D4 & InfluxBD integation

Spike91000, 2024-11-21:
Bonjour

 

Je viens de flasher un Denky D4 v1.3 avec le firmware [tasmota32-teleinfo-denkyd4.factory.bin v14.8 & souhaiterais importer les relevés dans une instance INFLUXDB.

Cependant l'integration INFLUXDB n'est plus disponible 
[image]

Débutant en domotique avec le flashing des ESP, ai-je utilisé la bonne version du firmware  ou cette option n'est elle plus intégré ? 

Cordialement

NicolasBernaerts, 2024-11-21:
Bonjour Spike,
La version avec intégration InfluxDB est actuellement en test.
Je ne l'ai pas encore publiée mais j'ai commencé à mettre à jour la documentation.
La version étant plutôt stable, je vais publier la version Denky D4 qui l'intègre.
Ce sera la v14.9beta9.

NicolasBernaerts, 2024-11-21:
Le binary DenkyD4 est publié.

Spike91000, 2024-11-22:
Merci Bcp
Je le testerai ce week-end

closed 2024-11-22 by Spike91000