NicolasBernaerts/tasmota #48: v14.8, TEMPO, problèmes couleur / retard HC HP

CTPA, 2024-11-24:
Bonjour,
J'ai ce firmware : tasmota32-teleinfo-ethernet.bin / en version 14.8

La "Couleur Demain" reste simplement à "Bleu" et la "Couleur Aujourd'hui" est passée le 21 & 22 nov en "Bleu" sans raison apparente.
[image]
la commutation Heure Pleine <> Heure Creuse est très régulièrement décalé de 0 jusqu'à 20 minutes comme ce matin (en moyenne 10min)
[image]

Merci pour votre travail

CTPA, 2024-11-25:
> S'il y a un décalage de quelques minutes, c'est que votre fournisseur ne change pas la prériode à l'heure pile.
C'est logique car toutes les régions ne basculent pas en même temps pour éviter des pics de conso.

Je l'ignorais. Ce qui est surprenant, c'est le compteur HC se met bien à compter a partir de 22h et s'arrête à 6h (horaires de mes HC)
Sur les 2 images suivantes on voit bien que malgré que le compteur nous indique toujours être en HP Bleu, il compte bien en HC.
[image]
[image]

retard de 6 min ce soir pour l'indication en HC
[image]

> Il faudrait :
noter la valeur de la clé STGE
m'indiquer votre type de contrat
m'indiquer les heures de changement dans votre contrat
Je pourrais alors regarder quel est le soucis.

clé STGE : 013AC401
contrat tempo triphasée 18kva
HP-HC => 22h - 6h

Extract console :
`"STGE":"013AC401","MSG1":"PAS DE MESSAGE","PRM":"21198552806784","RELAIS":"000","NTARF":"02","NJOURF":"00","NJOURF+1":"00","PJOURF+1":"00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE"`

Merci

NicolasBernaerts, 2024-11-30:
Sur les linky, il arrive régulièrement qu'il y ait un décalage entre l'évolution des compteurs et la période publiée.
ENEDIS vous répondra que le seul qui fasse foi est le compteur.

Vous pourrez vérifier ce décalage en analysant la donnée NTARF. C'est elle qui indique la période en cours, elle est founie par le compteur à titre informatif. C'est cette donnée qui est utilisée pour déterminée la période en cours sur mon firmware.
D'après vos captures, cette donnée doit être mise à jour par le compteur avec un certain décalage.

Du coup, pour garantir la pertinence de la période en cours, je devrais peut être tracer les évolutions de compteurs pour ajuster la période en cours indépendemment de la donnée NTARF publiée. Je vais analyser cette évolution.

Petite question complémentaire : je vois que vous avez Période 5 et 6 d'affiché. Cela signifie que votre identifiant de contrat n'est pas dans la base des contrats connus par le firmware. Quel est votre nom de contrat et auprès de quel fournisseur ?

CTPA, 2024-11-30:
Oui j'avais même l'ensemble des périodes inconnues au passage de la 14.8béta à la 14.8, mais elle se complète au fur et à mesure que les jours bleu blanc rouge arrivent.

Je suis chez EDF et effectivement le nom n'est pas en majuscule :
`"name":"Tempo","period":"HC BLEU","color":"Bleu","hour":"Creuse","tday":"Bleu","tmrw":"Bleu"`

Merci pour les modifications et la publication de la 14.9 en version Ethernet.

Je test et je vous fait un retour.

CTPA, 2024-12-01:
Voici un premier retour, de l'interface EthInfo en 14.9beta9

Avant la commande : `energyconfig reset`
[image]
Après la commande : `energyconfig reset`
[image]

je vous transmettrais les historique "home assistant" dès qu'il y aura 2-3 commutations.

NicolasBernaerts, 2024-12-01:
Je viens d'uploader une v14.9beta10 qui corrige un bug sur la détection de période basée sur les incréments compteurs.
Je vous préconise de la mettre à jour pour vos tests HA.

CTPA, 2024-12-02:
Bonsoir,
Un premier petit retour, j'ai mis a jour hier soir à 20h avec la v14.9beta10 et les 2 premières commutation se sont bien passer :
[image]
@22:02:03 j'ai bien eu la commutation pleine vers creuse
@06:02:03 j'ai bien eu la commutation creuse vers pleine
A noté qu'il n'y avait pas de changement de couleur.

Par contre ce soir à 22h30 toujours pas commutation, malgré le fait que le compteur heure creuse compte bien
[image]
[image]

j'ai extrait ces 4 messages consoles :
```
22:17:02.468 MQT: Compteur_Maison_Linky/tele/SENSOR = {"Time":"2024-12-02T22:17:02","Eth":"C8:F0:9E:54:C7:EB","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":235,"I1":7.00,"P1":1552,"W1":0,"U2":237,"I2":1.00,"P2":235,"W2":0,"U3":233,"I3":0.00,"P3":104,"W3":0,"U":235,"I":8.00,"P":1891,"W":0,"C":0.00,"YDAY":1029740,"TDAY":91580},"CONTRACT":{"serial":31876067693,"name":"Tempo","period":"Pleines Bleu","color":"Bleu","hour":"Pleine","tday":"Bleu","tmrw":"Bleu","CONSO":103922549,"HC BLEU":53476039,"HP BLEU":45805355,"HC BLANC":2518106,"HP BLANC":1173422,"HC ROUGE":632297,"HP ROUGE":317331},"ALERT":{"Load":0,"Volt":0,"Preavis":1,"Label":"PM"}}

22:22:02.437 MQT: Compteur_Maison_Linky/tele/SENSOR = {"Time":"2024-12-02T22:22:02","Eth":"C8:F0:9E:54:C7:EB","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":234,"I1":6.00,"P1":1425,"W1":0,"U2":237,"I2":1.00,"P2":265,"W2":0,"U3":233,"I3":0.00,"P3":105,"W3":0,"U":234,"I":7.00,"P":1795,"W":0,"C":0.00,"YDAY":1029740,"TDAY":91737},"CONTRACT":{"serial":31876067693,"name":"Tempo","period":"Pleines Bleu","color":"Bleu","hour":"Pleine","tday":"Bleu","tmrw":"Bleu","CONSO":103922706,"HC BLEU":53476196,"HP BLEU":45805355,"HC BLANC":2518106,"HP BLANC":1173422,"HC ROUGE":632297,"HP ROUGE":317331},"ALERT":{"Load":0,"Volt":0,"Preavis":1,"Label":"PM"}}

22:27:02.431 MQT: Compteur_Maison_Linky/tele/SENSOR = {"Time":"2024-12-02T22:27:02","Eth":"C8:F0:9E:54:C7:EB","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":235,"I1":6.05,"P1":1423,"W1":0,"U2":235,"I2":1.00,"P2":237,"W2":0,"U3":232,"I3":3.33,"P3":774,"W3":0,"U":234,"I":10.39,"P":2434,"W":0,"C":0.00,"YDAY":1029740,"TDAY":91892},"CONTRACT":{"serial":31876067693,"name":"Tempo","period":"Pleines Bleu","color":"Bleu","hour":"Pleine","tday":"Bleu","tmrw":"Bleu","CONSO":103922861,"HC BLEU":53476350,"HP BLEU":45805355,"HC BLANC":2518106,"HP BLANC":1173422,"HC ROUGE":632297,"HP ROUGE":317331},"ALERT":{"Load":0,"Volt":0,"Preavis":1,"Label":"PM"}}

22:32:02.433 MQT: Compteur_Maison_Linky/tele/SENSOR = {"Time":"2024-12-02T22:32:02","Eth":"C8:F0:9E:54:C7:EB","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":233,"I1":7.00,"P1":1598,"W1":0,"U2":236,"I2":1.00,"P2":220,"W2":0,"U3":232,"I3":0.00,"P3":114,"W3":0,"U":233,"I":8.00,"P":1932,"W":0,"C":0.00,"YDAY":1029740,"TDAY":92046},"CONTRACT":{"serial":31876067693,"name":"Tempo","period":"Pleines Bleu","color":"Bleu","hour":"Pleine","tday":"Bleu","tmrw":"Bleu","CONSO":103923015,"HC BLEU":53476505,"HP BLEU":45805355,"HC BLANC":2518106,"HP BLANC":1173422,"HC ROUGE":632297,"HP ROUGE":317331},"ALERT":{"Load":0,"Volt":0,"Preavis":1,"Label":"PM"}}
```

Toujours dans la console j'ai poussé la commande : energyconfig reset
La commande est suivit d'un redémarrage, et là, tout est rentré dans l'ordre. 
[image]

```
22:47:02.467 MQT: Compteur_Maison_Linky/tele/SENSOR = {"Time":"2024-12-02T22:47:02","Eth":"C8:F0:9E:54:C7:EB","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":234,"I1":6.00,"P1":1417,"W1":0,"U2":232,"I2":5.00,"P2":1114,"W2":0,"U3":233,"I3":1.00,"P3":103,"W3":0,"U":233,"I":12.00,"P":2634,"W":0,"C":0.00,"YDAY":1029740,"TDAY":92508},"CONTRACT":{"serial":31876067693,"name":"Tempo","period":"Creuses Bleu","color":"Bleu","hour":"Creuse","tday":"Bleu","tmrw":"Bleu","CONSO":103923477,"HC BLEU":53476966,"HP BLEU":45805355,"HC BLANC":2518106,"HP BLANC":1173422,"HC ROUGE":632297,"HP ROUGE":317331},"ALERT":{"Load":0,"Volt":0,"Preavis":1,"Label":"PM"}}

22:52:02.440 MQT: Compteur_Maison_Linky/tele/SENSOR = {"Time":"2024-12-02T22:52:02","Eth":"C8:F0:9E:54:C7:EB","METER":{"PH":3,"ISUB":30,"PSUB":6000,"PMAX":6000,"U1":234,"I1":6.04,"P1":1415,"W1":0,"U2":232,"I2":4.81,"P2":1117,"W2":0,"U3":233,"I3":0.48,"P3":114,"W3":0,"U":233,"I":11.35,"P":2646,"W":0,"C":0.00,"YDAY":1029740,"TDAY":92658},"CONTRACT":{"serial":31876067693,"name":"Tempo","period":"Creuses Bleu","color":"Bleu","hour":"Creuse","tday":"Bleu","tmrw":"Bleu","CONSO":103923627,"HC BLEU":53477118,"HP BLEU":45805355,"HC BLANC":2518106,"HP BLANC":1173422,"HC ROUGE":632297,"HP ROUGE":317331},"ALERT":{"Load":0,"Volt":0,"Preavis":1,"Label":"PM"}}
```
Demain à 6h00 on sera en jours rouge, du coup on verra si le changement de couleur fonctionne

NicolasBernaerts, 2024-12-02:
Il y a bien un problème ... votre cosphi est à 0 ainsi que la puissance active à 0W alors que la puissance apparente est bien réelle. Il doit y avoir un problème d'interprétation des étiquettes du tempo triphasé.

CTPA, 2024-12-02:
Effectivement je n'avais pas remarqué.
Voici une trame complète :
```
<html>
<body>
<!--StartFragment-->
ADSC | 031876067693 | E
-- | -- | --
VTIC | 02 | J
DATE | H241203004951 | @
NGTF | TEMPO | F
LTARF | HC BLEU | ^
EAST | 103936569 | 9
EASF01 | 053490058 | D
EASF02 | 045805355 | F
EASF03 | 002518106 | ;
EASF04 | 001173422 | 9
EASF05 | 000632297 | C
EASF06 | 000317331 | 9
EASF07 | 000000000 | (
EASF08 | 000000000 | )
EASF09 | 000000000 | *
EASF10 | 000000000 | "
EASD01 | 039959361 | M
EASD02 | 034866678 | Q
EASD03 | 016681100 | 9
EASD04 | 012429430 | <
IRMS1 | 022 | 2
IRMS2 | 020 | 1
IRMS3 | 016 | 7
URMS1 | 236 | E
URMS2 | 233 | C
URMS3 | 237 | H
PREF | 18 | H
PCOUP | 18 | "
SINSTS | 13913 | W
SINSTS1 | 05290 | G
SINSTS2 | 04754 | L
SINSTS3 | 03869 | S
SMAXSN | H241203004902 13479 | @
SMAXSN1 | H241203003004 04981 | '
SMAXSN2 | H241203004802 04479 | 1
SMAXSN3 | H241203004858 04108 | 2
SMAXSN-1 | H241202002829 13981 | !
SMAXSN1-1 | H241202002831 05281 | E
SMAXSN2-1 | H241202031358 04765 | R
SMAXSN3-1 | H241202005232 04294 | H
CCASN | H241203003000 12942 | <
CCASN-1 | H241203000000 01852 | U
UMOY1 | H241203004000 236 | )
UMOY2 | H241203004000 232 | &
UMOY3 | H241203004000 235 | *
STGE | 2D3A8000 | W
DPM2 | 241203060000 00 |  
FPM2 | 241204060000 00 | #
MSG1 | PAS DE MESSAGE | <
PRM | 21198552806784 | C
RELAIS | 001 | C
NTARF | 01 | N
NJOURF | 00 | &
NJOURF+1 | 00 | B
PJOURF+1 | 00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | .
PPOINTE | 00004005 06004006 22004005 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | '

<!--EndFragment-->
</body>
</html>
```

CTPA, 2024-12-05:
Voilà l'historique des 4 derniers jours en version 14.9beta10 avec un redémarrage le 2 décembre à 22h47.
[image]

 - L'information "Contrat Couleur" est bonne mais en retard le 3 décembre, le jour de la commutation en Rouge à 8h37 au lieu de 6h00.
 - L'information "Contrat Heure" marche pour 3 commutations puis semble se freezer
 - Les informations "Couleur Aujourd'hui/Demain" semble fonctionner depuis le dernier redémarrage
 - L'information "Contrat Période" n'a aucune erreur mais à des retards
 
 Je continu l'expérimentation.

NicolasBernaerts, 2025-01-12:
Bonjour,
Je viens de déposer une v14.9beta11 qui devrait corriger le problème.
Tenez moi au courant.

CTPA, 2025-01-12:
Bonjour,
Merci pour votre retour, je fais la MAJ et je vous tiens au courant.

Dans l'espoir que les graphiques soient plus explicites que les précédents, j'ai créé deux variables :
EDJ_JourNuit : basée sur l'heure de la journée => 6h - 22h
EDF_Tempo : qui définit la couleur de la journée en fonction de la dérivée des compteurs bleu/blanc/rouge

[image]

Observations : La couleur n'est pas toujours correcte, notamment le 11 janvier où elle est passée à blanc avec environ 11 heures de retard. Le décalage est synchrone avec le passage heures creuses/pleines (aussi avec 11 heures de retard).

NicolasBernaerts, 2025-01-12:
Sachant que vous faites la mise du correctif aujourd'hui, soit le 12, il est normal que les données du 11 ne soient pas bonnes.

CTPA, 2025-01-12:
Oui oui,
Nous nous sommes mal compris, c'était plus pour vous donner des éléments sur la version précédente comme je m'y étais engagé plus haut.
:-)

CTPA, 2025-01-12:
Je n'ai pas encore fait la MAJ.
Le firmware [tasmota32-teleinfo-ethernet.bin] est toujours en version 14.9beta10

NicolasBernaerts, 2025-01-12:
Ok, je vais publier la version ethernet

Bibinsa, 2025-01-12:
Bonjour, et encore merci pour votre travail sur le code Teleinfo pour Tasmota.

J'utilise cette version depuis quelques mois sur un Wemos ESP8266.
Je constate aussi le décalage HP/HC.
L'information n'est pas remontée en MQTT certaines fois (je pensais avoir trouvé une logique avec les passage HC/HP et changement de couleur mais ça n'est pas que là).

(je contourne le problème avec Home Assistant en faisant rebooter le Wemos actuellement).

Est-ce que vous pouvez compiler le fix pour tasmota-teleinfo-4m (bin.gz ou bin).

Merci :)

NicolasBernaerts, 2025-01-12:
Je viens d'uploader une v14.9beta11 pour ESP8266 4M et 16M.
Tenez moi au courant.

NicolasBernaerts, 2025-01-13:
Je viens d'uploader le binaire tasmota-teleinfo.bin.gz.
Mais non testé ... merci du retour.

Bibinsa, 2025-01-13:
> Je viens d'uploader une v14.9beta11 pour ESP8266 4M et 16M. Tenez moi au courant.

Merci.
Testé aujourd'hui, ça semble bon.
Tempo HC Bleu vers HP Rouge
Tempo HP Rouge vers HC Rouge

Pas de soucis.
Juste obligé de flasher le Wemos en USB, OTA non fonctionnel.

NicolasBernaerts, 2025-01-14:
> Impeccable, merci. Ca tourne bien pour le moment, ça a fix le problème de couleur que j'avais aussi en Tempo, et mon index heures creuses rouge qui était bloqué à 0.

Vous êtes en tempo historique ?

NicolasBernaerts, 2025-01-14:
> > Je viens d'uploader une v14.9beta11 pour ESP8266 4M et 16M. Tenez moi au courant.
> 
> Merci. Testé aujourd'hui, ça semble bon. Tempo HC Bleu vers HP Rouge Tempo HP Rouge vers HC Rouge
> 
> Pas de soucis. Juste obligé de flasher le Wemos en USB, OTA non fonctionnel.

L'OTA peut planter quand l'ESP n'a pas assez de RAM. Dans ce cas, supprimez TinfoRX de la configuration du module, rebooter et flasher. Cela devrait passer.

closed 2025-01-18 by NicolasBernaerts

CTPA, 2025-01-19:
J'ai fait un test sur une semaine, et ça marche nickel, bleu/blanc/rouge, sans de retard, sans d'erreur, vraiment nickel

[image]

Les couleurs d'aujourd'hui et demain font n'importe quoi, mais ce serra l'objet d'un autre bug report :-)

Un grand merci pour votre travail!

Bibinsa, 2025-01-19:
Bonjour, idem.
Depuis le patch beta11 je n'ai plus de problème de décalage.
C'est noté pour l'OTA, à la prochaine version je ferai l'essai.

Merci pour le travail effectué !