NicolasBernaerts/tasmota #49: Influxdb integration : Senseur non trouvé dans Influxdb

Spike91000, 2024-11-29:
Bonjour 

Je teste l’intégration de mon Denky D4 dans Influxdb. 
Je ne retrouve pas toutes les valeurs transmise 
[image]

Dans Influxdb, ,je ne trouve que les senseurs lié à la conso ( courant, tension, W, VA) et certains liés au contrat  mais pas les données liés aux consommations HC, HP etc.
[image]
[image]

Ai je un probleme de configuration ( Denky ou Influxdb) ?

NicolasBernaerts, 2024-11-30:
Bonjour,

Tout d'abord une première remarque : votre capture d'écran présente le topic LIVE.
Celui-ci est très particulier et il n'est sans doute pas nécessaire de le publier.

Pour la publication InfluxDB, je vais rajouter la liste des données publiées dans la page Github de présentation du projet.

Cette intégration est très récente et a été faite suite à une demande spéciffique de l'universite de Grenoble. Seules les données dynamiques (évoluant régulièrement) les plus importantes sont publiées. En particulier je ne publie pas les données "statiques" qui n'ont que très peu d'intérêt dans une base de données temporelle.

Si des données apparaissant dans le topic SENSOR seraient intéresssantes coté InfluxDB, dites le moi et je pourrais les ajouter.

Spike91000, 2024-12-02:
Bonjour

Merci pour la réponse.
Je comprend mieux pourquoi les données n’étaient pas visibles :)
Je débute en domotique et découvre les possibilités au fur et à mesure.  

Je compte utiliser Influxdb pour collecter et stocker les données ( conso EDF, station Météo) pour les utiliser via Home-Assistant ou les visualiser via Grafana.
J'imagine bien que les besoins de l'université ne soient pas les memes que celles d'un particulier bricoleur. 
D'autant plus que je ne connais l'impact sur la taille de base de données si on y intègre d'autres données. 

Ce qui m’intéresse particulièrement ce sont les index heures creuses/pleines , la conso totale du jour,

NicolasBernaerts, 2024-12-02:
Bonsoir,
Je vais voir comment ajouter ces données au flux de mise à jour InfluxDB.

NicolasBernaerts, 2025-01-12:
Bonjour,
Je viens de déposer une v14.9beta11 qui ajoute des données coté intégration InfluxDB.
Tenez moi au courant.

NicolasBernaerts, 2025-01-26:
No feedback

closed 2025-01-26 by NicolasBernaerts