NicolasBernaerts/tasmota #50: Hisorique monophase vers domoticz

RolPat24, 2024-12-10:
Bonjour Nicolas 
Tout d'abord, un grand merci pour avoir développé cette magnifique version.

Depuis peu, je suis passé autoconsomateur et j'ai installé la version 8266 1M 14.8, j'avais des index bizarres avec la 8.5.1.
J'ai gardé les mêmes IDX Domoticz mais les messages pour Domoticz ne partent pas, 
je suis donc passé à la 14.9 beta3 et idem, j'ai testé différentes configs et aussi les VA sans success.

[image]
[image]
[image]
La commande domo voit bien l'intégration activée.

Par contre , j'ai testé les messages HA et je les vois bien dans le broker , je fais peut-être un erreur de paramétrage.

Merci

NicolasBernaerts, 2024-12-10:
Bonsoir RolPat24,
Ce qui est ettonant est que vous soyez toujours en mode historique en tant qu'auto-consommateur.
En effet, le mode historique ne fournit que très peu de données par rapport au mode standard.
Concernant vos soucis, pourrriez vous envoyer une copie d'écran de votre configuration Teleinfo ?
Voyez-vous des publications domoticz en mode console ?

[image]

RolPat24, 2024-12-11:
Bonjour Nicolas
Je suis en mode historique car je ne revend pas le surplus de ma production, je ne peux voir le peu d'injection que sur le Linky.
J'ai un contrat CASI (Convention d'AutoConsommation Sans Injection) avec Enedis, je ne suis pas considéré comme producteur
donc je n'ai pas droit au mode standard !!!!!!
Justement en mode console, je ne vois pas les messages domoticz/in et si je sélectionne l'intégration Home Assistant à la place de
Domoticz, alors là je vois bien les messages HA .
J'ai essayé le choix Total global conso. à la place Total conso. Bleu , pas de message non plus

[image]

Merci

NicolasBernaerts, 2024-12-11:
Je pense avoir compris le problème.
Avec votre configuration, l'ESP est trop stressé et n'a pas le temps de publier les messages Domoticz qui sont depriorises.

Décochez Données TeleInfo brutes 
Cochez évolution +- 100 VA
Définissez télémétrie à 30s

Vous publications Domoticz devraient être publiées toutes les 30 secondes

RolPat24, 2024-12-12:
Bonsoir,

j'ai appliqué les recommandations mais je ne vois toujours pas de messages Domoticz .

J'avais installé la version 14.9 beta3 sur un ESP 07s qui auparavant avait la version 8.5.1.
Aujourd'hui j'ai mis la 14.9 beta3 sur une Wemos D1 mini et toujours pas de messages Domoticz.
Ce qui est surprenant, c'est que les autres intégrations émettent des messages corrects. 

Demain je vais tester la teleinfo proposée par le Web Installer Tasmota et je ferai un petit retour d'info.

NicolasBernaerts, 2024-12-12:
La version 14.9 beta 3 étant assez ancienne et les travaux d'intégration de Domoticz étant assez récents, je vais publier une version récente esp8266 4m afin de vous permettre de faire des tests sur la dernière version.

Vincent835, 2024-12-12:
Bonjour, 

Mon retour d’expérience.
J’utilise aussi un esp8266 (sonoff basic modifié).
J’ai eu un souci avec les index domoticz.
Voir issue #36
J’ai finalement fait un reset 6, rechargé la config, et ça marché à nouveau.
Depuis j’ai changé de version pour la 14.9.beta3. Je n’ai aucun problème. 

[image]

Vincent

NicolasBernaerts, 2024-12-13:
Effectivement en relisant le post vous êtes passé d'un Tasmota standard à mon fork.
Sachant que je réutilise des zones mémoires dédiées à d'autres fonctions (il y a très peu de ROM disponible sur un esp8266 pour sauvegarder les paramètres), il est très important de faire un reset complet après le 1er flash.
Sinon, vous pouvez avoir des dysfonctionnements inexplicables.
Si ce n'est pas précisé dans la page de description sur GitHub, je vais l'ajouter ...

Merci Vincent !

RolPat24, 2024-12-13:
Bonjour Nicolas,

Petite précision: 
je n'ai pas upgradé mon ancienne version 8.5.1, pour la bonne raison que l'upgrade OTA me donne une erreur serveur et l'upgrade fichier binaire me dit pas assez de mémoire.
J'ai donc gardé mon ancien ESP07S en 8.5.1 et je suis donc passé par l'UART USB-Série avec un nouveau ESP07S de test .
Sachant que, chaque fois avant le flash du binaire, je fais un flash avec un blank-4M.bin. 
J'ai fait la même manip pour les tests avec les 2 WemosD1mini mais avec un blank-1M.bin.

Ce matin je télécharge la version tasmota-teleinfo-4m.bin.gz 14.9 beta10, je dézippe et je flashe le même ESP07S de test .
Je paramètre comme d'ab, j'ouvre la console, et là miracle je vois les "domoticz/in".
Si je remets une version antérieure sur le même ESP07S de test, je ne vois plus les messages domoticz ???

Conclusion, pour le moment, mon installation fonctionne correctement avec uniquement la version 14.9 beta10.
Je ferai un petit retour d'info pour dire si c'est toujours OK.

Merci Nicolas pour ce cadeau de Noël un peu en avance.
Bonnes fêtes de fin d'année

PS : Je ne ferais donc pas d'essai avec la version téléinfo du site Tasmota.

closed 2025-01-05 by NicolasBernaerts