NicolasBernaerts/tasmota #51: Téléfinfo | v14.8 | TEMPO | incohérence d'unité

othrd, 2024-12-20:
Bonjour,

J'utilise la v14.8 de téléinfo sur un ESP32. Je viens de basculer sur un abonnement EDF Tempo.
Je viens de passer sur un jour blanc pour la première et j'ai constaté un comportement étrange : 

Côté Linky , le compteur indique 1kWh pour les HP Blanc
[image]

Côté ESP, l'interface indique 0.001 kWh
[image]

Dans la console la sortie MQTT retourne 1W 
[image]

Cela est peut-être lié au premier kWh , je surveillerai par la suite.

NB: je n'ai pas constaté d'erreur côté HP / HC Bleu

Merci pour votre travail

othrd, 2024-12-21:
Bonjour,
Je viens de constater le même comportement sur les HC JW lors du premier passage aux heures creuser la nuit dernière.

Ce matin l'ESP indiquait 0,001kWh:
[image]

MQTT retournait 1Wh :
[image]

Alors que dans les messages j'avais la bonne valeur 7082 Wh:
[image]

Après reboot l'affichage côté ESP et le retour MQTT sont corrects :
[image]

[image]

NicolasBernaerts, 2024-12-21:
Je crois avoir compris ...
Avec le contrat Tempo historique, l'étiquette du contrat change en fonction de la période ! BBR9, BBR(, ...
C'est le seul contrat qui fait cela, heureusement.
Donc au changement de période l'ESP pense à un nouveau contrat et l'increment ne se fait plus. Un redémarrage remet à 0 le contrat détecté et tout est de nouveau ok.
Je pensais avoir géré cette spécificité, mais avec l'auto découverte des contrats inconnus, cela a peut être sauté. Je vais vérifier.
Quelle version d'ESP utilisez vous ?

othrd, 2024-12-21:
J'utilise un ESP32-WROOM-32, plus précisément ce modèle: 
[image]

J'ai intallé ce firmware : tasmota32-teleinfo.factory.bin

Concernant le mode historique j'avais regardé les specs du mode standard dans la doc de RTE et j'avais cru comprendre qu'il n'exportait pas les données en Tempo:
[image]

Ce oblige donc a rester en mode historique pour le contrat Tempo?

NicolasBernaerts, 2024-12-21:
En fait c'est je contraire. Le mode standard fourni beaucoup plus de données que le mode historique, en particulier pour les contrats tempo.
Je vais essayer de publier une version esp32 avec le correctif tempo historique. Vous me direz si c'est mieux.

othrd, 2024-12-21:
Ok, merci pour votre support.
Conseillez-vous d'utiliser plutôt le mode historique ou standard avec le firmware ?

NicolasBernaerts, 2025-01-05:
Les 2 sont possibles.
Mais le mode standard fournit beaucoup plus d'informations (tension, calendrier, ...).

othrd, 2025-01-06:
Retour d'utilisation après 2 semaines:
J'ai rebooté l'ESP après chaque premier passage de période (HC rouge, HP Rouge etc)
Maintenant les périodes se mettent à jour correctement chaque jour, les incréments sont ok, tout fonctionne vraiment très bien.

Je propose de clôturer l'issue.

NicolasBernaerts, 2025-01-06:
C'est parfait.
La prochaine version stable intègrera la gestion des étiquettes dynamiques de contrat Tempo historique.

closed 2025-01-06 by NicolasBernaerts