NicolasBernaerts/tasmota #52: Tempo | Ecowatt - Denky - Teleinfo 14.9beta9 - pas de mise à jour

sosandroid, 2024-12-25:
Fonctionnement de l'interrogation API RTE semble bloqué.

- Création du compte et de l'appli sur le site RTE avec association des APIs à interroger
- Copie de la clef privée en base 64 et insertion grâce à la commande associée
- Activation du module à l'affichage
- Pas de mise à jour
- Forcer l'update via la commande associée
- Pas d'effet apparent

Ai-je raté un truc ?
[image]
[image]

NicolasBernaerts, 2024-12-25:
L'icône rouge signifie qu'il n'y a aucune connexion a un broker MQTT. Le problème vient sans doute de là. Configurez un broker MQTT et vérifiez ensuite.

sosandroid, 2024-12-25:
Merci,
L'API RTE n'avais pas de lien avec MQTT pour moi. J'ai désactivé la fonction car pas dans mon setup. EmonCMS ne supporte pas la transmission MQTT/JSON. C'est donc un script Berry qui s'en charge.

Lors de l'activation de MQTT vers tes.mosquitto.org, le Denky s'est pendu. Il faudra que j'aille le récupérer demain pour tenter un retour aux paramètres usine...

A suivre...

sosandroid, 2024-12-25:
Après avoir récupéré et reconfiguré, je constate la connexion aux APIs RTE. Ces dernières renvoient une erreur 500 (de leur coté).

Je dirais donc que c'est fonctionnel lorsque MQTT est en fonctionnement ...
Je vais recouper MQTT et désactiver les APIs

````
22:19:59.601 CMD: tempo_update
22:19:59.607 MQT: denky_DEADBEEF/stat/RESULT = {"tempo_update":"Done"}
22:20:01.504 RTE: Tempo - URL https://digital.iservices.rte-france.com/open_api/tempo_like_supply_contract/v1/tempo_like_calendars?start_date=2024-12-24T00:00:00+01:00&end_date=2024-12-27T00:00:00+01:00
22:20:01.506 RTE: Tempo - Connexion done [https://digital.iservices.rte-france.com/open_api/tempo_like_supply_contract/v1/tempo_like_calendars?start_date=2024-12-24T00:00:00+01:00&end_date=2024-12-27T00:00:00+01:00]
22:20:02.127 RTE: Tempo - Error  [500]
````

sosandroid, 2024-12-25:
Closed
A moins que ce soit une optim possible

closed 2024-12-25 by sosandroid

NicolasBernaerts, 2024-12-26:
Dans la prochaine version je vais supprimer la vérification de connexion MQTT.
Seule une vérification RTC sera réalisée car nous parlons d'un calendrier.