NicolasBernaerts/tasmota #53: Integration HASS: perte sensors apres redemarrage mqtt server et HA

fastm3, 2024-12-31:
Version 14.8
On redemarre le serveur mqtt -> Les sensors en auto discovery sur HA sont toujours available.
Par contre, si on redemarre ensuite HA -> les sensors en auto discovery deviennent tous unavailable.
Avec un publish manuel du topic de l'autodiscovery, le sensor HA correspondant reprends vie.
Un redémarrage de l'esp regle donc aussi le probleme car il republie l'autodiscovery de tous les sensors et c'est la seule solution en pratique.
Pas expert de mqtt mais n'y a t'il pas possibilité que l'autodiscovery résiste a un reboot du serveur mqtt. 
Il me semble que le "retain" flag doit etre setté pour les topics d'autodiscovery sans pouvoir valider que c'est le probleme.

NicolasBernaerts, 2025-01-05:
Bonjour,
La version 14.8 est une version assez ancienne.
Depuis la version 14.9, la publication HomeAssistant se fait en mode "retain", ce qui devrait régler le problème.
Beaucoup d'évolutions étant en cours, je n'ai pas publié de version stable 14.9 pour le moment.
Quel chipset utilisez vous ?

fastm3, 2025-01-05:
Ah ok, je ne pensais pas que la "beta" puisse corriger cela. Actuellement, j'ai un S3 mini 16M , qui remplace tres avantageusement avec votre firmware un esp8266 dans un tres tres ancien montage dans boitier rail din . J'avais utilisé le binaire dispo par simplificité...Je vais voir avec le 14.9 en essayant de le compiler. Je ferai un retour. Merci pour ce super outil.

NicolasBernaerts, 2025-01-05:
Les sources ne sont pas publiés à la dernière version car ils évoluent beaucoup en ce moment. Je vais publier une version 14.9beta pour le ESP32S3. Vous pourrez l'utiliser pour vos tests en attendant que je publie une version stable.

fastm3, 2025-01-05:
👍

NicolasBernaerts, 2025-01-12:
Bonjour,
Je viens de déposer une v14.9beta11 qui devrait gérer le mode retain.
N'utilisant pas HA, je n'ai pas pu réaliser tous les tests coté HA.
Tenez moi au courant.

fastm3, 2025-01-12:
Je viens de refaire le test avec la nouvelle version. Les sensors restent availables avec la v14.9beta11 . Top. Merci bcp.

closed 2025-01-12 by NicolasBernaerts