NicolasBernaerts/tasmota #54: Décodage STGE

sosandroid, 2025-01-09:
Bonjour

aujourd'hui = 20250109 vers 19h

Aujourd'hui je viens de remarquer qu'une probable erreur de décodage a eu lieu.
Sur la Home, aujourd'hui = rouge, demain = bleu
Sur RTE, aujourd'hui = blanc, demain = rouge
STGE = **42**3ACD01 => 0x42 => 0100 **0010** => 0010 => **00** demain 0 : inconnu, **10** aujourd'hui 2 : blanc, 

C'est une erreur de décodage ou j'ai raté une source de data ?

Denky D4
Teleinfo 14.9beta9

[image]

NicolasBernaerts, 2025-01-09:
Bonsoir,
Je suis en train de revoir complètement le décodage du calendrier en mode standard et en mode historique.
Je devrais publier une nouvelle version pour test très prochainement.

NicolasBernaerts, 2025-01-12:
Bonjour,
Je viens de déposer une v14.9beta11 qui devrait corriger le problème.
Tenez moi au courant.

sosandroid, 2025-01-12:
Installé, 
test en cours

sosandroid, 2025-01-13:
Premièrement merci d'avoir décorrélé l'API RTE de la publication MQTT.
Pour le moment, cela a l'air de fonctionner.

La valeur STGE courante à 18h47 : 833AD501
[image]
[image]

sosandroid, 2025-01-13:
Avec une trame c'est mieux
````json
{
    "Time": "2025-01-13T19:05:53",
    "METER": {
        "PH": 3,
        "ISUB": 15,
        "PSUB": 3000,
        "PMAX": 3840,
        "U1": 241,
        "I1": 0,
        "P1": 31,
        "W1": 31,
        "U2": 222,
        "I2": 1,
        "P2": 229,
        "W2": 229,
        "U3": 240,
        "I3": 4,
        "P3": 1058,
        "W3": 1058,
        "U": 234,
        "I": 5,
        "P": 1318,
        "W": 1318,
        "C": 1,
        "YDAY": 25399,
        "TDAY": 15119,
        "PP": 0,
        "PW": 0,
        "PYDAY": 169,
        "PTDAY": 6201
    },
    "CONTRACT": {
        "serial": "--",
        "name": "Tempo",
        "period": "Pleines Rouge",
        "color": "Rouge",
        "hour": "Pleine",
        "tday": "Bleu",
        "tmrw": "Bleu",
        "CONSO": 18275745,
        "HC BLEU": 8003687,
        "HP BLEU": 9015502,
        "HC BLANC": 449919,
        "HP BLANC": 313633,
        "HC ROUGE": 292003,
        "HP ROUGE": 201002,
        "PROD": 6767489
    },
    "RTE": {
        "tday": "Rouge",
        "tmrw": "Rouge"
    },
    "CAL": {
        "level": 3,
        "hp": 1,
        "tday": {
            "0": 10,
            "1": 10,
            "2": 10,
            "3": 10,
            "4": 10,
            "5": 10,
            "6": 10,
            "7": 10,
            "8": 10,
            "9": 10,
            "10": 10,
            "11": 10,
            "12": 31,
            "13": 31,
            "14": 31,
            "15": 31,
            "16": 31,
            "17": 31,
            "18": 31,
            "19": 31,
            "20": 31,
            "21": 31,
            "22": 31,
            "23": 31,
            "24": 31,
            "25": 31,
            "26": 31,
            "27": 31,
            "28": 31,
            "29": 31,
            "30": 31,
            "31": 31,
            "32": 31,
            "33": 31,
            "34": 31,
            "35": 31,
            "36": 31,
            "37": 31,
            "38": 31,
            "39": 31,
            "40": 31,
            "41": 31,
            "42": 31,
            "43": 31,
            "44": 30,
            "45": 30,
            "46": 30,
            "47": 30
        },
        "tmrw": {
            "0": 30,
            "1": 30,
            "2": 30,
            "3": 30,
            "4": 30,
            "5": 30,
            "6": 30,
            "7": 30,
            "8": 30,
            "9": 30,
            "10": 30,
            "11": 30,
            "12": 30,
            "13": 30,
            "14": 30,
            "15": 30,
            "16": 30,
            "17": 30,
            "18": 30,
            "19": 30,
            "20": 30,
            "21": 30,
            "22": 30,
            "23": 30,
            "24": 30,
            "25": 30,
            "26": 30,
            "27": 30,
            "28": 30,
            "29": 30,
            "30": 30,
            "31": 30,
            "32": 30,
            "33": 30,
            "34": 30,
            "35": 30,
            "36": 30,
            "37": 30,
            "38": 30,
            "39": 30,
            "40": 30,
            "41": 30,
            "42": 30,
            "43": 30,
            "44": 30,
            "45": 30,
            "46": 30,
            "47": 30
        }
    },
    "ALERT": {
        "Load": 0,
        "Volt": 0,
        "Preavis": 0,
        "Label": ""
    }
}
````

NicolasBernaerts, 2025-01-13:
Merci pour le retour. 

Effectivement j'ai de corrélé les 2 calendriers en prévision des nouveaux contrats qui ne suivront pas forcément les heures standard Tempo.

Je vais faire un essai pour la couleur grise. 

Remarque : c'est incroyable ... à 18h47 votre compteur n'annonce pas encore la couleur du lendemain !

NicolasBernaerts, 2025-01-13:
est-ce que vous pourriez faire un screen d'une trame TIC (dans messages) ?
quelles sont les valeurs de DPM1 ... DPM3 et FPM1 ... FPM3 ?

sosandroid, 2025-01-14:
Aujourd'hui, 19h34
STGE 83xxxx pas d'annonce encore

[image]

Je referai la même après 22h au changement de tarif

🏷️ | 📄 | ✅
-- | -- | --
ADSC | xxx | 4
VTIC | 02 | J
DATE | H250114193304 | B
NGTF | TEMPO | F
LTARF | HP ROUGE | %
EAST | 018300635 | )
EASF01 | 008003687 | B
EASF02 | 009015502 | 9
EASF03 | 000449919 | H
EASF04 | 000313633 | 8
EASF05 | 000302424 | 5
EASF06 | 000215470 | :
EASF07 | 000000000 | (
EASF08 | 000000000 | )
EASF09 | 000000000 | *
EASF10 | 000000000 | "
EASD01 | 009966975 | S
EASD02 | 003291847 | C
EASD03 | 002484222 | :
EASD04 | 002557591 | E
EAIT | 006769964 | 4
ERQ1 | 002415378 | Y
ERQ2 | 000779717 | "
ERQ3 | 000148406 | T
ERQ4 | 001432296 | Y
IRMS1 | 000 | .
IRMS2 | 001 | 0
IRMS3 | 005 | 5
URMS1 | 244 | D
URMS2 | 219 | G
URMS3 | 239 | J
PREF | 09 | H
PCOUP | 09 | "
SINSTS | 01438 | V
SINSTS1 | 00032 | <
SINSTS2 | 00206 | @
SINSTS3 | 01198 | L
SMAXSN | H250114105145 03140 | 2
SMAXSN1 | H250114072906 00100 | $
SMAXSN2 | H250114130643 01750 | *
SMAXSN3 | H250114105141 03280 | &
SMAXSN-1 | H250108195206 02460 | ^
SMAXSN1-1 | H250108195150 01970 | R
SMAXSN2-1 | H250108070943 01850 | R
SMAXSN3-1 | H250108150326 01220 | D
SINSTI | 00000 | <
SMAXIN | H250114133349 02120 | ,
SMAXIN-1 | H250108112224 02000 | ?
CCASN | H250114193000 01278 | G
CCASN-1 | H250114190000 01276 |  
CCAIN | H250114193000 00000 | +
CCAIN-1 | H250114190000 00000 | F
UMOY1 | H250114193000 241 | /
UMOY2 | H250114193000 221 | .
UMOY3 | H250114193000 238 | 7
STGE | 833AD501 | ^
DPM2 | 250114060000 00 | !
FPM2 | 250115060000 00 | $
MSG1 | PAS DE MESSAGE | <
PRM | xxx | L
RELAIS | 000 | B
NTARF | 06 | S
NJOURF | 00 | &
NJOURF+1 | 00 | B
PJOURF+1 | 00004001 06004002 16004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | 1
PPOINTE | 00004005 06004006 16004005 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | *

sosandroid, 2025-01-14:
23h15
Annonce arrivée sur Linky

STGE xFxxxxxxx => rouge demain
Seuls sont présents ces deux là
DPM2 | 250114060000 00 | !
FPM2 | 250115060000 00 | $

[image]

🏷️ | 📄 | ✅
-- | -- | --
ADSC | xxx | 4
VTIC | 02 | J
DATE | H250114231424 | >
NGTF | TEMPO | F
LTARF | HC ROUGE | X
EAST | 018305381 | ,
EASF01 | 008003687 | B
EASF02 | 009015502 | 9
EASF03 | 000449919 | H
EASF04 | 000313633 | 8
EASF05 | 000303980 | =
EASF06 | 000218660 | >
EASF07 | 000000000 | (
EASF08 | 000000000 | )
EASF09 | 000000000 | *
EASF10 | 000000000 | "
EASD01 | 009966975 | S
EASD02 | 003291847 | C
EASD03 | 002485778 | K
EASD04 | 002560781 | @
EAIT | 006769964 | 4
ERQ1 | 002415812 | R
ERQ2 | 000779717 | "
ERQ3 | 000148406 | T
ERQ4 | 001432296 | Y
IRMS1 | 000 | .
IRMS2 | 001 | 0
IRMS3 | 005 | 5
URMS1 | 243 | C
URMS2 | 222 | A
URMS3 | 243 | E
PREF | 09 | H
PCOUP | 09 | "
SINSTS | 01522 | P
SINSTS1 | 00084 | C
SINSTS2 | 00244 | B
SINSTS3 | 01194 | H
SMAXSN | H250114105145 03140 | 2
SMAXSN1 | H250114072906 00100 | $
SMAXSN2 | H250114130643 01750 | *
SMAXSN3 | H250114105141 03280 | &
SMAXSN-1 | H250108195206 02460 | ^
SMAXSN1-1 | H250108195150 01970 | R
SMAXSN2-1 | H250108070943 01850 | R
SMAXSN3-1 | H250108150326 01220 | D
SINSTI | 00000 | <
SMAXIN | H250114133349 02120 | ,
SMAXIN-1 | H250108112224 02000 | ?
CCASN | H250114230000 01240 | 4
CCASN-1 | H250114223000 01230 | S
CCAIN | H250114230000 00000 | #
CCAIN-1 | H250114223000 00000 | C
UMOY1 | H250114231000 242 | )
UMOY2 | H250114231000 222 | (
UMOY3 | H250114231000 240 | )
STGE | AF3A9100 | *
DPM2 | 250114060000 00 | !
FPM2 | 250115060000 00 | $
MSG1 | PAS DE MESSAGE | <
PRM | xxx | L
RELAIS | 001 | C
NTARF | 05 | R
NJOURF | 00 | &
NJOURF+1 | 00 | B
PJOURF+1 | 00004001 06004002 16004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | 1
PPOINTE | 00004005 06004006 16004005 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | *

</body>
</html>

NicolasBernaerts, 2025-01-14:
Cela prouve donc que pour les contrats tempo, STGE est renseigné, mais pas les étiquettes DPM et FPM correspondantes. C'est juste complètement hors des règles annoncées pour les contrats standard. 
Je vais voir comment gérer ...
Merci

[image]

[image]

NicolasBernaerts, 2025-01-14:
Je viens de publier une nouvelle version Denky D4 qui devrait gérer le flag STGE sans les étiquettes DPM et FPM associée. Est-ce que c'est ok ?

sosandroid, 2025-01-15:
Installé, en attente de bascule du tarif vers 22h
STGE pas encore OK

(PS, j'avais le Denky complètement freeze vers 17h30 selon EmonCMS)

sosandroid, 2025-01-15:
Mise à jour effectuée, capture d'écran et des messages TIC associés

Juste après la mise à jour
STGE = 833AD501

[image]

Un peu avant le changement de tarif

STGE  = AF3A9100 

[image]

**18h52**
🏷️ | 📄 | ✅
-- | -- | --
ADSC | xxx | 4
VTIC | 02 | J
DATE | H250115185212 | B
NGTF | TEMPO | F
LTARF | HP ROUGE | %
EAST | 018320401 | "
EASF01 | 008003687 | B
EASF02 | 009015502 | 9
EASF03 | 000449919 | H
EASF04 | 000313633 | 8
EASF05 | 000313116 | 5
EASF06 | 000224544 | <
EASF07 | 000000000 | (
EASF08 | 000000000 | )
EASF09 | 000000000 | *
EASF10 | 000000000 | "
EASD01 | 009966975 | S
EASD02 | 003291847 | C
EASD03 | 002494914 | C
EASD04 | 002566665 | G
EAIT | 006772587 | /
ERQ1 | 002418409 | W
ERQ2 | 000780337 | X
ERQ3 | 000148499 |  
ERQ4 | 001432378 | Z
IRMS1 | 000 | .
IRMS2 | 001 | 0
IRMS3 | 001 | 1
URMS1 | 242 | B
URMS2 | 217 | E
URMS3 | 237 | H
PREF | 09 | H
PCOUP | 09 | "
SINSTS | 00584 | W
SINSTS1 | 00031 | ;
SINSTS2 | 00236 | C
SINSTS3 | 00316 | C
SMAXSN | H250115174352 01880 | B
SMAXSN1 | H250115155051 00610 | $
SMAXSN2 | H250115174353 01570 | 1
SMAXSN3 | H250115010927 01360 | +
SMAXSN-1 | H250109203929 05100 | [
SMAXSN1-1 | H250109203926 02970 | U
SMAXSN2-1 | H250109073755 02000 | K
SMAXSN3-1 | H250109215136 01430 | I
SINSTI | 00000 | <
SMAXIN | H250115132515 02260 | ,
SMAXIN-1 | H250109151738 00320 | P
CCASN | H250115183000 00594 | G
CCASN-1 | H250115180000 00674 | !
CCAIN | H250115183000 00000 | +
CCAIN-1 | H250115180000 00000 | F
UMOY1 | H250115185000 242 | 2
UMOY2 | H250115185000 218 | 6
UMOY3 | H250115185000 237 | 8
STGE | 833AD501 | ^
DPM2 | 250115060000 00 | "
FPM2 | 250116060000 00 | %
MSG1 | PAS DE MESSAGE | <
PRM | xxx | L
RELAIS | 000 | B
NTARF | 06 | S
NJOURF | 00 | &
NJOURF+1 | 00 | B
PJOURF+1 | 00004001 06004002 16004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | 1
PPOINTE | 00004005 06004006 16004005 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | *

**21h55**
🏷️ | 📄 | ✅
-- | -- | --
ADSC | *** | 4
VTIC | 02 | J
DATE | H250115221645 | C
NGTF | TEMPO | F
LTARF | HC ROUGE | X
EAST | 018324385 | 1
EASF01 | 008003687 | B
EASF02 | 009015502 | 9
EASF03 | 000449919 | H
EASF04 | 000313633 | 8
EASF05 | 000313439 | =
EASF06 | 000228205 | :
EASF07 | 000000000 | (
EASF08 | 000000000 | )
EASF09 | 000000000 | *
EASF10 | 000000000 | "
EASD01 | 009966975 | S
EASD02 | 003291847 | C
EASD03 | 002495237 | B
EASD04 | 002570326 | <
EAIT | 006772587 | /
ERQ1 | 002418789 | "
ERQ2 | 000780337 | X
ERQ3 | 000148499 |  
ERQ4 | 001432428 | V
IRMS1 | 000 | .
IRMS2 | 008 | 7
IRMS3 | 004 | 4
URMS1 | 242 | B
URMS2 | 216 | D
URMS3 | 240 | B
PREF | 09 | H
PCOUP | 09 | "
SINSTS | 02847 | [
SINSTS1 | 00032 | <
SINSTS2 | 01762 | H
SINSTS3 | 01052 | A
SMAXSN | H250115174352 01880 | B
SMAXSN1 | H250115155051 00610 | $
SMAXSN2 | H250115174353 01570 | 1
SMAXSN3 | H250115010927 01360 | +
SMAXSN-1 | H250109203929 05100 | [
SMAXSN1-1 | H250109203926 02970 | U
SMAXSN2-1 | H250109073755 02000 | K
SMAXSN3-1 | H250109215136 01430 | I
SINSTI | 00000 | <
SMAXIN | H250115132515 02260 | ,
SMAXIN-1 | H250109151738 00320 | P
CCASN | H250115220000 01206 | 6
CCASN-1 | H250115213000 01312 | T
CCAIN | H250115220000 00000 | #
CCAIN-1 | H250115213000 00000 | C
UMOY1 | H250115221000 240 | '
UMOY2 | H250115221000 218 | -
UMOY3 | H250115221000 239 | 1
STGE | AF3A9100 | *
DPM2 | 250115060000 00 | "
FPM2 | 250116060000 00 | %
MSG1 | PAS DE MESSAGE | <
PRM | *** | L
RELAIS | 001 | C
NTARF | 05 | R
NJOURF | 00 | &
NJOURF+1 | 00 | B
PJOURF+1 | 00004001 06004002 16004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | 1
PPOINTE | 00004005 06004006 16004005 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE | *

NicolasBernaerts, 2025-01-15:
C'est beaucoup mieux !
A priori votre fournisseur publie les heures de pointe en hexa décimal, les HC commencent à 16h.
Une commande 
energyconfig calhexa 1
devrait régler le problème.

sosandroid, 2025-01-15:
Merci du support !
J'ai injecté la config

sosandroid, 2025-01-18:
Validé. Je ferme, j'en ouvre un autre vis-à-vis de Berry script

closed 2025-01-18 by sosandroid