NicolasBernaerts/tasmota #55: Accès aux data via Berry

sosandroid, 2025-01-18:
Il y a plusieurs publication de data par le firmware, je synthétise ici celles qui m'intéressent

````
SENSOR
{
    "Time": "2025-01-18T23:51:20",
    "METER": {
    },
    "CONTRACT": {
    },
    "RTE": {
    },
    "CAL": {
    },
    "ALERT": {
    }
}
RTE
{
	"Time":"2025-01-18T23:51:20",
	"TEMPO":{
	}
}
LIVE 
{
	"METER":{
	}
}
````

Pour y accéder via Berry, j'utilise un script comme ceci

````berry
def print_obj(value, trigger)
  print(value)
end

tasmota.add_rule('TEMPO', print_obj)
tasmota.add_rule('RTE', print_obj)
tasmota.add_rule('METER', print_obj)
tasmota.add_rule('CONTRACT', print_obj)
````

les règles METER et TEMPO fonctionnent
les règles RTE et CONTRACT ne fonctionnent pas

A quoi est-ce dû ? Ai-je raté quelque chose ?

L'objectif est de pousser une sous-ensemble vers EmonCMS en requête HTTP

NicolasBernaerts, 2025-01-26:
Bonsoir, 
En regardant rapidement sur le site emoncms, il semble que vous puissiez ajouter simplement un broker MQTT sur l'installation pour être compatible avec les flux MQTT.
Cela semble être de loin le plus simple sachant que mon fork oublié toutes les données en MQTT.
Vous aurez alors une solution nativement opérationnelle dans aucun besoin de scripting Berry.

sosandroid, 2025-01-26:
Bonsoir,

Merci d'avoir regardé.
Effectivement il y a un broker mqtt installé. EmonCMS l'écoute mais attend un schéma 
/Device/topic/data
/Device/topic2/data

Il ne sait pas traiter une schéma 
/Device/{Json data}

NicolasBernaerts, 2025-01-26:
D'après
https://docs.openenergymonitor.org/emoncms/mqtt.html
il semblerait qu'une publication MQTT sous 
emon/#
est directement visible sous EmonCms.
Avez vous essayé de configurer tasmota pour publier sous ce topic de base ?

NicolasBernaerts, 2025-02-08:
Fil abandonné