NicolasBernaerts/tasmota #56: Recuperation valeur couleur du jour tempo par mqtt sur HA incorrecte

fastm3, 2025-01-20:
14.9beta11
Probleme recup couleur du jour tempo. J'ai remis en place RTE et l'api . Tout est ok. 
Neanmoins sur les infos publiés par mqtt, quand j'essaie de recuperer la couleur du jour . J'ai une mauvaise info avec le sensor defini par discovery.
Samedi , on etait en blanc . RTE.today  est bien blanc mais le sensor en autodiscovery "sensor.teleinfo_couleur_aujourdhui" utilise CONTRACT.tday qui est bleu ( erreur ).  Ils devraient avoir la meme valeur, non ?

Aujourd'hui, RTE.tday et tmrw est bien rouge mais CONTRACT.tday reste bleu.  Normal ?  Et comme le sensor en autodiscovery est basé sur ce topic ( et pas celui de RTE correct ) , la valeur remontée dans home asssistant est incorrecte.
Peut etre serait il possible de declarer en mqtt autodiscovery  quand activé RTE.today et RTE.tmrw qui ont les bonnes valeurs ?
Le contournement actuel est simple. Déclarer en manuel ces 2 sensors mas les 2 en autodiscovery resteraient alors incorrects.

Mon topic en MQTT exemple:
{
	"Time": "2025-01-18T12:14:50",
	"METER": {
		"PH": 1,
		"ISUB": 60,
		"PSUB": 12000,
		"PMAX": 12000,
		"U1": 229,
		"I1": 3.41,
		"P1": 782,
		"W1": 0,
		"U": 229,
		"I": 3.41,
		"P": 782,
		"W": 0,
		"C": 0.0,
		"YDAY": 22115,
		"TDAY": 5662
	},
	"CONTRACT": {
		"serial": 812164578022,
		"name": "Tempo",
		"period": "Pleines Blanc",
		"color": "Blanc",
		"hour": "Pleine",
		"tday": "Bleu",
		"tmrw": "Bleu",
		"CONSO": 23847742,
		"HC BLEU": 11578651,
		"HP BLEU": 8328891,
		"HC BLANC": 1204904,
		"HP BLANC": 1785272,
		"HC ROUGE": 629798,
		"HP ROUGE": 320226
	},
	"RTE": {
		"tday": "Blanc",
		"tmrw": "Bleu"
	},
	"RELAY": {
		"R1": 0,
		"R2": 0,
		"R3": 0,
		"R4": 0,
		"R5": 0,
		"R6": 0,
		"R7": 0,
		"R8": 0,
		"P1": 0,
		"L1": "Creuses Bleu",
		"P2": 0,
		"L2": "Pleines Bleu",
		"P3": 0,
		"L3": "Creuses Blanc",
		"P4": 1,
		"L4": "Pleines Blanc",
		"P5": 0,
		"L5": "Creuses Rouge",
		"P6": 0,
		"L6": "Pleines Rouge"
	},
	"CAL": {
		"level": 2,
		"hp": 1
	},
	"ALERT": {
		"Load": 0,
		"Volt": 0,
		"Preavis": 0,
		"Label": ""
	}
}

sosandroid, 2025-01-20:
Sauf erreur de ma part, dans CONTRACT, les données sont issues du décodage STGE. Données que j'ai constatée moyennement fiable.
La valeur de RTE est plus fiable.

NicolasBernaerts, 2025-01-21:
Effectivement, tday et tmrw dans CONTRACT étaient issues de STGE qui ne s'avère absolument pas fiable. 
Je suis en train de les modifier pour les extraire du calendrier calculé. Cela devrait être beaucoup plus fiable. 
Mais pour les contrats Tempo, la donnée la plus fiable est clairement celle issue du plugin calendrier RTE.

NicolasBernaerts, 2025-01-21:
Je parle du plugin RTE intégré dans mon fork Tasmota.
Je vais regarder pour ajouter les 2 infos RTE côté HA.
Pour la publication sous homeassistant/sensor/DCDA0C19E020_CONTRACT_NAME, je respecte les préconisations de publication HA. Si vous renommez votre Tasmota en TELEINFO sous WIFI / Nom d'hôte cela devrait répondre à votre demande.

fastm3, 2025-01-21:
oui, l'ajout des sensors rte en autodiscovery mqtt serait top. C'est noté pour le reste.  Merci.

NicolasBernaerts, 2025-03-20:
Je viens de publier une version v14.9 où la déclaration Home Assistant des données TEMPO de RTE a été ajoutée.
Cela permettra de disoser des données RTE directement depuis Tasmota dans HA.

fastm3, 2025-03-20:
Ok. J'ai testé l'update. J'ai juste mis un peu de temps a m'y retrouver cherchant une reference à RTE.  On a donc desormais "tempo aujourd'hui" et "couleur aujourd'hui" , c'est ok. Remarque, j'aurai juste fait référence a RTE dans le nouveau nom pour savoir qui est qui. Pas evident de déterminer de suite quelle entité a utiliser.  Merci bcp pour le suivi.

NicolasBernaerts, 2025-03-20:
J'ai mis TEMPO car RTE publie 3 calendriers différents : TEMPO, POINTE et ECOWATT. Chacun aura son étiquette de calendrier.

NicolasBernaerts, 2025-03-31:
Clôturé faute de mise à jour