NicolasBernaerts/tasmota #57: Rémontée et publication des données KO en mode TIC standard

Adsum02, 2025-03-13:
Bonjour,

tout d'abord merci pour le travail effectué :)

Je me heurte à un problème de remontée et de publication des données (sur l'interface et sur MQTT) depuis le passage en mode TIC standard.

Quelques infos de base sur mon Hardware/software:

[image]

Côté conf MQTT, elle est correcte (et n'a pas changé):

[image]

Je ne "vois" plus rien dans l'interface:

[image]

Là où avant j'avais des informations de conso de ce type:

[image]

Et j'ai l'impression qu'il manque des infos dans la trame MQTT:
02:05:13.514 MQT: tele/denky_F5BE34/SENSOR = {"Time":"2025-03-13T02:05:13","ESP32":{"Temperature":53.3},"METER":{"PH":1,"ISUB":45,"PSUB":9000,"PMAX":9000},"CONTRACT":{"serial":21961741927,"name":"HC SEM WE MERCR","period":"HC SEM WE MERCR","color":"Inconnu","hour":"Pleine","tday":"Inconnu","tmrw":"Inconnu","CONSO":0},"CAL":{"level":0,"hp":1,"tday":{"0":1,"1":1,"2":1,"3":1,"4":1,"5":1,"6":1,"7":1,"8":1,"9":1,"10":1,"11":1,"12":1,"13":1,"14":1,"15":1,"16":1,"17":1,"18":1,"19":1,"20":1,"21":1,"22":1,"23":1,"24":1,"25":1,"26":1,"27":1,"28":1,"29":1,"30":1,"31":1,"32":1,"33":1,"34":1,"35":1,"36":1,"37":1,"38":1,"39":1,"40":1,"41":1,"42":1,"43":1,"44":1,"45":1,"46":1,"47":1},"tmrw":{"0":1,"1":1,"2":1,"3":1,"4":1,"5":1,"6":1,"7":1,"8":1,"9":1,"10":1,"11":1,"12":1,"13":1,"14":1,"15":1,"16":1,"17":1,"18":1,"19":1,"20":1,"21":1,"22":1,"23":1,"24":1,"25":1,"26":1,"27":1,"28":1,"29":1,"30":1,"31":1,"32":1,"33":1,"34":1,"35":1,"36":1,"37":1,"38":1,"39":1,"40":1,"41":1,"42":1,"43":1,"44":1,"45":1,"46":1,"47":1}},"ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""},"TempUnit":"C"}

La conf téléinfo est configurée pour publier à chaque évolution de 500W (comme conseillé) mais à part le message MQTT au redémarrage, je n'en ai pas d'autre (alors que je démarrage éteint un groupe de chauffage de 4500 W donc bien au delà du seuil qui déclenche la variation).

Du positif (quand même !), au niveau des messages TIC ça semble bon:

[image]

J'ai tenté plusieurs actions:

- Reset contrat (j'avais un message de warning m'indiquant que le contrat avait changé)
- Upgrade Firmware via OTA
- Mise à jour du firmware via upload de fichier

Est-ce que j'ai mal/pas fait une action suite au changement de contrat ?
Est-ce que c'est mon contrat qui est mal interprété ?

Merci d'avance pour le retour et bonne nuit (ou journée) !

NicolasBernaerts, 2025-03-14:
Hello,
Merci pour le feedback.
Je n'ai jamais eu de retour sur ce type de contrat.
J'aurais besoin d'un log des données TELEINFO pour reproduire le problème.
Pouvez vous enregistrer le flux TeleInfo pendant 5 mn et me l'envoyer que je reproduise ?
Commande :
tcp_start 888
par exemple ...

Adsum02, 2025-03-14:
Hello,

oui bien sûr :)

Voici !

[teleinfo.log](https://github.com/user-attachments/files/19254341/teleinfo.log)

J'ai fait varier la consommation pendant la prise de traces pour avoir un peu de matière aussi.

Une piste:
Quand je regarde la console dans l'interface par rapport à la trace du tcp, j'ai l'impression qu'il se perd dans un champ, peut-être en lien avec le nom du contrat.

21:33:53.506 MQT: tele/denky_F5BE34/SENSOR = {"Time":"2025-03-14T21:33:53","ESP32":{"Temperature":53.3},"METER":{"PH":1,"ISUB":45,"PSUB":9000,"PMAX":9000},"CONTRACT":{"serial":21961741927,"name":"HC SEM WE MERCR","_**period":"HC SEM WE MERCR"**_,"color":"Inconnu","hour":"Pleine","tday":"Inconnu","tmrw":"Inconnu","CONSO":0},"CAL":{"level":0,"hp":1,"tday":{"0":1,"1":1,"2":1,"3":1,"4":1,"5":1,"6":1,"7":1,"8":1,"9":1,"10":1,"11":1,"12":1,"13":1,"14":1,"15":1,"16":1,"17":1,"18":1,"19":1,"20":1,"21":1,"22":1,"23":1,"24":1,"25":1,"26":1,"27":1,"28":1,"29":1,"30":1,"31":1,"32":1,"33":1,"34":1,"35":1,"36":1,"37":1,"38":1,"39":1,"40":1,"41":1,"42":1,"43":1,"44":1,"45":1,"46":1,"47":1},"tmrw":{"0":1,"1":1,"2":1,"3":1,"4":1,"5":1,"6":1,"7":1,"8":1,"9":1,"10":1,"11":1,"12":1,"13":1,"14":1,"15":1,"16":1,"17":1,"18":1,"19":1,"20":1,"21":1,"22":1,"23":1,"24":1,"25":1,"26":1,"27":1,"28":1,"29":1,"30":1,"31":1,"32":1,"33":1,"34":1,"35":1,"36":1,"37":1,"38":1,"39":1,"40":1,"41":1,"42":1,"43":1,"44":1,"45":1,"46":1,"47":1}},"ALERT":{"Load":0,"Volt":0,"Preavis":0,"Label":""},"TempUnit":"C"}

Dans le "period", il ne devrait pas plutôt récupérer "HP SEMAINE" à la place de "HC SEM WE MERCR" ?

Pour info mon contrat est celui-ci: https://particulier.edf.fr/fr/accueil/electricite-gaz/zen-week-end-plus.html
Et le jour choisi est le mercredi, d'où le "MERCR" dans le nom du contrat.

Merci du check, je reste dispo pour quoi que ce soit !

Adsum02, 2025-03-14:
Si ça peut aider aussi, j'ai 6 index de consommation différents à l'affichage sur le Linky (en faisant défiler avec les boutons):

- HC SEMAINE
- HP SEMAINE
- HC WEEK END
- HP WEEK END
- HC MERCREDI
- HP MERCREDI

NicolasBernaerts, 2025-03-15:
J'ai reproduit le problème grâce à votre enregistrement. Pourriez vous faire un autre enregistrement de quelques sur une autre période ? Merci

Adsum02, 2025-03-15:
Bonjour,

super nouvelle !

Voici, cette fois-ci on est en "HP WEEK END"

[teleinfo_15032025.log](https://github.com/user-attachments/files/19260195/teleinfo_15032025.log)

Merci !

NicolasBernaerts, 2025-03-16:
Merci c'est top !
Je suis totalement preneur pour des enregistrements de périodes inconnues.
J'ai une piste concernant le problème.

Adsum02, 2025-03-16:
De rien 😊

Il me reste « HC SEMAINE », « HC MERCREDI » et « HP MERCREDI », je pourrais envoyer les traces de chacune de ces périodes d’ici mercredi.

Ça porte sur quoi la piste ?

Adsum02, 2025-03-16:
Ah oui je comprends, difficile de prévoir sans avoir les contrats exotiques et leurs spécificités aussi.

NicolasBernaerts, 2025-03-16:
Cela se présente mieux :-)

[image]

NicolasBernaerts, 2025-03-16:
La publication du calendrier Linky annonce des changements de tarif en format hexadécimal aux heures suivantes : 
- minuit
- 0h56
- 7h56
- 13h56
- 14h56
Est-ce que c'est bien le cas pour vous ?

Adsum02, 2025-03-17:
> Cela se présente mieux :-)
> 
> [image]
> 
> J'attends votre enregistrement HC SEMAINE pour valider que la détection HC / HP fonctionne bien.

Pas mal !

Et voilà l'enregistremement pour "HC SEMAINE", c'est tout frais :)

[HC_SEMAINE.log](https://github.com/user-attachments/files/19276537/HC_SEMAINE.log)

Adsum02, 2025-03-17:
> La publication du calendrier Linky annonce des changements de tarif en format hexadécimal aux heures suivantes :
> 
> * minuit
> * 0h56
> * 7h56
> * 13h26
> * 14h26
> 
> Est-ce que c'est bien le cas pour vous ?
> 
> [image]

Pour le changement à minuit je ne sais pas trop (changement auto pour chaque jour ?) mais pour le reste, c'est bien ça oui:

[image]

Adsum02, 2025-03-19:
Bonjour @NicolasBernaerts !

J'espère que vous allez bien :)

Je viens de prendre les traces pour les deux périodes de mercredi au cas où:

[HC_MERCREDI.log](https://github.com/user-attachments/files/19330145/HC_MERCREDI.log)

[HP_MERCREDI.log](https://github.com/user-attachments/files/19330150/HP_MERCREDI.log)

Bonne nuit !

NicolasBernaerts, 2025-03-19:
Merci !

J'espère pouvoir publier une version bêta qui corrige le problème de l'auto découverte des contrats dans la soirée.

NicolasBernaerts, 2025-03-20:
Bonjour,
J'ai uploadé une v14.10beta1 pour DenkyD4 qui devrait régler le problème de l'autodécouverte du contrat.

Adsum02, 2025-03-20:
Hello, ça à l’air de fonctionner 😊

Sur les calendriers tout est sur la même période par contre, je ne sais pas si c’est normal:

[image]

Je laisse tourner pour voir ce que ça donne dans le temps.

Merci et bonne journée !

NicolasBernaerts, 2025-03-20:
C'est logique car une seule période est connue pour le moment. Laissez tourner la semaine et les informations devraient apparaître au fur et à mesure.
Les informations sont stockées sur le FS.
Elles sont donc conservées après reboot.

Adsum02, 2025-03-20:
Super, merci 😊

Adsum02, 2025-03-26:
Bonjour,

Après une semaine, j’ai bien tous les index visibles et cohérents, c’est tout bon 😊

Un détail au cas où, je ne sais pas si c’est un comportement normal mais pour que la première remontée d’un index (pour le week-end ou le mercredi par exemple) soit visible dans HA, j’ai du faire un restart via l’interface. Après ça, tout était visible et synchronisé.

Merci pour tout !

NicolasBernaerts, 2025-03-26:
Bonsoir,
C'est effectivement logique car actuellement la publication des index HA ne se fait qu'une seule fois au démarrage. Donc les index découverts en cours de route ne sont pas publiés.
Je vais voir s'il est possible de republier suite à une découverte.

closed 2025-03-30 by Adsum02