NicolasBernaerts/tasmota #58: Consommations disparues

jefftut123, 2025-03-15:
Bonjour,

Tout d'abord félicitations pour tout cet énorme travail.
Un petit soucis avec le 14.9 sur un ESP32 4M. Il n'y a plus d'affichage du volet Consommations alors qu'il existait sur la 14.8

A votre disposition pour tout log nécessaire.

Cordialement,

Jeff

NicolasBernaerts, 2025-03-15:
Bonjour, 
Effectivement je suis en train de refondre totalement les graphiques. 
Les courbes sont refondues mais pas les historiques. Si vous souhaitez conserver les historiques il est préférable de rester en v14.8 en attendant la nouvelle version.

jefftut123, 2025-03-15:
Bonsoir,
Merci pour la rapidité de votre réponse. Bon courage pour cette refonte car j'imagine que c'est quelque peu complexe. Encore merci pour votre formidable travail.
Bien cordialement,

Jeff

NicolasBernaerts, 2025-03-15:
Merci Jeff 😀

NicolasBernaerts, 2025-04-13:
Bonsoir,
La version v14.10 publiée ce soir amène une refonte complète des graph temps réel et des graphs d'historiques.
Les données historisées intégrent maintenant la production et les périodes de conso (type Tempo).
De ce fait, les anciennes données ne pourront être visualisées avec cette nouvelle version.
Elles seront toutefois conservées sur le FS.