NicolasBernaerts/tasmota #59: Perte Accès "Consommation" des données historiques suite update 14.9

djelloul78, 2025-03-15:
Bonjour, 

Avant toute chose, je tenais à vous remercier pour ce formidable travail réalisé avec une telle précision de restitution des données TIC via Tasmota.
Le résultat obtenu est incroyable, et la restitution des informations TIC ainsi que l'exploitation que l'on peut en faire grâce à vous et les travaux de "Hallard" est vraiment super, surtout au regard de la simplicité d'accostage électronique. 

Je fais actuellement face à un problème auquel je ne sais pas dire si celui-ci est dû à une mauvaise utilisation de ma part ou si cette évolution est normale.
À la première mise en œuvre de ce projet, j'avais flashé le firmware en version 14.8 sur un Wemos Mini en base ESP32.
Tout fonctionne comme décrit dans le GitHub, notamment la rubrique "Consommation" qui permet de consulter la consommation mesurée via TIC sur les précédents jour, semaines et mois.

Depuis la publication du firmware en version 14.9, j'ai exécuté une mise à jour via OTA en me trompant de fichier binaire en utilisant le lien [tasmota-teleinfo.bin](https://github.com/NicolasBernaerts/tasmota/raw/refs/heads/master/teleinfo/binary/tasmota-teleinfo.bin) plutôt que [tasmota32-teleinfo.bin](https://github.com/NicolasBernaerts/tasmota/raw/refs/heads/master/teleinfo/binary/tasmota32-teleinfo.bin).
À la suite de cette erreur, le module Tasmota s'est configuré en Safe Mode en attente d'un firmware valide à flasher via OTA.
Une fois mon erreur identifiée, j'ai appliqué la bonne URL et cette fois la mise à jour s'est bien déroulé. 
Cependant, à la suite de cette manipulation, la rubrique "Consommation" affichant les données historiques n'apparaît plus dans l'interface Tasmota du module.
Pourtant, les fichiers *.CSV semblent toujours présents au sein de ce dernier car je peux les observer via l'outil "Gestionnaire de Fichiers" de la rubrique "Tools".

Je voudrais juste savoir si ce phénomène est normal ou si mon erreur de manipulation m'a fait perdre l'accès à la rubrique "Consommation" qui va me contraindre à reflasher le module via USB, qui le fera perdre les données déjà acquises, afin de pouvoir récupérer cette fonctionnalité? 

Merci beaucoup pour le temps que vous m'accorderez et merci encore pour le partage de vos travaux qui sont vraiment super utile pour qui sait les exploiter.

NicolasBernaerts, 2025-03-15:
Bonjour, 
Effectivement je suis en train de refondre totalement les graphiques. 
Les courbes sont refondues mais pas les historiques. Si vous souhaitez conserver les historiques il est préférable de rester en v14.8 en attendant la nouvelle version.

closed 2025-03-15 by NicolasBernaerts

djelloul78, 2025-03-15:
Merci beaucoup pour votre rapide réponse qui explique pourquoi je n'observais plus les données historiques suite à cette mise à jour.

Je patienterai jusqu'à la prochaine release qui remettra cette fonctionnalité sur le devant de la scène, qu'importe la version. 

Merci à nouveau.

NicolasBernaerts, 2025-04-13:
Bonsoir,
La version v14.10 publiée ce soir amène une refonte complète des graph temps réel et des graphs d'historiques.
Les données historisées intégrent maintenant la production et les périodes de conso (type Tempo).
De ce fait, les anciennes données ne pourront être visualisées avec cette nouvelle version.
Elles seront toutefois conservées sur le FS.

djelloul78, 2025-04-13:
Bonsoir, 

Merci pour ce retour. 
Je test ce firmware dès mon retour à domicile et me permettrait de vous faire un feedback si cela vous intéresse 

Encore merci pour votre investissement personnel à faire bien les choses pour toute une communauté d'utilisateurs de Linky (quasiment tous les logements français) et débrouillard avec la galaxie Tasmota.

NicolasBernaerts, 2025-04-14:
N'hésitez pas a me faire un feedback :-)