NicolasBernaerts/tasmota #61: Réafectation automatique du GPIO "Tinfo Rx" sur "OUTPUT" à la mise sous tension

Zboubie, 2025-04-17:
Bonjour,
J'utilise le shield WeMos Teleinfo ( https://www.tindie.com/products/hallard/wemos-teleinfo/ ) de Charles Hallard + un ESP32-S2 mini.
La pin 11 de l'ESP32 est assignée a "TInfo Rx"

Avec le firmware original Tasmota, tout fonctionne correctement.

Cependant, lorsque j'utilise la version 14.9 ou 14.10 (je n'ai pas essayé avec les versions antérieures).
Si le Linky est branché à la mise sous tension de l'ESP32, en regardant via GPIO Viewer,  je me suis aperçu que la pin 11 "Tinfo Rx", passe de "Input with Pull-Up"  à "OUTPUT". Ce qui bloque la remontée de télémétrie du Linky.

Le seul moyen de corriger ce problème est de désaffecter la PIN 11, débrancher le linky. Redémarrer électriquement l'ensemble, puis rebrancher le Linky. A partir de ce moment, la pin 11 ne change pas de type

Le problème, c'est qu'en cas de coupure de courant, toute la manip est a refaire ...

Je me demande s'il ne s'agit pas du même problème qu'avec l'utilisation de l'ESP8266 qui désactive la teleinfo si le linky est branché lors de la mise sous tension.

Sébastien

NicolasBernaerts, 2025-04-18:
Bonjour Sébastien,

C'est très étonnant car sur un ESP32 S3, la GPIO44 sélectionnée est bien vue comme input.

[image]

[image]

Zboubie, 2025-04-18:
Pourtant j'ai essayé avec 2 ESP32-S2 acheté auprès de différents vendeurs, et j'ai le exactement le même comportement. 

Avec un ESP8266, si le Linky est branché à la mise sous tension, ca bug aussi. Soit la Teleinfo passe sur "Désactivé" soit la configuration du module bascule sur le profile générique. 
Si je configure l'ESP8266 et qu'ensuite je branche le linky, là ca fonctionne un certains temps. (j'ai l'impression qu'au bout d'un moment l'ESP fini par rebooter, et donc RAZ de la config)

NicolasBernaerts, 2025-04-21:
En regardant le pinout de l' esp32s2 mini, vous utilisez gpio6.

[image]

Cette gpio6 semble être connectée a la mémoire SPI et il est déconseillé de l'utiliser. Peut être est le problème au boot si un signal existe sur cette gpio6.

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Zboubie, 2025-04-22:
Je ne suis pas sure qu'elle soit utilisée. La 6 n'est ni configurée, ni électriquement connectée (seule la rangée de pin extérieures de l'ESP est raccordée au bornier du shield GPIO3, 5, 7, 9...)

[image]

[image]

NicolasBernaerts, 2025-04-22:
Vous parliez de la pin 11 qui est logiquement gpio06. Bonne recherche.

Zboubie, 2025-04-22:
mea culpa! J'avais bien formulé dans le titre, puis utilisé la mauvaise terminologie dans le message.

En fait, je ne savais pas que certaines GPIO étaient réservées. 
Le shield quand à lui une trace a couper/souder pour passer de la GPIO11 à la GPIO37 (PIN "Rx" sur le shield) celui-ci risque t'il de poser problème ?
[image]

Ce que je ne comprends pas vraiment, c'est que le firmware officiel Tasmota-Teleinfo n'est pas affecté par ce problème

NicolasBernaerts, 2025-04-22:
Essayez de switcher sur la GPIO RX via la coupure et la soudure. Il n'y a pas de risque et cela évitera d'utiliser une pin réservée.

jujuinparis, 2025-04-28:
Hello

je me permets d'intervenir dans le sujet car je me prends la tête depuis hier et je pense que j'ai le meme souci que Zboubie

J'avais un Wemos D1 sous Tasmota Teleinfo assez ancien (12 ou 13 je crois) et j'ai voulu mettre à jour hier pour avoir le support TLS over MQTT et la drame ...

Je suis passé sur ton firmware et impossible de le faire fonctionner : 

le TInfoRX est sur GPIO13, cela fonctionne parfaitement sur tasmota 14 avec teleinfo de Charles

Avec ton firmware, il enregistre bien le TinfoRX sur GPIO13 mais dès que je passe à Mode Historique, cela reboot et il me remet en Sonoff Basic et toute la config est perdue.

Un idée avant de retourner au firmware de Charles ? 

Merci !

jujuinparis, 2025-04-28:
Hello

je confirme 👍 

- Si le module est branché au linky, dès que je reboote pour appliquer le mode Historique (pas de vitesse selectionnée), il me remet en sonoff basic donc impossible de le faire fonctionner
- Si le module n'est pas branché au linky. je fais la configuration, je le branche et la cela fonctionne parfaitement

NicolasBernaerts, 2025-04-28:
Je viens de publier une esp8266 4M en version v14.11beta1.
Elle intègre différentes optimisations pour essayer de régler le problème de boot avec réception TIC.
Etant dans le TGV, aucun moyen de la tester ...
Dites moi si c'est mieux.
Si le problème persiste, je ferai des optimisations mémoire plus violentes pour le version esp8266.

NicolasBernaerts, 2025-04-29:
J'ai réussi à reproduire clairement le problème sur un esp8266 16M.
Les coupables sont les courbes graph + histo qui sont beaucoup trop demandeuses en RAM (environ 20k) pour ces processeurs.
Sous Tasmota, si un esp8266 descend en dessous de 10kB dispo (sur 41kB), il devient instable.
Une solution simple et testée serait de supprimer purement et simplement les graphs et histo de la version esp8266.
Mais je vais essayer de voir si je ne peux pas conserver les histo et le graph live à minima.
A suivre ...

jujuinparis, 2025-05-02:
hello

désolé pour la réponse tardive.

J'ai remis ma board sous téléinfo de Charles car cela me demandait de modifier également du code sous Jeedom pour adapter à ce plugin

j'ai une nouvelle board Wemos S2 qui doit arriver d'ici quelques jours, je partirai sur ton fork et je te dirai ce qu'il en est

Merci en tout cas pour le fix rapide !

NicolasBernaerts, 2025-05-04:
J'ai testé en production la dernière version publiée sur un esp8266 en mode historique et sur un esp8266 16M en mode standard sans aucun problème au boot. Suite à ces tests je clos la fiche.

closed 2025-05-04 by NicolasBernaerts