NicolasBernaerts/tasmota #66: Plantage complet en sélectionnant GPIO16 comme Rx sur ESP32 Wroom

TDSnet, 2025-09-24:
Bonjour,
J'utilise Teleinfo sur Tasmota 15.0.1 depuis quelques semaines sans trop de problème sur une ESP32 4M de type Wroom32 (Devkit). 
L'UART2 de ce module est sur GPIO16 pour Rx. En basculant sur votre version, l'ESP32 plante définitivement (boot permanent) lorsque l'on configure GPIO16 en Tinfo Rx. 
J'ai testé avec le BIN Tasmota32-teleinfo.bin (15.1b11) et une version compilée localement. (15.0)

[image]

Olivier

NicolasBernaerts, 2025-09-25:
Si vous sélectionnez GPIO3 tout fonctionne correctement ?

TDSnet, 2025-09-25:
En sélectionnant GPIO3 ca ne plante plus a la sauvegarde de la config. Je ferais des tests pour vérifier le fonctionnement. 
De toute facon je suis bon pour refaire les circuits imprimés... C'est surprenant cette différence entre la version officielle qui fonctionne sans problème avec l'utilisation de GPIO16 (UART2 RX préconisé par plusieurs utilisateurs) et votre version qui plante au redémarrage des que l'on choisit cette broche. 
Je vous tiens au courant des progrés pour le preciser dans la doc... Merci pour tout
Olivier

NicolasBernaerts, 2025-09-25:
Je viens de faire un test sur un ESP32 WROOM sans PSRAM :
GPIO16 ou 17 : aucune réception, mais pas de plantage
GPIO21 ou 22 : réception Teleinfo ok

J'ai vu sur des forums qu'il faut éviter GPIO16 et 17 car ces GPIO sont utilisées sur certaines implémentations ESP32 pour la gestion PSRAM. Cela expliquerait le plantage si vous avez un ESP32 avec PSRAM. La différence entre la version officielle et mon fork vient sans doute du fait que mes paramètres de compilation autorisent la PSRAM.

[image]

Donc en conclusion, il faut éviter GPIO16 et GPIO17 ...

NicolasBernaerts, 2025-09-28:
C'est assez ettonant car chez moi GPIO03 (RxD) fonctionne sans problème.
Il est à noter que sur un certain nombre de modules duplicate, RxD et TxD sont inversés au niveau de l'impression.

closed 2025-09-28 by NicolasBernaerts