NicolasBernaerts/tasmota #67: Denky V15.1 tempo issue

loorisr, 2025-09-28:
Bonjour,

J'ai migré de la version 14.11beta5 vers la v15.1beta11 et les compteurs tempo ne sont pas relevé. Par contre dans l'onglet message ils apparaissent bien et sont corrects. je ne sais pas d'où vient le problème.
J'ai une carte Denky D4

[image]

Merci pour ce super fork de tasmota!

NicolasBernaerts, 2025-09-28:
Hello,
C'est très etonnant car en rejouant un flux Tempo Conso + Prod, j'ai toutes les données qui sont correctement traitées.

[image]

Est-ce que vous pourriez enregistrer un flux TIC, la procédure est expliquée dans ma doc Github (tcp_start 888 ...), et me l'envoyer par mail afin que je le rejoue et regarde quel peut être le problème ?

loorisr, 2025-09-29:
Bonjour,

Effectivement, problème réglé!

Merci!

closed 2025-09-29 by loorisr